grblHAL/ESP32 #2: weak symbol crashing the MCU

kdomanski, 2021-03-21:
Since the recent merger of the testing branch into master in the original grblHAL monorepo, the grblHAL crashes on my ESP32 boards with the following stack trace:
```
GrblHAL 1.1f ['$' for help]
<Idle|MPos:0.000,0.000,0.000|Bf:35,1023|FS:0,0|Pn:XYZ|Ov:100,100,100>
ets Jun  8 2016 00:22:57
rst:0x1 (POWERON_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:2
load:0x3fff0030,len:1316
load:0x40078000,len:12224
ho 0 tail 12 room 4
load:0x40080400,len:3020
entry 0x400805e4
Guru Meditation Error: Core  1 panic'ed (InstrFetchProhibited). Exception was unhandled.
Core  1 register dump:
PC      : 0x00000000  PS      : 0x00060b30  A0      : 0x800d4c27  A1      : 0x3ffb5600
A2      : 0x3ffc3078  A3      : 0x00000000  A4      : 0x00000150  A5      : 0x3ffc31c8
A6      : 0x3ffbede8  A7      : 0x00000015  A8      : 0x800e06c6  A9      : 0x000203fb
A10     : 0x00000001  A11     : 0x3ffc2434  A12     : 0x3ff000c4  A13     : 0x3ffbed50
A14     : 0x00000001  A15     : 0x3ffc073c  SAR     : 0x00000009  EXCCAUSE: 0x00000014
EXCVADDR: 0x00000000  LBEG    : 0x400012e5  LEND    : 0x40001309  LCOUNT  : 0x800e0ec7
Backtrace:0xfffffffd:0x3ffb5600 0x400d4c24:0x3ffb5620 0x400d4b33:0x3ffb5640
ELF file SHA256: a30a8c1319f4e97b
CPU halted.
```

A bit of investigation showed that the MCU crashes on the call to `my_plugin_init()` in `driver.c`. Commenting that call out makes it work normally again. I experienced the issue on multiple boards.

terjeio, 2021-03-21:
I have commented it out as well - compiler or linker bug? Or is there a function with the same name in the framework?
It is an empty function and it worked with IDF v3.x, adding an empty function locally in driver.c works for some strange reason. Removing the weak attribute does not help.

terjeio, 2022-01-21:
This has been fixed recently.

closed 2022-01-21 by terjeio