grblHAL/ESP32 #3: HY Spindle + ESP32 + RS485 converter: RS485_DIR_ENABLE

fraro, 2021-06-07:
I'm stuck setting up my machine because I always get a spindle speed timeout alarm with this setup. I have a 3.3V capable TTL<->RS485 converter that only has TX/RX pins, no direction pin.
[image]

I found in driver.h:
#if SPINDLE_HUANYANG
#define RS485_DIR_ENABLE 1

which means that regardless of my setting in CMakeLists.txt
OPTION(RS485_DIR_OUT "Compile with RS485 direction pin enabled" OFF)
it's always set to ON (=have direction pin), correct?

I assume the timeout error is because of the direction pin missing?

Thanks!

terjeio, 2021-06-07:
> I assume the timeout error is because of the direction pin missing?

No, it should not be so since your board has auto direction sensing. The timeout is due to the VFD not responding to a request in time. Does it get it? Wiring is ok? Baud rate ok? VFD setup ok? Have I somehow messed up the code?

Which board are you compiling for?

terjeio, 2021-06-07:
> I found in driver.h:
#if SPINDLE_HUANYANG
#define RS485_DIR_ENABLE 1

This code is ok as there is a outer #ifdef around it. It is set to 1 if RS485_DIR_OUT is defined _and_ SPINDLE_HUANYANG is > 0, if not it is set to 0 as enabling the dir output does not make sense otherwise.

I have tested the code with my spindle simulator and it works as it should. Note that there are two incompatible variants of the Huanyang spindle protocol, CMakeLists.txt is set up for v1:

`target_compile_definitions("${COMPONENT_LIB}" PUBLIC SPINDLE_HUANYANG=1)`

fraro, 2021-06-07:
You are absolutely right. I missed the outer #ifdef RS485_DIR_ENABLE. I will double-check connection  and settings. Is there any helpful debugging I can enable?

closed 2021-06-07 by fraro

fraro, 2021-06-09:
Actually, this is what I did to get rid of the alarm: shorting TX and RX on the ESP side, with only RX connected to the RS485 converter. I can now at least switch on/off the spindle (using bCNC), setting rpm does not yet work (despite PD002 set to 2), spindle runs full speed only.
The debug code did print any received bytes count when having TX and RX connected correctly, only when looped.

So the VFD is receiving commands, but does not talk back. But this looks like a hardware or settings issue. Comm Params (PD163=1 164=1 165=3 with $374=2) seem to be ok.
Does GRBL read or write any registers/settings of the VFD via Modbus?

Thanks  a lot!

terjeio, 2021-06-10:
> Does GRBL read or write any registers/settings of the VFD via Modbus?

Yes, on every real-time report request the RPM register is read in `spindleGetState()`. The read is non-blocking meaning that the result returned is from the previous request.

```
    mode_cmd.adu[1] = ModBus_ReadInputRegisters;
    mode_cmd.adu[2] = 0x03;
    mode_cmd.adu[3] = 0x00;
    mode_cmd.adu[4] = 0x00;
    mode_cmd.adu[5] = 0x00;
    mode_cmd.tx_length = 8;
    mode_cmd.rx_length = 8;
```