grblHAL/ESP32 #4: Installation instuctions

ithinkido, 2021-06-26:
I do seem to be able to get the installation instructions to work for the ESP32. The first issue I found  that is an easy fix is 

```git clone -recursive git://github.com/grblHAL/ESP32```

should be with two - ( --)

```git clone --recursive git://github.com/grblHAL/ESP32```

But the second one,...

I am not a big git user, so I do not really know what is expected here. On running 
```git submodule update --remote```
I get this error
```fatal: Needed a single revision```
```Unable to find current origin/master revision in submodule path 'components/asio/asio'```

There have been a lot of changes to the repo structure lately , so I sure that this is just a small issue  that has come from the move, but would you mind giving the build/ install instructions a look over for this board.

terjeio, 2021-06-27:
Thanks, I have added the missing '-'.

I believe you are running the `git submodule update --remote` command from the wrong directory as there is no `components` subfolder in grblHAL (there is in the esp frameowrk), or have you managed to mix sources?

There is a missing cd the instructions - it should be like this:

terjeio, 2021-06-27:
I posted the previous comment too early, updated now. Sorry for that.