grblHAL/ESP32 #10: Is this code functional for anyone

m-g-m, 2021-11-21:
Honest question before I waste more hours debugging endless guru meditations.
Does this code work for anyone ? If so what hardware are you using ?
Thanks !

terjeio, 2021-11-21:
Oops, the latest version is broken - I'll try to get a fix out with an update later today.

If you want to give it a try now then the fix is replacing all occurences of `#ifndef VFD_SPINDLE` with` #if VFD_SPINDLE != 1` in [driver.c](https://github.com/grblHAL/ESP32/blob/master/main/driver.c) and in the selected board map file if present there.

troth530, 2021-11-22:
I have been trying to get this compiled and working with a ESP32-WROVER-B.  Had an issue with Ethernet and SDCard and may be related to FTP.  Here is the [issue](https://github.com/grblHAL/ESP32/issues/8#issue-1043763878).   but now cannot get WIFI working.  So I don't know if WEBUI is working and not sure if this is ready to go out of the box.

terjeio, 2021-11-22:
The version I committed yesterday works for me:

[image]

FTP is not started unless enabled with $70, 15 enables all protocols.

@troth530 - what are your settings in [CMakelists.txt](https://github.com/grblHAL/ESP32/blob/master/main/CMakeLists.txt)?

m-g-m, 2021-11-22:
OK, success with the latest check-in from today 11/21/21. It compiled after fixing the board include #ifdef . Basic machine stepping is working with ESPDUINO32 and protoneer-style shield.

troth530, 2021-11-22:
@terjeio I need to build with SDcard hardware to verify FTP functionality works which I haven't done yet.  

In CMakelist.txt, I am using default for everything except Networking (line 18) and SDCard (line 21) set to ON.  I am using ESP32-WROVER-B and WIFI is not broadcasting a SSID or working at all as far as I can tell.

terjeio, 2021-11-22:
@troth530 Default wifi mode is station, you will have to enable access point (AP) mode to make it broadcast anything:

`OPTION(SoftAP "Enable soft AP mode" OFF)`

I have not tested this in a long time, hopefully it still works...
Station mode works, for that you have to enter SSID and password in settings to connect to a access point:

```
*help wifi
---- WiFi:

$73: WiFi Mode:
    0 - Off
    1 - Station

$74: WiFi Station (STA) SSID, max: 64

$75: WiFi Station (STA) Password, max: 32
ok

```

troth530, 2021-11-22:
@terjeio 
Thank you for your help.  
Yes, I've tried that setting in CMakelist,txt as well but it is still not broadcasting a SSID.
With SoftAP set to ON.  This is my result.
Also settings in my_machine.h do not seem to trigger a need to recompile.

[VER:1.1f.20211029:]
[OPT:VNMSL,35,1024,3,0]
Target buffer size found
[NEWOPT:ENUMS,RT+,TC,SED,WIFI,SD]
[FIRMWARE:grblHAL]
[NVS STORAGE:*FLASH]
[DRIVER:ESP32]
[DRIVER VERSION:211029]
[BOARD:BDRING v4]
[WIFI MAC:24:6F:28:45:74:14]
[IP:0.0.0.0]
[PLUGIN:SDCARD v1.02]
ok
[MSG:WIFI STA DISCONNECTED]
[MSG:WIFI AP SCAN COMPLETED]
$73
$73=1
ok
$74
$74='GRBL'

terjeio, 2021-11-22:
AP mode works for me, both WebUI and FTP
Try a settings reset: $RST=* to set default values for AP mode, then restart - I had to do that.

[image]

The FTP port will not show up for you unless you have todays commit. Earlier versions uses port 21.

troth530, 2021-11-23:
@terjeio I think your code is different, for example $73 had 4 options for me, I pulled from your repo about 2 weeks ago using these commands.
   git clone --recursive git://github.com/grblHAL/ESP32
   git pull --recurse-submodules
   git submodule update --remote

Thank you!

$help wifi
---- WiFi:
$73: WiFi Mode:
    0 - Off
    1 - Station
    2 - Access Point
    3 - Access Point/Station
$74: WiFi Station (STA) SSID, max: 64
$75: WiFi Station (STA) Password, max: 32
$76: WiFi Access Point (AP) SSID, max: 64
$77: WiFi Access Point (AP) Password, max: 32

terjeio, 2021-11-23:
The `$help wifi` command outputs available commands/options and those depend on AP mode option enabled or not.
When enabled I get:
```
---- WiFi:
$73: WiFi Mode:
    0 - Off
    1 - Station
    2 - Access Point
    3 - Access Point/Station
$74: WiFi Station (STA) SSID, max: 64
$75: WiFi Station (STA) Password, max: 32
$76: WiFi Access Point (AP) SSID, max: 64
$77: WiFi Access Point (AP) Password, max: 32
```
which is the same as you get.

Did you reset the settings with `$RST=*` and try again?
Is $73 set to to 2? Other settings are ok to? The settings shown in the screen dump above are those that works for me.

I commited an update yesterday, if the one you have is not working please update to that.

troth530, 2021-11-23:
@terjeio, I stashed what I had, updated to the latest commit and it compiled with 
Networking, SDcard and SoftAP all 3 set to ON.  I believe this will meet my needs, but need to confirm.
With the 3 above including SDcard set ON, FtpDaemon set ON will not compile, but I don't know if it's needed.
I will confirm the hardware behaves as expected.
Yet, any changes to my_machine.h does not force it to recompile, are all changes intended through CMakeList.txt?
I will close the other issue.
Thank you!

terjeio, 2021-11-23:
> With the 3 above including SDcard set ON, FtpDaemon set ON will not compile, but I don't know if it's needed.

You get compilation errors?

> Yet, any changes to my_machine.h does not force it to recompile, are all changes intended through CMakeList.txt?

Yes, since _my_machine.h_ is disabled in [CMakeLists.txt](https://github.com/grblHAL/ESP32/blob/8799e21c46a884ef0c86eef32ba8fcbe9e2081b6/main/CMakeLists.txt#L141) by default.

troth530, 2021-11-23:
@terjeio here is the error with FtpDaemon set ON (SDcard set ON).
I see my_machine.h disabled now in the board section.
Thank you!

Compile error snippet
...
  ^~~~~~~
[52/57] Building C object esp-idf/main/CMakeFiles/__idf_main.dir/wifi.c.obj
**FAILED: esp-idf/main/CMakeFiles/__idf_main.dir/wifi.c.obj**
ccache C:\Users\troth530\.espressif\tools\xtensa-esp32-elf\esp-2021r2-8.4.0\xtensa-esp32-elf\bin\xtensa-esp32-elf-gcc.exe -DBOARD
...

terjeio, 2021-11-23:
@troth530  This compiler error makes no sense for me, no line number where it fails?
Corrupted file? Submodules not up to date? Does it go away if you disable FTP?

terjeio, 2021-11-23:
@troth530 I believe I've found the reason, the networking plugin somehow was not updated yesterday. My local master were not pushed by the script I use, I have now done it manually.

troth530, 2021-11-23:
@terjeio I believe there is more missing.  
I recall I had to manually enable FatFS dependency on a Teensy build.  Here is the error.
Thank you!

[56/57] Linking CXX executable grbl.elf
FAILED: grbl.elf
cmd.exe /C "cd . && C:\Users\troth530\.espressif\tools\xtensa-esp32-elf\esp-2021r2-8.4.0\xtensa-esp32-elf\bin\xtensa-esp32-elf-g++.exe  -mlongcalls -Wno-frame-address   @CMakeFiles\grbl.elf.rsp  -o grbl.elf  && cd ."
c:/users/troth530/.espressif/tools/xtensa-esp32-elf/esp-2021r2-8.4.0/xtensa-esp32-elf/bin/../lib/gcc/xtensa-esp32-elf/8.4.0/../../../../xtensa-esp32-elf/bin/ld.exe: esp-idf/main/libmain.a(ftpd.c.obj):(.literal.cmd_cdup+0x4): undefined reference to `f_chdir'
c:/users/troth530/.espressif/tools/xtensa-esp32-elf/esp-2021r2-8.4.0/xtensa-esp32-elf/bin/../lib/gcc/xtensa-esp32-elf/8.4.0/../../../../xtensa-esp32-elf/bin/ld.exe: esp-idf/main/libmain.a(ftpd.c.obj): in function `cmd_cdup':
c:\users\troth530\esp\esp-idf\esp32\build/../main/networking/ftpd.c:640: undefined reference to `f_chdir'
c:/users/troth530/.espressif/tools/xtensa-esp32-elf/esp-2021r2-8.4.0/xtensa-esp32-elf/bin/../lib/gcc/xtensa-esp32-elf/8.4.0/../../../../xtensa-esp32-elf/bin/ld.exe: esp-idf/main/libmain.a(ftpd.c.obj): in function `cmd_cwd':
c:\users\troth530\esp\esp-idf\esp32\build/../main/networking/ftpd.c:635: undefined reference to `f_chdir'
c:/users/troth530/.espressif/tools/xtensa-esp32-elf/esp-2021r2-8.4.0/xtensa-esp32-elf/bin/../lib/gcc/xtensa-esp32-elf/8.4.0/../../../../xtensa-esp32-elf/bin/ld.exe: esp-idf/main/libmain.a(vfs.c.obj):(.literal.vfs_getcwd+0x0): undefined reference to `f_getcwd'
c:/users/troth530/.espressif/tools/xtensa-esp32-elf/esp-2021r2-8.4.0/xtensa-esp32-elf/bin/../lib/gcc/xtensa-esp32-elf/8.4.0/../../../../xtensa-esp32-elf/bin/ld.exe: esp-idf/main/libmain.a(vfs.c.obj): in function `vfs_getcwd':
c:\users\troth530\esp\esp-idf\esp32\build/../main/networking/vfs.c:155: undefined reference to `f_getcwd'
collect2.exe: error: ld returned 1 exit status
ninja: build stopped: subcommand failed.
ninja failed with exit code 1

terjeio, 2021-11-24:
@troth530 That helps, same issue here as with the [iMXRT1062](https://github.com/grblHAL/iMXRT1062#sd-card-plugin) - _ffconf.h_ has to be edited and `#define FF_FS_RPATH` changed to 2. I see now that I did that back in may when I started porting ftpd.
On my machine _ffconf.h_  is located in _esp\esp-idf\components\fatfs\src_ in the ESP SDK, on my machine the SDK is in my user folder. I will add a note about this in the readme.

It is a pity that the FatFs source is not written in a way that allow setting configuration symbols like this in CMakeLists.txt, perhaps the author ChaN should be asked to change that?

troth530, 2021-11-24:
@terjeio correct.  In fact on the Teensy I used your patched utility/sd_sdhc.c  and had to #define FF_FS_PATH to 2. In addition to those items you had documented, for the Teensy I had to make several manual changes so the include statements would point to the correct location I'd placed the FatFS- maybe due to how I organized the files manually and my inexperience with GIT.  

On my computer there is also ffconf.h in the same location as yours in the ESP SDK and I made the change to FF_FS_PATH=2. Do you believe it is preferred to use the one in the SDK or FatFS by WMXZ-EU? 

However it still does not explain my issues with wifi not broadcasting, am I missing a manual step to turn wifi on for this device due to my inexperience?  FTPDaemon=ON compile error seems a downstream artifact of wifi not working and possibly something different in my code versus your code.  Confirming that I have used $RST=* and $73 is set to 2, I cannot set $70 to 15 (only can set o 7).  I have confirmed same behavior of wifi not broadcasting with both ESP WROVER-B and WROOM.

terjeio, 2021-11-25:
> Do you believe it is preferred to use the one in the SDK or FatFS by WMXZ-EU?

The SDK is used by the ESP32 driver, the WMXZ-EU by the iMXRT1062 driver - they cannot be swapped around as there is low level driver code added by the framework used by the drivers that is MCU specific.

> However it still does not explain my issues with wifi not broadcasting, am I missing a manual step to turn wifi on for this device due to my inexperience?

Add your CMakelists.txt and your settings here, preferably as attachments, and I'll check. If you are on Windows and have [ioSender](https://github.com/terjeio/ioSender) installed use _Help > About > To clipboard_ to get both `$I` and `$$` output and paste the result into a file,.

troth530, 2021-11-26:
@terjeio
The only setting changes are those in CMakeList.txt, if you'd like any other settings please let me know.  As always, Thank you!

[settings.txt](https://github.com/grblHAL/ESP32/files/7610343/settings.txt)
[CMakeLists.txt](https://github.com/grblHAL/ESP32/files/7610345/CMakeLists.txt)

terjeio, 2021-11-26:
Settings looks ok, I have flashed with your settings and it works for me. Can you try with the binary I made? I have uploaded it as _grblHAL ESP32 FTP.zip_ [here](http://www.io-engineering.com/downloads).

There is one snag likely due to the fact that I run the ESP32 "naked" (not inserted in a breakout board) - it crashes in a loop when I connect the SD card. I guess this is due to some missing pull-up resistors... When I run it when compiled for and inserted in my CNC Boosterpack board this does not happen. 

Here is the output to a terminal I get when I connect and run a short WebUI session, after I connected I issued a `$I` command from the terminal.

```
ets Jun  8 2016 00:22:57

rst:0x1 (POWERON_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:2
load:0x3fff0030,len:4
load:0x3fff0034,len:1380
load:0x40078000,len:11140
ho 0 tail 12 room 4
load:0x40080400,len:3864
entry 0x40080658

GrblHAL 1.1f ['$' or '$HELP' for help]
[MSG:WIFI AP READY]
[MSG:WIFI AP CONNECTED]
[VER:1.1f.20211125:]
[OPT:VNMSL,35,1024,3,0]
[NEWOPT:ENUMS,RT+,TC,SED,WIFI,FTP,SD]
[FIRMWARE:grblHAL]
[NVS STORAGE:*FLASH]
[DRIVER:ESP32]
[DRIVER VERSION:211124]
[BOARD:BDRING v4]
[WIFI MAC:3C:71:BF:4C:98:78]
[IP:192.168.5.1]
[PLUGIN:SDCARD v1.03]
[PLUGIN:ESP32 WebUI v0.03]
ok
[MSG:WEBSOCKET STREAM ACTIVE]
<Idle|MPos:0.000,0.000,0.000|Bf:35,1023|FS:0,0|Pn:PXYZ|Ov:100,100,100>
<Idle|MPos:0.000,0.000,0.000|Bf:35,1023|FS:0,0|Pn:PXYZ>
<Idle|MPos:0.000,0.000,0.000|Bf:35,1023|FS:0,0|Pn:PXYZ>
<Idle|MPos:0.000,0.000,0.000|Bf:35,1023|FS:0,0|Pn:PXYZ>
<Idle|MPos:0.000,0.000,0.000|Bf:35,1023|FS:0,0|Pn:PXYZ>
<Idle|MPos:0.000,0.000,0.000|Bf:35,1023|FS:0,0|Pn:PXYZ>
<Idle|MPos:0.000,0.000,0.000|Bf:35,1023|FS:0,0|Pn:PXYZ>
[MSG:SERIAL STREAM ACTIVE]
```

troth530, 2021-11-26:
@terjeio your bin file works great, I used ESP download tool to program.  For now, I am running my development boards naked as well.

With my bin file no Wifi, no SSID broadcast.  I also tried fresh install on another computer using **esp-idf-tools-setup-offline-2.12.exe**  Still no Wifi when I compile.

[image]
.

terjeio, 2021-11-27:
@troth530 What is your IDF version?  You can find it in _components\esp_common\include\esp_idf_version.h_

I am using 4.2 and compiling with VSCode.

troth530, 2021-11-27:
@terjeio ESP IDF Version is 4.3.1.  I did some looking around in WebUI with your build (the bin file you uploaded) and the machine does not respond to jog commands, commands at the User Interface or read back grbl settings. It does however show information in the ESP Status window as you've shown.

Do you have instructions how to downgrade IDF version and install VSCode?

Thank you.

terjeio, 2021-11-27:
@troth530 Please wait a bit - I have rewritten the event handling to not use deprecated calls, I want you to try that first. I will complete that tomorrow.

What is the status shown in the WebUI? Is it "IDLE"? If in "ALARM" state this could be the reason there is no reaction to jog commands. Check this [wiki page](https://github.com/grblHAL/core/wiki/First-Run-Grbl-Settings) for more.

Here is the console output when I execute a jog:

[image]

What do you get?

troth530, 2021-11-27:
@terjeio , I get no response in WebUI, I can see the $J input commands but no response from the ESP32.
IOSender and Lightburn jogging works as expected, so it's not coming up in alarm state .
Could be related to default Windows 10 firewall settings, I'll look into some SSH settings.
Thank you!

terjeio, 2021-11-27:
Port 81 for the websocket is blocked? Try connecting with ioSender to that.

terjeio, 2021-11-28:
@troth530 In IDF versions after 4.2 Espressif have changed the returned wifi mode in a call that leads to AP mode not beeing set up:

https://github.com/grblHAL/ESP32/blob/52f2d8366c03f476101bcdcb11dc467b07a14d65/main/wifi.c#L390-L416

The test a line 416 fails due to this. Add `currentMode = WIFI_MODE_NULL;` before line 412 to fix it.I'll add it in the next commit.

troth530, 2021-11-29:
@terjio
This addition before line 412 (currentMode = WIFI_MODE_NULL; to wifi.c) in fact does turn on WiFi finally!

I still cannot compile with FtpDaemon=ON here is the error, same as before may be an IDF compiler error. 
> 
> ../main/networking/ftpd.c:419:37: error: 'vfs_stat_t' {aka 'struct <anonymous>'} has no member named 'st_mtim'; did you mean 'st_mtime'?
>                  s_time = gmtime(&st.st_mtime);
>                                      ^~~~~~~~
> ninja: build stopped: subcommand failed.

Bluetooth=ON makes the device constantly reboot

Connecting to HTTP 192.168.5.1:80 still has same behavior and jogging does not work in WebUI
I can connect to IOSender using websocket 81 with my bin file, I previously could not connect to IOSender with your bin file using websocket 81.
Here is what I see in WebUI, looks like a configuration file is missing or I need to login as Admin?

[image]

Thank you!

terjeio, 2021-11-29:
> I still cannot compile with FtpDaemon=ON here is the error, same as before may be an IDF compiler error.

See issue #8 - the latest commit has a workaround for this IDF/compiler issue.

> Bluetooth=ON makes the device constantly reboot

I'll try this again - enabling Bluetooth on a "naked" board may trigger an ESP32 issue that has not been resolved. It has been reported by several people, including myself.

> Here is what I see in WebUI, looks like a configuration file is missing or I need to login as Admin?

No, it is due to the websocket connection not established/not working.

I you have a terminal window open when you start the WebUI do you get this message:
`[MSG:WEBSOCKET STREAM ACTIVE]`
followed by status reports?

Which browser are you using? Is [compatible](https://github.com/luc-github/ESP3D-WEBUI/wiki/Browser-compatibility-for-ESP3D) with the WebUI?

troth530, 2021-11-30:
@terjeio , Your latest updates have fixed the FTP issue.  I saw how you fixed it and "Silly" is not a harsh enough word.

yes I see this message in a terminal window.  [MSG:WEBSOCKET STREAM ACTIVE]
I have tried Win7 and Win 10 IE11, IE12, Chrome.  
It is a winsocket connection error.  I turned on some debugging in Chrome.

Thank you!

[image]

terjeio, 2021-11-30:
@troth530 The websocket issue is on me - a bad initial handshake with the browser, still accepted by Firefox which I use.
Can you replace _networking/WsStream.c_ with this version and try again:

[WsStream.zip](https://github.com/grblHAL/ESP32/files/7623636/WsStream.zip)

I have verified this with Edge on Win10, the previous version did not work with that.

FYI I have seen some random crashes while running the WebUI (the dreaded meditating guru wakes up and providing no stack trace). I will start running long time tests again to try to isolate this. Luckily I now have the WebUI running on a STM32F756 that is debuggable, this may help in tracking down the reason...

terjeio, 2021-11-30:
More tuning:
[WsStream.zip](https://github.com/grblHAL/ESP32/files/7627506/WsStream.zip)

I have found one "bug" that causes a crash, if connecting a second WebUI while there is motion going on from the first - boom.
There might be other scenarioes that triggers it as well, luckily there is an easy fix for it, [this line](https://github.com/grblHAL/core/blob/f88b154b9603f84709c4092b0ec2868a1b52a1a5/state_machine.c#L148) has to be changed to:
`ISR_CODE sys_state_t state_get (void)`

Reason is that when attempting to load the second WebUI code from flash `state_get()` is called from an interrupt context and since this is residing in flash as well, likely in a page that has to be loaded into RAM, a conflict arises. The guru then wakes up... The fix moves the `state_get` function permanently into RAM.
I'll have to check the code again to see if there has been any other changes that may trigger this as well.

troth530, 2021-11-30:
@terjeio   **Guru Scores!**
We appreciate all the hard work that goes into making this code work for so many devices.

Thank you!

troth530, 2021-12-02:
@terjeio I need to reset everything and try a second PC.  Recently anything with WIFI on will compile but I get CPU halted error when trying to connect through serial and of course it will not broadcast SSID.  I had manually made changes to wifi.c and WsStream.c and updated only the networking directory.  Some additional changes you may have made to wifi.c are possibly the source of my latest problems.
Thank you.

terjeio, 2021-12-02:
That is odd. What do you get in the terminal?
Have you tried with Soft AP off as well?

I have replaced deprecated code in the IDF with the new, and also fixed some issues with Access point/Station mode. Yesterday I terminated a 24+ hour test that went without problems, and when I now flash a fresh build today it still works.

Perhaps one thing to try is to erase the whole flash as there might be some corrupted settings creating problems?

troth530, 2021-12-03:
@terjeio I finally got it working, erasing full flash was one of many things I had tried.
Can you verify that when using chrome with WebUI  that if you close the page with WebUI it will not crash the controller.
I found that either closing the page it or disconnecting from the AP it crashes, but my build is a little out of date.
Thank you!

terjeio, 2021-12-03:
> I finally got it working, erasing full flash was one of many things I had tried.

Great, I guess I should swicth to a 16 bit CRC for the settings checksums - I have prepared the codebase for that. This as the current 8 bit version is not very good, and too boot the legacy version of that was broken.

> Can you verify that when using chrome with WebUI that if you close the page with WebUI it will not crash the controller.

I do not have Chrome on my machine so cannot test with that.

> I found that either closing the page it or disconnecting from the AP it crashes, but my build is a little out of date.

The latest version behaves as it should for me, I made some changes to how connect/disconnect is handled in that. Here is output on the terminal for a short WebUI session in AP mode where I disconnected from the AP:

```
GrblHAL 1.1f ['$' or '$HELP' for help]
[MSG:WIFI AP READY]
[MSG:WIFI AP CONNECTED]
[MSG:WEBSOCKET STREAM ACTIVE]
<Idle|MPos:0.000,0.000,0.000|Bf:35,1023|FS:0,0|Ov:100,100,100>
<Idle|MPos:0.000,0.000,0.000|Bf:35,1023|FS:0,0>
<Idle|MPos:0.000,0.000,0.000|Bf:35,1023|FS:0,0>
[MSG:SERIAL STREAM ACTIVE]
[MSG:WIFI AP DISCONNECTED]
<Idle|MPos:0.000,0.000,0.000|Bf:35,1023|FS:0,0>
```

I then get _Connection error_ in the WebUI console. What do you get?

Note that the last real-time report is from me requesting it from the terminal, just to check that it is back in control.

terjeio, 2021-12-03:
@troth530 I found a new "bug" - if a disconnect/reconnect happens while a SD card job is running I get a crash due to a stack overflow. This is from the SD card plugin and can be fixed, but how?
My intuition is to terminate the job when a client disconnects as nothing is in control of the machine - I have tested that and it works.

Leaving the job running and allowing a client to connect again will likely be hard as grblHAL does not allow input from more than one stream at a time.

terjeio, 2021-12-03:
@troth530 I have spent the day working on a fix for the WebUI so that it can be restarted without terminating the current job.
However, if another sender connects after the WebUI has been disconnected a running job will be promptly terminated when the first non real-time command character is received, this to allow the sender to query settings etc.
For this I have an issue to resolve, the character that termintates the job is currently lost and will cause an error for the first command sent. I will commit an update when this is solved.

troth530, 2021-12-03:
> I found a new "bug" - if a disconnect/reconnect happens while a SD card job is running I get a crash due to a stack overflow. This is from the SD card plugin and can be fixed, but how?
> My intuition [...]
@terjeio , one feature I like about the SD card is you can start a job, mount card, start from SD card then disconnect and the job will continue autonomously from the controller completely offline.  I have this working properly on Teensy.  So please consider job coming from SD card continuing after disconnect as a valid usage.

> Leaving the job running and allowing a client to connect again will likely be hard as grblHAL does not allow input from more than one stream at a time.
Makes sense, I think in general most controls like jogging are locked out when a job is running, so it seems reasonable that new WebUI session or even new WIFI AP client will be locked out when job is running.

I look forward to what you come up with to resolve.
As always, Thank You!

troth530, 2021-12-04:
@terjeio, after a fresh rebuild and clone from your repo on a new machine I have no more issues compiling or flashing. 

As far as browsers with the latest code, I have mixed results.  
IE10 on Win10 will not connect (similar to the problem before we saw with Chrome that you resolved as a winsock handshake)
Chrome on Win10 will connect, but closing the window crashes the ESP32 most of the time, not all.
Firefox on Win10 will connect and closing the window returns control to the serial stream.  Disconnecting from the AP with the winsock connection in both open and closed shows WIFI AP DISCONNECTED (as expected) and does not appear to be crashing the ESP32.  So I'd say Firefox is very promising, but I'll continue testing.

Before the ESP32 would crash just by disconnecting from the AP, but that may have been on older code and I am only using the latest code for testing now.  So I will continue to verify behavior on Firefox.  It's been a while since I used Firefox regularly - my old friend.

The best thing about ESP32 is WIFI and the WebUI, so please don't get discouraged!
Thank you!

terjeio, 2021-12-04:
IE10 is not in the [compatibility list](https://github.com/luc-github/ESP3D-WEBUI/wiki/Browser-compatibility-for-ESP3D) so I guess you can forget about that, I have IE11 on my laptop and that shows a garbled UI as expected but does not crash the controller. Edge seems to be ok. I guess I have to download Chrome again to check what it is doing even if I do not like Google much...

> The best thing about ESP32 is WIFI and the WebUI, so please don't get discouraged!

Well, the ESP32 is a pain to work with - no debugger and seemingly very flaky (easy to crash). The iMXRT1062 and STM32F756 (cabled ethernet) versions feels a lot more robust, a 12 hr job running on the iMXRT1062 has survived numerous restarts and reloads - and even my computer going to sleep overnight.

terjeio, 2021-12-04:
New commit up now.

I believe I have solved WebUI disconnect, another client connecting issue while job is running. When the new client connects and send a non realtime command the current job will be put into hold then terminated.

I have not installed Chrome and checked why it crashes the ESP32, I'll wait for your feedback first.

troth530, 2021-12-05:
@terjeio now it seems Firefox and Edge are working, but Edge has the same issue as Chrome - when the window with the winsock connection is closed it will crash the controller.  I think you'll see the same result with IE11 as with IE10 which is no connect.  Firefox works perfectly for all needs.

I'm not testing with full hardware yet, but wouldn't it be preferred that the running job is not terminated when a new connection is made and that there is a WebUI button to allow the new client the option to stop the job instead of it automatically being stopped.

I have another issue it looks like SDCard is not working.  I am using the Adafruit SPI 3V breakout board connecting the SPI signals but even IOSender does not see the same SD card which I had working with Teensy.  Is there a different file system needed.  I will say that there are 47KOhm pullups on all of the SPI pins (not 4.7k as the document suggests).  Here are the details:
https://learn.adafruit.com/adafruit-microsd-spi-sdio/pinouts

Also FTP does not work when connected to the AP.  I use these commands
      C:\Windows\System32>ftp
      ftp> OPen
      To 192.168.5.1
       ftp: connect :Connection refused

The goal is to connect through AP and use WebUI or FTP to send a file to SD card, then run the job and disconnect.  Then possibly reconnect later to monitor the job.

Thank you!

terjeio, 2021-12-05:
> ... but Edge has the same issue as Chrome - when the window with the winsock connection is closed it will crash the controller.

I just installed Edge on my laptop, closing the window will crash the network stack, not the controller. I can still get a response for real time report request (typing `?`) in the terminal window. My hunch is that this might be due to Chrome and Edge trying to send telemetry back home. FYI my STM32F756 controller works with Edge so it is ESP32 related.

> I'm not testing with full hardware yet, but wouldn't it be preferred that the running job is not terminated when a new connection is made and that there is a WebUI button to allow the new client the option to stop the job instead of it automatically being stopped.

IMO it has to be stopped when a different client connects since it cannot communicate with the controller if not. I may add a setting for that if required. There is a button to stop a running job in the WebUI:

[image]

Best is to put the job into a feed hold before stopping, if not the controller is reset.

There is no way for a new client to ask the WebUI client to stop the job since it is not connected when the takeover happens. Again, a running job started by the WebUI is terminated when a new client connectes _after_ the WebUI client has disconnected. If a WebUI client reconnects it is not terminated. If a new client connects while the WebUI is connected it may receive the real time reports depending on the connection type. 

> I have another issue it looks like SDCard is not working.

It works for me and I do not have any pullup resistors at all. I am using the CNC Boosterpack map.

> Also FTP does not work when connected to the AP. I use these commands

Again, this works for me, I have been using WinSCP for testing - and I just verified with command line ftp.
Is ftp enabled in the $70 setting ($70=15)? Note that ftp requires the SD card plugin and IIRC a card mounted.

terjeio, 2021-12-05:
Ok, found the network stack "crash" - a call to tcp_close() in the stack never returned. I have reoganized the code and now it seems to work with ESP32 too. The hang happens when the client sends an empty packet to close the connection, not the websocket close command as expected. I guess it hangs with ESP32 due to grblHAL running as a FreeRTOS task, the other platforms supported runs "bare metal".

Can you try on your end with this version of WsSTream.c?

[WsStream.zip](https://github.com/grblHAL/ESP32/files/7655612/WsStream.zip)

troth530, 2021-12-05:
@terjeio that fixes the issue with Chrome and Edge.  Who cares about IE.  So Chrome, Edge, Firefox all seem to be working as expected for WebUI  Great work!.

However, I cannot test if a job continues if reconnecting a websock client that started a running job due to being unable to get SD Card working.  I pulled all the pullup resistors so the IO is directly connected and using this wiring, I'm assuming 19=GPIO_19 and so on.  The SD Card is formatted as Fat32.  
#define PIN_NUM_MISO 19
#define PIN_NUM_MOSI 23
#define PIN_NUM_CLK  18  
#define PIN_NUM_CS   5

Thank you.

terjeio, 2021-12-05:
>  I'm assuming 19=GPIO_19 and so on. 

Yes, and it is the same configuration I have for my board.

> The SD Card is formatted as Fat32.

Have you tried with more than one card? BTW mine are FAT formatted and <= 2GB. I just reformatted one to FAT32 and that works as well.

troth530, 2021-12-06:
@terjeio SD card mounted and works through WebUI.   

Testing of FTP- when WebUI is closed and SD is mounted, ftp connects but won't put or get files, $70=15 (of course).  In any case I see that FTP is not needed for WebUI to be able to transfer files.  Also SD card streaming is available in IOSender, but upload button is greyed out so the SD card can be read, but not written to.  Another hint possibly is that with FTP enabled- unplugging the USB from then the serial port is connected will cause EEPROM read error 7 (with reset to defaults) upon plugging USB back in.  Overall it could be that a lot of the Guru meditation errors I've been having not related to the other issues you've fixed are FTP related as I was trying to get SDcard working believing FTP was needed to move files to SD.

More testing with prototype hardware in upcoming week.
Thank you!

terjeio, 2021-12-07:
> Testing of FTP- when WebUI is closed and SD is mounted, ftp connects but won't put or get files, $70=15 (of course). 

Strange, it is working for me but is a little bit unstable. This could be due to the SD card clock frequency beeing a bit too high - the signal does not look good on my digital scope but it is only 100 MSPS so it could be due to that. The SD card mount code has to be rewritten to allow reducing it since the code provided by the IDF does not. If http upload does not fail there may be some problem with how my ftp code interacts with the IDF provided network stack as I do not see the same issue with e.g. the iMXRT062 driver. BTW iMXRT062 driver achieves way faster transfer rates - with exactly the same ftp code...

An idea: perhaps moving the SD card to the [HSPI pins](https://docs.espressif.com/projects/esp-idf/en/latest/esp32/api-reference/peripherals/sdio_slave.html#connections) will result in better performance?

> Also SD card streaming is available in IOSender, but upload button is greyed out so the SD card can be read, but not written to.

SD Card upload via FTP is only available in the [edge version](http://www.io-engineering.com/downloads). The current release only supports YModem. Which version do you have? Note that the FTP upload will fail in AP mode if the sender is not running on the same computer as the one that "owns" the connection.

> Another hint possibly is that with FTP enabled- unplugging the USB from then the serial port is connected will cause EEPROM read error 7 (with reset to defaults) upon plugging USB back in. 

Is [EEPROM support](https://github.com/grblHAL/ESP32/blob/31e078131b76f9fc4069be2799f25a05beb9af29/main/CMakeLists.txt#L39) enabled in CMakeLists.txt? If so do you have one connected? Only 2K EEPROMs are supported for now.
I do not see any error 7s when I am testing.

I have "hardened" code for handling removing/inserting the SD card. Currently you have to issue  a `$FU` command to unmount it before removing it and `$FM` after inserting again. Failure to do so may wake up the guru. I have found that the code I use for mounting/unmounting was added to the IDF so that Espressif could write [a simpler example](https://docs.espressif.com/projects/esp-idf/en/latest/esp32/api-reference/storage/fatfs.html#using-fatfs-with-vfs-and-sd-cards). Foilware... I may opt to write a lowlevel FatFs driver and bypass the IDF instead.

I have committed an update with the changes.

troth530, 2021-12-11:
@terjeio 
I had perfect results with SD Card iMXRT062 with FTP to SD Card.  Also spooling gCode was almost as fast as Ethernet (and USB).  

Noted on the need to unmount before removing SD card.

EEPROM is not enabled, seems a normal grbl error 7 when reading the settings on startup.  But as I said this error did not remain when I turned off FTP.

I do have a problem you may be able to double check on your end.  gCode spooling is very, very, very slow. 
I have fine tuned STM Blue Pill which runs this gCode in 31-32 seconds, STM Black pill which runs this gCode in 24 seconds and Teensy 4,1 which runs this code in 22 seconds.   On the ESP32 it takes 11minutes, 30 seconds.  I have not tuned the buffers (BLOCK_BUFFER_SIZE,  SEGMENT_BUFFER_SIZE, RX, TX) but I was thinking you may be able to compare with your STM F4.  Please use $32=1, $100 and $101 80 steps/mm, $110 and $111 as 30000mm/min and $120 and $121 as 10000mm/s/s.  I call this the unconstrained test to see if the firmware and controller will spool fast enough so the steppers become the bottleneck in performance.  Right now greyscale is limited to somewhere around 10,000mm/min (166mm/sec) due to the gCode spooling speed slow,
[greyscale_raster_test_500.zip](https://github.com/grblHAL/ESP32/files/7696279/greyscale_raster_test_500.zip)

Thank You!

terjeio, 2021-12-11:
> EEPROM is not enabled, seems a normal grbl error 7 when reading the settings on startup. But as I said this error did not remain when I turned off FTP.

Does this happen every time on reset or only after reflashing? Are all settings lost? There are separate checksums for different blocks of settings, I wonder which fails...
FYI: I am compiling/flashing with ESP-IDF 4.3 and I do not see this, could it be due to the [grbl partition](https://github.com/grblHAL/ESP32/blob/master/partitions.csv) used for settings not beeing properly placed due to differences in flash size?.

> I do have a problem you may be able to double check on your end. gCode spooling is very, very, very slow.

Is your ESP32 broken?
I have run the same test, albeit with acceleration settings set to a more realistic 1000m/s/s.
Check mode: 0:06 from SD card, 0:49 via USB/UART (544Kb/115Kbits/s -> 0:47), 1:43 via telnet, 2:23 via websocket
Run, planner buffer = 36: 1:27 from SD card, 1:28 via USB/UART, 2:02 via telnet, 2:33 via websocket
Run, planner buffer = 400: 0:55 from SD card, N/A<sup>*</sup> via USB/UART, 1:50 via telnet, 2:27 via websocket

<sup>*</sup> The infamous Guru randomly wakes up and halts the controller even when networking is disabled.

I have run quick tests with the MSP432E401Y which also uses FreeRTOS, network transfer speeds are slightly slower than ESP32. This makes wonder if it is the RTOS that is slowing down the rates compared to drivers that runs on "bare metal".

I have made a simple MCU "load" plugin that counts how many times the protocol loop is run in 10 ms, the iMXRT1062 driver manages around 10000 times when running this test, 20000 when idle. Perhaps I should characterize the other drivers in the same way?

terjeio, 2021-12-11:
> Run, planner buffer = 400: 0:55 from SD card, N/A* via USB/UART, 1:50 via telnet, 2:27 via websocket

Via USB/UART is 1:26 after I managed to leave the Guru meditating in peace. I had to add an ESP32 specific workaround in the core (and driver) for that. It is related to either the processor or the IDF (randomly) not handling floating point data in an ISR context - [even if not referencing it](https://github.com/grblHAL/ESP32/blob/31e078131b76f9fc4069be2799f25a05beb9af29/main/driver.c#L880).

Also, I lost the settings once after reflashing numerous times today - I have no idea why...

troth530, 2021-12-11:
@terjeio  same result with planner buffer=400
I'm using this WROVER device with CH340.  
[https://www.amazon.com/gp/product/B07RXTSZRC](https://www.amazon.com/gp/product/B07RXTSZRC
)
I have the same results with this WROOM device with Silicon labs interface.
[https://www.aliexpress.com/item/32992390199.html](url)

I don't suspect damage I am using your code directly with IDF 4.3 as per compilation instructions.  In cases of both devices the baud rate is set 115200, I have 2 of each device and have only gotten similar results.  
I use this command to erase 
     idf.py -p COM8 erase_flash
When switching between devices I use idf.py clean to redo the partition table and to flash
  idf.py -p COM8 flash

[VER:1.1f.20211203:]
[OPT:VNMSL,399,1024,3,0]
Target buffer size found
[NEWOPT:ENUMS,RT+,TC,SED,SD]
[FIRMWARE:grblHAL]
[NVS STORAGE:*FLASH]
[DRIVER:ESP32]
[DRIVER VERSION:211203]
[DRIVER OPTIONS:v4.3.1-dirty]
[BOARD:BDRING v4]
[PLUGIN:SDCARD v1.04]
ok
Starting stream
[MSG:Pgm End]
Stream completed in 11:28

Willing to buy another device to test, what do you recommend?
Thank you!

terjeio, 2021-12-11:
> idf.py -p COM8 erase_flash

That clears the settings too, I do not do that - I just flash.

> Stream completed in 11:28

Is this with the default settings (after they are reset) or after you have reconfigured?
How long does it take to complete the job in Check mode? Running from the SD card?

> Willing to buy another device to test, what do you recommend?

I do not think you need one. If check mode takes a long time too IMO then there must be something wrong with your USB connection.

troth530, 2021-12-12:
@terjeio interesting.  Slowing the speeds and accel way down seems to keep the Guru from showing himself, but does it make sense that the ESP32 can only spool gCode at half the speed of STM32?  I am now getting around 1:11 via USB/UART and I observed better using websocket around 1:00.  This is now in a practical range of operation for some machines but not all.  Basically limited to 200mm/sec, with accel approximately 2000mm/s/s.  I did observe that it was sensitive to acceleration, so I changed the M4 to M3 to see if it helped and it did, but that may imply a processing issue, not a communications issue.

terjeio, 2021-12-12:
> ... but does it make sense that the ESP32 can only spool gCode at half the speed of STM32?

Over a USB connection yes it does, ESP32 has an USB <> UART  chip in between the USB port and the MCU so the configured baud rate is real, with the USB stack in the MCU there is no such bottleneck and the baud rate setting usually has no meaning. I have seen transfer rates > 150Kbytes/s with iMXRT1062, ESP32 is limited to 115Kbits/s (which is ~11Kbytes/s) as it is configured now. The baud rate setting can be increased for ESP32 here if you want to try that:

https://github.com/grblHAL/ESP32/blob/31e078131b76f9fc4069be2799f25a05beb9af29/main/driver.c#L1681

>  I did observe that it was sensitive to acceleration, so I changed the M4 to M3 to see if it helped and it did, but that may imply a processing issue, not a communications issue.

Hopefully the workaround I made will fix that - I will commit that later as it required changes to the core.

troth530, 2021-12-13:
@terjeio 
> Over a USB connection yes it does, ESP32 has an USB <> UART chip in between 
Makes perfect sense

Unfortunately, increasing Baud shows no benefit tried 460800 and no benefit.  

>Hopefully the workaround I made will fix that - I will commit that later as it required changes to the core.

Looking forward to that.

For the control pins- reset_pin, feed_hold_pin and cycle_start_pin, are external pullups required?  For some reason reset pin does not perform control reset when pulled low (so I used the reset pin on the ESP32) but it's behavior is a full controller restart.  feed_hold works, but then cycle_start does not restart execution of the gCode.  To complicate matters I should mention that on the WROVER device I am using, GPIO36 and 39 are senseVP and senseVN.

Thank you!

terjeio, 2021-12-13:
> Unfortunately, increasing Baud shows no benefit tried 460800 and no benefit.

I halved the time to transfer data in Check mode when I went to 230400, to 460800 showed no improvement over 230400.

The workaround is now committet alongside a lot of other changes that I hope does not break anything. These are a using new core based stream switcher (from UART to Websocket to SD Card etc. and back), support for ganged/auto squared axes, Trinamic stepper driver support++

> For the control pins- reset_pin, feed_hold_pin and cycle_start_pin, are external pullups required? 

IMO yes. Relying on the weak internal pullups is inviting problems. More current through the switches are generally better, at least above the wetting current. A capacitor across the switch may help increasing the wetting current too. I see some using optocouplers adding the capacitor on the transistor side of the optocoupler for noise filtering, better to have it accross the switch as most optocouplers act as a low-pass filter anyway.

>  For some reason reset pin does not perform control reset when pulled low (so I used the reset pin on the ESP32) but it's behavior is a full controller restart. feed_hold works, but then cycle_start does not restart execution of the gCode.

Does the LED indicators change in ioSender depending on the switch state?
Also, there is some interlocking betwteen the switches, e.g. if feed hold is asserted cycle start will not work.

> To complicate matters I should mention that on the WROVER device I am using, GPIO36 and 39 are senseVP and senseVN.

I'll have to check this as I do not know what senseVP and senseVN is for.

[23 more replies]

closed 2023-04-16 by terjeio