grblHAL/ESP32 #11: How to build using PlatformIO

MeKeCNC, 2021-11-22:
If anyone can compile esp32 with PlatformIO, I would appreciate it if you share project file.

drom89, 2022-09-29:
Any new here ? :) 
Im trying it for 2 days with VScode ESP-IDF and VScode and PlatformIO and just can't get it to work.

terjeio, 2022-09-30:
@drom89 I have given up on VSCode and use Eclipse, the IDF command line also work.

Are VSCode using the 4.3 version of the IDF? It is not uncommon for Espressif to introduce breaking changes in new versions...

And yes, building with platformio would be nice - I guess I have to figure out how to set up a configuration file for it if nobody else can do it.

drom89, 2022-09-30:
The plugin version is v1.5.0
https://marketplace.visualstudio.com/items?itemName=espressif.esp-idf-extension

In the terminal i can see .espressif\python_env\idf4.4_py3.8_env\Scripts\python.exe, so i thing it is using version 4.4

terjeio, 2023-01-29:
The [Web Builder](http:/svn.io-engineering.com:8080/) uses PlatformIO to build the firmware, [using a template .ini file](https://github.com/grblHAL/ESP32/blob/master/platformio.tpl) which is filled in with options from the builder.
The options sent to the backend can be viewed in the browser console, open it with F12, this could be used in combination with the template to create a standalone .ini-file?

closed 2023-01-29 by terjeio