grblHAL/ESP32 #13: Ganging using 4 drivers

timm052, 2021-12-11:
How would i go about configuring a XYYZ set up using an individual driver for each motor. I'm trying to create a board board profile for the CNC xPro V5 based on this GRBL-ESP32 file. https://github.com/Spark-Concepts/xPro-V5-Firmware

terjeio, 2021-12-13:
I now have added a board profile and tentative support for both ganged/auto squared axes and TMC5160 stepper drivers. Since I do not have the hardware to verify these changes with I'll have to ask you to do that.
Build configuration is in [CMakeLists.txt](https://github.com/grblHAL/ESP32/blob/master/main/CMakeLists.txt).

Note that I have not added support for enabling SD Card and Trinamic drivers at the same time, try with Trinamic drivers first.

timm052, 2021-12-14:
Thank so much i'll get this tested in the next few days

timm052, 2021-12-16:
Got it compiled and flashed everything works except i cannot get the steppers to move. They seem to be recognized and i was able to change the amount of current going to them but I cannot get them to move at all. I've messed around with inverting the direction pins but that doesn't seem to be having an effect. Although i found if i enabled a pull-up on the reset button the steppers would move very slightly when i pressed the reset button. I could be missing some super obvious setting though.

terjeio, 2021-12-16:
Try $4=7

timm052, 2021-12-16:
That appears to have done it. What setting does that change?

terjeio, 2021-12-16:
Motor enable signals - opposite of legacy Grbl where it is motor disable.
Send `$$=4` for info.

timm052, 2021-12-16:
Okay so i it seems to all be working except the motor ganging Y2 will only move in one direction no matter which button is pressed it always moves the same way. While Y1 behaves normally

terjeio, 2021-12-16:
Is the pin definition for the Y2 motor direction correct?
`#define M3_DIRECTION_PIN    GPIO_NUM_32`

timm052, 2021-12-16:
Yep just double checked against the default firmware and the grbl hal pins and they're all correct.

terjeio, 2021-12-16:
Setting direction is done here:

https://github.com/grblHAL/ESP32/blob/0f6429343f25aae3382991c870e2f80c2f531b74/main/driver.c#L761-L788

Can you copy line 782 to below line 766 and try again? If not working then the pin is likely not set as output at startup as it should.

timm052, 2021-12-17:
Yeah no luck on copying the 786 to 766 gonna have a shot at switching the driver pins see if the issues move to the other driver

timm052, 2021-12-18:
Went back to GRBL ESP32 and the direction pin is working there so i assume there is some weird software thing going on somewhere.

terjeio, 2021-12-18:
The pin is set as input due to an incorrectly sized variable, replace [these lines](https://github.com/grblHAL/ESP32/blob/0f6429343f25aae3382991c870e2f80c2f531b74/main/driver.c#L1686-L1689) with:

```
    for(idx = 0; idx < (N_AXIS + N_ABC_MOTORS); idx++)
        rmt_set_source_clk(idx, RMT_BASECLK_APB);

    uint64_t mask = 0;
```

I guess I should make a "motherboard" for my [Nucleo-64 BOB](https://github.com/terjeio/CNC_Breakout_Nucleo64) for testing...

timm052, 2021-12-19:
Works thanks so much. The Y2/A driver, however, seems to not like resetting after a reset the other drivers all work but the Y2 fails to come back up and needs to reset the controller too. Not a huge issue but i thought I'd mention it. Also is there any way to disable stealth chop which seems to be on by default on ZY1Y2 but not X

terjeio, 2021-12-19:
>  the Y2 fails to come back up and needs to reset the controller too.

It does not turn anymore?

>  Also is there any way to disable stealth chop ...

Yes, basic configuration is in [trinamic.h](https://github.com/grblHAL/Plugins_motor/blob/8c4afcc879a79fe98668b91f868ed47c7d64bfc2/trinamic.h#L58-L71). 

>  ... which seems to be on by default on ZY1Y2 but not X

Did you see that in the _Trinamic tuner_ tab in ioSender when querying the driver status? If so it disables StealthChop for the selected axis and reenables it when leaving the tab.

timm052, 2021-12-19:
> It does not turn anymore?
It fails to respond to commands after a reset either manual or caused by an alarm and requires a full power cycle to bring it back online

terjeio, 2021-12-19:
Ok, what happens if you send `M122I` to reinitialize the drivers?

timm052, 2021-12-19:
> Yes, basic configuration is in [trinamic.h](https://github.com/grblHAL/Plugins_motor/blob/8c4afcc879a79fe98668b91f868ed47c7d64bfc2/trinamic.h#L58-L71).

Got it disabled but i had to go into the tmc5160.h file to do it

timm052, 2021-12-19:
> Ok, what happens if you send `M122I` to reinitialize the drivers?

hmmm that brings them back but they have extreme vibrations and run terribly until i do a full reset

terjeio, 2021-12-19:
I have fixed a number of bugs in the Trinamic code and just committed an update, hopefully this will fix your problems.

timm052, 2021-12-20:
That seems to have resolved all outstanding issues. Thanks so much for the help is there anywhere i can donate to support this project?

closed 2021-12-20 by timm052