grblHAL/ESP32 #15: CNC XPro V5 0-10v PWM Converter Issue

timm052, 2021-12-20:
Sorry, one last issue The CNC Xpro has a built-in PWM Converter that outputs a 0-10v signal which is use to drive my kmb spindle there's also a relay that can be linked to the enable pin. I cannot seem to get this working correctly and it seems very intermittent the relay often gets stuck on and the pwm settings disappear from the config screen. I have managed to get the spindle to turn on but it on goes to max speed and gets stuck and stops me from disabling it again.

The approx description of how it works can be found here
https://github.com/Spark-Concepts/xPro-V5/wiki/Front_Panel

timm052, 2021-12-20:
Dw trouble extended trouble shooting has gotten to my head i had it turned off at the power........

timm052, 2021-12-20:
I can get it working but it seems very inconsistent sometimes just coming on at max speed and sometimes pulsing the speed up and down but mostly not enabling at all. The relay seems to be fairly consistent at working but sometimes it gets stuck on as well.

terjeio, 2021-12-20:
>  I cannot seem to get this working correctly and it seems very intermittent the relay often gets stuck on ...

For me it looks like the relay is controlled by a switch, not the MCU.

> ... and the pwm settings disappear from the config screen.

With ioSender? If version 2.0.36p21 please update to 2.0.36p22.

> I can get it working but it seems very inconsistent sometimes just coming on at max speed and sometimes pulsing the speed up and down but mostly not enabling at all. 

When running gcode or when idle? Both? If when running gcode ensure `$32=0` (laser mode is off).
The enable signal and the PWM output are both stable when I check with my scope, also when running gcode.

> The relay seems to be fairly consistent at working but sometimes it gets stuck on as well.

This is odd, if controlled by the MCU as well as the switch I would like to se a real schematic...

timm052, 2021-12-20:
> For me it looks like the relay is controlled by a switch, not the MCU.

It automatically triggers when the spindle enable pin is activated

> With ioSender? If version 2.0.36p21 please update to 2.0.36p22.

Update seems to have corrected things

> When running gcode or when idle? Both? If when running gcode ensure `$32=0` (laser mode is off). The enable signal and the PWM output are both stable when I check with my scope, also when running code.

This is just turning on the spindle via M3 with a set speed

> This is odd, if controlled by the MCU as well as the switch I would like to se a real schematic...

Yeah that would be great it's a real shame it's not open source it's not like it's doing anything special.

terjeio, 2021-12-20:
Could it be related to [this issue](https://github.com/Spark-Concepts/xPro-V5/issues/91)?

And can you check with a voltmeter or scope on the spindle outputs? Are any of those stable?

timm052, 2021-12-20:
No it works fine on the default firmware so I don't think that's the issue. I did a bit of probing with the voltmeter and it's definitely an issue with the 0-10v as I'm getting no output on that pin when it's not working. I'll do a fresh flash tomorrow make sure it's not some wrong setting somewhere

terjeio, 2021-12-20:
I am misreading the "schematics" in the link you provided, the relay is either controlled by the coolant or spindle enable output.
What I find a bit odd is that the relay ground pin seems to be routed through the PWM output for the VFD, could this be the a hint?

[image]

Perhaps you should try inverting the spindle enable output with `$16=1`?

timm052, 2021-12-20:
I have it wired slightly differently  with 24v going through the relay due to my kmb spindle and 0-10v always connected I'll do some additional experimentation

timm052, 2021-12-21:
After 12 hours turned off it all seems to have resolved itself not sure what was going on. Thanks once again for all the help i think i can offically call it workinjg

closed 2021-12-21 by timm052