grblHAL/ESP32 #16: CMake issue

fraro, 2021-12-20:
I have built grbHAL/ESP32 successfully several times. Now I did a fresh install on another machine and always getting
Unknown CMake command "idf_component_register".
during idf.py build
cmake version 3.18.4
esp-idf is release/v4.3
running on debian/bullseye, Python 3.9.2
Any hints appreciated!

terjeio, 2021-12-21:
[Here](https://www.esp32.com/viewtopic.php?t=12084) is one from a search I did. There are [others too.](https://duckduckgo.com/?t=ffsb&q=Unknown+CMake+command+%22idf_component_register%22.&ia=web)

fraro, 2022-01-11:
Turned out to be a user error. I was used to run the build from inside the "main" directory, but first time it has to be run from the root dir.

closed 2022-01-11 by fraro