grblHAL/ESP32 #17: MKS DLC32 and MKS TinyBee Board support

Cncninja, 2022-01-17:
Hello, the MKS DLC32 is a 3 axis Grbl board using an ESP32 it currently supports/runs 1.1f  I have 2 of these boards and FluidNC has a drop-in YAML for them, but I like GrblHal and IOsender. I'm just not sure how to port the stock 1.1f or FluidNC file over to grblHal.  Also they just came out with the TinyBee witch is basically a 5 axis version of the DLC32. Any help would be appreciated.

terjeio, 2022-01-17:
You have to create a map file. Name it _my_machine_map.h_ and select that in [CMakelists.txt](https://github.com/grblHAL/ESP32/blob/cd6f050f15733a8a561b087c5ba01ac181d6ef69/main/CMakeLists.txt#L15). When verified then either post it here or create a PR for it (after giving it a new name that match the board name).

dianlight, 2022-01-23:
Hello, 
 I try to create the MKS_DLC32 map file but I have some issue on I2S port.  
What not I understand is how set an I2S pin if I use the I2SO(..) macro I'have the follow compiler error:

```
...
./grblHal/main/ioports.c:22:
./grblHal/main/ioports.c: In function 'ioports_event':
./grblHal/main/driver.h:262:25: error: implicit declaration of function 'i2s_out_state'; did you mean 'gc_get_state'? [-Werror=implicit-function-declaration]
 #define DIGITAL_IN(pin) i2s_out_state(pin)
                         ^~~~~~~~~~~~~
./grblHal/main/ioports.c:278:60: note: in expansion of macro 'DIGITAL_IN'
                 input->interrupt_callback(in_map_rev(idx), DIGITAL_IN(input->pin));
                                                            ^~~~~~~~~~
cc1: some warnings being treated as errors
[1246/1292] Building C object esp-idf/main/CMakeFiles/__idf_main.dir/grbl/gcode.c.obj
ninja: build stopped: subcommand failed.
ninja failed with exit code 1
```

terjeio, 2022-01-23:
A missing include and a function rename caused the error. Fix committed.

dianlight, 2022-01-24:
Thx now the function is find but I have another problem. I don't find information about how specify the i2s pin in a right way.
If I set pin ad natural number as in `bdring_i2s_6_axis_map.h` I can build but the firmware enter in a cycle loop error if I set in the "esp32 way" using the macro I2SO() I have another compile error because I2S_OUT_PIN_BASE is not set.

-- Loop Error using `#define X_STEP_PIN 1`:
```
rst:0x3 (SW_RESET),boot:0x1b (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:2
load:0x3fff0030,len:1400
load:0x40078000,len:12812
ho 0 tail 12 room 4
load:0x40080400,len:3020
entry 0x400805e4
ets Jun  8 2016 00:22:57
```

-- Compile Error using `#define X_STEP_PIN I2SO(1)`:
```
....
In file included from ./grblHal/main/driver.c:33:
./grblHal/main/driver.c: In function 'i2s_set_step_outputs':
./grblHal/main/i2s_out.h:56:22: error: 'I2S_OUT_PIN_BASE' undeclared (first use in this function); did you mean 'I2S_OUT_NUM_BITS'?
 #    define I2SO(n) (I2S_OUT_PIN_BASE + n)
                      ^~~~~~~~~~~~~~~~
./grblHal/main/./driver.h:265:47: note: in definition of macro 'DIGITAL_OUT'
 #define DIGITAL_OUT(pin, state) i2s_out_write(pin, state)
                                               ^~~
./grblHal/main/./mks_dlc32_v2p0_map.h:53:20: note: in expansion of macro 'I2SO'
 #define X_STEP_PIN I2SO(1)
                    ^~~~
./grblHal/main/driver.c:706:17: note: in expansion of macro 'X_STEP_PIN'
     DIGITAL_OUT(X_STEP_PIN, step_outbits.x);
                 ^~~~~~~~~~
./grblHal/main/i2s_out.h:56:22: note: each undeclared identifier is reported only once for each function it appears in
 #    define I2SO(n) (I2S_OUT_PIN_BASE + n)
....
```

dianlight, 2022-01-24:
I look in the code and try a solution; I add in `*_map.h` file the `I2S_OUT_PIN_BASE` and I was able to compile and use the firmware. The only issue is the homing phase. If I start $H the firmware freeze when the hard limit is trigged.
I was able to test hard limit and works correctly, and also axis move... only in homing there are problems.

I attach mu _map.h file for reference

```
/*
  mks_dlc32_2_0_map.h - An embedded CNC Controller with rs274/ngc (g-code) support

  Driver code for ESP32

  Part of grblHAL

  Copyright (c) 2022 Terje Io

  Grbl is free software: you can redistribute it and/or modify
  it under the terms of the GNU General Public License as published by
  the Free Software Foundation, either version 3 of the License, or
  (at your option) any later version.

  Grbl is distributed in the hope that it will be useful,
  but WITHOUT ANY WARRANTY; without even the implied warranty of
  MERCHANTABILITY or FITNESS FOR A PARTICULAR PURPOSE.  See the
  GNU General Public License for more details.

  You should have received a copy of the GNU General Public License
  along with Grbl.  If not, see <http://www.gnu.org/licenses/>.
*/

#if VFD_SPINDLE
#error "Board BOARD_MKS_DLC32_V2P0 does not have support for VFD spindle."
#endif

#define BOARD_NAME "MKS DLC32 2.0"

#define USE_I2S_OUT
#define I2S_OUT_PIN_BASE 64

// timer definitions
#define STEP_TIMER_GROUP TIMER_GROUP_0
#define STEP_TIMER_INDEX TIMER_0

#if SDCARD_ENABLE

// Pin mapping when using SPI mode.
// With this mapping, SD card can be used both in SPI and 1-line SD mode.
// Note that a pull-up on CS line is required in SD mode.
#define PIN_NUM_MISO GPIO_NUM_12
#define PIN_NUM_MOSI GPIO_NUM_13
#define PIN_NUM_CLK GPIO_NUM_14
#define PIN_NUM_CS GPIO_NUM_15

#endif // SDCARD_ENABLE

#define I2S_OUT_BCK GPIO_NUM_16
#define I2S_OUT_WS GPIO_NUM_17
#define I2S_OUT_DATA GPIO_NUM_21

#define X_STEP_PIN I2SO(1)
#define X_DIRECTION_PIN I2SO(2)
#define X_ENABLE_PIN I2SO(0)
#define X_LIMIT_PIN GPIO_NUM_36

#define Y_STEP_PIN I2SO(5)
#define Y_DIRECTION_PIN I2SO(6)
#define Y_ENABLE_PIN I2SO(0)
#define Y_LIMIT_PIN GPIO_NUM_35

#define Z_STEP_PIN I2SO(3)
#define Z_DIRECTION_PIN I2SO(4)
#define Z_ENABLE_PIN I2SO(0)
#define Z_LIMIT_PIN GPIO_NUM_34

// Define ganged axis or A axis step pulse and step direction output pins.
#if N_ABC_MOTORS >= 1
#error "Board BOARD_MKS_DLC32_V2P0 does not have support for ABC Motors"
#endif

// Define spindle enable and spindle direction output pins.

#define SPINDLEPWMPIN GPIO_NUM_32
#define SPINDLE_ENABLE_PIN GPIO_NUM_27

// Define flood and mist coolant enable output pins.

#define COOLANT_MIST_PIN I2SO(7)
#define COOLANT_FLOOD_PIN GPIO_NUM_5

// Define user-control CONTROLs (cycle start, reset, feed hold) input pins.

// N/A

// Define probe switch input pin.
#if PROBE_ENABLE
#define PROBE_PIN GPIO_NUM_22
#endif

#if I2C_ENABLE
// Define I2C port/pins
#define I2C_SDA GPIO_NUM_0
#define I2C_SCL GPIO_NUM_4
#define I2C_CLOCK 100000
#endif

```

dianlight, 2022-01-25:
I have test all for map and the only problem is the homing.
When the limit switch is hit the CPU go in panic.
I try differents config ( single axis, no pull off etc etc ) but no changes, I have can't verify but I think is a bug in the ESP32 driver.

Panic error from serial console:

```
Guru Meditation Error: Core  1 panic'ed (LoadProhibited). Exception was unhandled.

Core  1 register dump:
PC      : 0x400837cb  PS      : 0x00060031  A0      : 0x80083a30  A1      : 0x3ffbef70  
A2      : 0x3ffc42e0  A3      : 0x3ffce8c4  A4      : 0x00000000  A5      : 0x3ffc4310  
A6      : 0x3ff000e0  A7      : 0x00000001  A8      : 0x800837c8  A9      : 0x3ffbef50  
A10     : 0x400e487c  A11     : 0x000000ff  A12     : 0x00060021  A13     : 0x00060023  
A14     : 0x00000001  A15     : 0xffffffff  SAR     : 0x00000020  EXCCAUSE: 0x0000001c  
EXCVADDR: 0x00000030  LBEG    : 0x4000c46c  LEND    : 0x4000c477  LCOUNT  : 0x00000000  

Backtrace:0x400837c8:0x3ffbef70 0x40083a2d:0x3ffbef90 0x40082be5:0x3ffbefb0 0x4008d25f:0x3ffb5800 0x40083dd9:0x3ffb5820 0x400db059:0x3ffb5850 0x400dc00c:0x3ffb5910 0x400e3212:0x3ffb5950 0x400e3271:0x3ffb5970 0x400e3641:0x3ffb5990 0x400dd345:0x3ffb59e0 0x400d7ab4:0x3ffb5a20 0x400d77c7:0x3ffb5a40

ELF file SHA256: 497cc5ec58499d4a

CPU halted.
```

terjeio, 2022-01-26:
Fix committed for you to test, I have added your board map too.

dianlight, 2022-01-27:
Thx, works perfectly.

I test is with a PWM spindle and a 20w Laser without issue.

I  notice you don't add the _map.h file in your commit #17 but only the reference. Is a missing?
Also in my_machine.h is missing in comment #define 

Do you want a PR for adding the _map?

L.

terjeio, 2022-01-27:
> Do you want a PR for adding the _map?

I just added it, I forgot `git add` before committing yesterday.

silvanopas, 2022-02-07:
Hello terjeio
Is it possible to communicate via WIFI with the MKS DLC 32 board?
I tried to activate it in mymachine.h but I don't see it ... surely something wrong ...

silvanopas, 2022-02-07:
> 

> > I tried to activate it in mymachine.h but I don't see it ... surely something wrong ...
> 
> Options are enabled in [CMakeLists.txt](https://github.com/grblHAL/ESP32/blob/master/main/CMakeLists.txt) for this driver.

Thanks

silvanopas, 2022-02-09:
> > Ho provato ad attivarlo in mymachine.h ma non lo vedo... sicuramente qualcosa non va...
> 
> Le opzioni sono abilitate in [CMakeLists.txt](https://github.com/grblHAL/ESP32/blob/master/main/CMakeLists.txt) per questo driver.

if active
OPTION (Networking "Wifi + protocols" ON)
and
OPTION(SDcard "SD Card Streaming" ON)

when compiling it gives me these errors, what am I wrong?

these are the mistakes:

FAILED: grbl.elf
cmd.exe /C "cd . && C:\Users\pasel\.espressif\tools\xtensa-esp32-elf\esp-2021r2-8.4.0\xtensa-esp32-elf\bin\xtensa-esp32-elf-g++.exe  -mlongcalls -Wno-frame-address   @CMakeFiles\grbl.elf.rsp  -o grbl.elf  && cd ."
c:/users/pasel/.espressif/tools/xtensa-esp32-elf/esp-2021r2-8.4.0/xtensa-esp32-elf/bin/../lib/gcc/xtensa-esp32-elf/8.4.0/../../../../xtensa-esp32-elf/bin/ld.exe: esp-idf/wpa_supplicant/libwpa_supplicant.a(base64.c.obj): in function `base64_encode':
D:/ESP/esp-idf/components/wpa_supplicant/src/utils/base64.c:166: multiple definition of `base64_encode'; esp-idf/main/libmain.a(base64.c.obj):d:\esp32\build/../main/networking/base64.c:39: first defined here
c:/users/pasel/.espressif/tools/xtensa-esp32-elf/esp-2021r2-8.4.0/xtensa-esp32-elf/bin/../lib/gcc/xtensa-esp32-elf/8.4.0/../../../../xtensa-esp32-elf/bin/ld.exe: esp-idf/wpa_supplicant/libwpa_supplicant.a(base64.c.obj): in function `base64_decode':
D:/ESP/esp-idf/components/wpa_supplicant/src/utils/base64.c:188: multiple definition of `base64_decode'; esp-idf/main/libmain.a(base64.c.obj):d:\esp32\build/../main/networking/base64.c:90: first defined here
c:/users/pasel/.espressif/tools/xtensa-esp32-elf/esp-2021r2-8.4.0/xtensa-esp32-elf/bin/../lib/gcc/xtensa-esp32-elf/8.4.0/../../../../xtensa-esp32-elf/bin/ld.exe: esp-idf/main/libmain.a(ftpd.c.obj):(.literal.cmd_cdup+0x4): undefined reference to `f_chdir'
c:/users/pasel/.espressif/tools/xtensa-esp32-elf/esp-2021r2-8.4.0/xtensa-esp32-elf/bin/../lib/gcc/xtensa-esp32-elf/8.4.0/../../../../xtensa-esp32-elf/bin/ld.exe: esp-idf/main/libmain.a(ftpd.c.obj): in function `cmd_cdup':
d:\esp32\build/../main/networking/ftpd.c:648: undefined reference to `f_chdir'
c:/users/pasel/.espressif/tools/xtensa-esp32-elf/esp-2021r2-8.4.0/xtensa-esp32-elf/bin/../lib/gcc/xtensa-esp32-elf/8.4.0/../../../../xtensa-esp32-elf/bin/ld.exe: esp-idf/main/libmain.a(ftpd.c.obj): in function `cmd_cwd':
d:\esp32\build/../main/networking/ftpd.c:643: undefined reference to `f_chdir'
c:/users/pasel/.espressif/tools/xtensa-esp32-elf/esp-2021r2-8.4.0/xtensa-esp32-elf/bin/../lib/gcc/xtensa-esp32-elf/8.4.0/../../../../xtensa-esp32-elf/bin/ld.exe: esp-idf/main/libmain.a(vfs.c.obj):(.literal.vfs_getcwd+0x0): undefined reference to `f_getcwd'
c:/users/pasel/.espressif/tools/xtensa-esp32-elf/esp-2021r2-8.4.0/xtensa-esp32-elf/bin/../lib/gcc/xtensa-esp32-elf/8.4.0/../../../../xtensa-esp32-elf/bin/ld.exe: esp-idf/main/libmain.a(vfs.c.obj): in function `vfs_getcwd':
d:\esp32\build/../main/networking/vfs.c:159: undefined reference to `f_getcwd'
collect2.exe: error: ld returned 1 exit status
ninja: build stopped: subcommand failed.
ninja failed with exit code 1

D:\ESP32>

terjeio, 2022-02-09:
The duplicate code is due to you having something enabled in the [sdkconfig](https://github.com/grblHAL/ESP32/blob/master/sdkconfig) that should not be enabled?

The missing references is likely due to a source modification that you have to do, [described here](https://github.com/grblHAL/ESP32#esp32-grblhal-driver), see under the Important heading.

silvanopas, 2022-02-10:
Nothing to do, keep giving me the same mistakes

sdkconfig is the same as yours

and

#define FF_FS_RPATH 2

terjeio, 2022-02-10:
You are compiling with ESP IDF? Which version?

silvanopas, 2022-02-10:
> You are compiling with ESP IDF? Which version?

v4.3

terjeio, 2022-02-10:
Ok, same as I have - and the project compiles fine with both the command line version and the Eclipse build I mainly use.

Post the CMakeLists.txt you use and I can try to build a binary for you.

silvanopas, 2022-02-10:
[CMakeLists.txt](https://github.com/grblHAL/ESP32/files/8040475/CMakeLists.txt)
thanks

terjeio, 2022-02-10:
Here you go:

[app-template.zip](https://github.com/grblHAL/ESP32/files/8041226/app-template.zip)

silvanopas, 2022-02-14:
Thanks Tejero
I tried to install it but it doesn't work and everything crashes.

silvanopas, 2022-02-14:
Other problem:

Through lightburn, if I perform a STOP during execution, this message sometimes appears and the engine runs away:

ALARM:3
Ripristina mentre sei in movimento. Grbl non può garantire la posizione. È probabile che avrai passi persi. Il re-homing è altamente raccomandato.
assert failed: taskSelectHighestPriorityTaskSMP tasks.c:3317 (xTaskScheduled == pdTRUE)
Backtrace:0x4008183a:0x3ffdc1a0 0x40092469:0x3ffdc1c0 0x40097c0d:0x3ffdc1e0 0x40094056:0x3ffdc300 0x40092560:0x3ffdc330 0x40092512:0x00000000 |<-CORRUPTED
ELF file SHA256: 6af66d48a9b7b41f
CPU halted.

terjeio, 2022-02-14:
Try this, Bluetooth and WiFi enabled at the same time does not work for some reason and causes constant reboots. 

[app-template.zip](https://github.com/grblHAL/ESP32/files/8061437/app-template.zip)

> the engine runs away:

By this you mean XYZ axes are moving?

terjeio, 2022-02-15:
Try this binary:
[app-template.zip](https://github.com/grblHAL/ESP32/files/8068940/app-template.zip)

Note that I do not have a I2S board to test with so I have not verified the motor signals.

silvanopas, 2022-02-16:
Are you sure you are compiling for the DLC32 mks?
I install your bins but they don't work and Lightburn keeps giving me this
message:

ets Jun 8 2016 00:22:57
rst: 0x10 (RTCWDT_RTC_RESET), boot: 0x3 (DOWNLOAD_BOOT (UART0 / UART1 /
SDIO_REI_REO_V2))
waiting for download

terjeio, 2022-02-17:
This is what I get when I flash the binary amd send a $i command:

silvanopas, 2022-02-17:
But what compiler are you using?
With what commands do you transmit the template-bin to the DLC32 card?

terjeio, 2022-02-17:
> But what compiler are you using?

I have no idea (gcc?), I building using ESP-IDF 4.3 with Eclipse - installed via Espressifs installer. I can also build and flash with the CLI tool (idf.py). And earlier I tried VSCode as well - but was not happy with that even if it kind of worked.

> With what commands do you transmit the template-bin to the DLC32 card?

I am using [flash_download_tool_3.9.2.exe](https://www.espressif.com/en/support/download/other-tools) to flash the image. So I am not using any command line commands.

BTW I have just committed the changes used for the build.

silvanopas, 2022-02-22:
> Sto usando [flash_download_tool_3.9.2.exe](https://www.espressif.com/en/support/download/other-tools) per eseguire il flashing dell'immagine. Quindi non sto usando alcun comando da riga.

Kindly tell me how to use flash_download_tool_3.9.2.exe to install your bin?

terjeio, 2022-02-22:
This works for me:

[image]

[image]

silvanopas, 2022-02-22:
I give up 😥

It doesn't work even with your parameters, the only way to install the
firmware on the MKS DLC32 is to use "idf.py -p COM3 flash" with the source
installed, unfortunately your template.bin can't install it.
Lightburn always replies to me:

Waiting for the connection ...
ets Jun 8 2016 00:22:57
rst: 0x10 (RTCWDT_RTC_RESET), boot: 0x3 (DOWNLOAD_BOOT (UART0 / UART1 /
SDIO_REI_REO_V2))
waiting for download

as if the firmware is not installed ...

silvanopas, 2022-02-22:
I solved
adding these 2 files, thanks anyway.
[image]

closed 2022-09-27 by terjeio

Ams-Laser, 2023-03-13:
Hello
new version of download tools is difrent
[image]
can someone help

dJOS1475, 2024-01-18:
> Hello new version of download tools is difrent [image] can someone help

Setup the files as above, then use the CombineBin button to make a complete FW file. Then you can use the tool MKS made:
https://github.com/makerbase-mks/MKS-DLC32/tree/main/doc

distebia, 2024-06-10:
Salve, scusate se riprendo questa vecchia discussione...ho generato il firmware da web build ma non riesco a caricarlo con nessuna versione di esp32 download flash tools (3.6.5-3.9.6 e 3.9.7 rilasciata 2 giorni fa). carico il bootloader, partition e firmware con gli indirizzi ricevuti con il firmware ma la scheda non accetta niente e non viene riconosciuta...ho provato sia DIO che DOUT, 8mb e anche 64mb...niente da fare..dove sbaglio?

dJOS1475, 2024-06-10:
Try my instructions over here:

https://github.com/dJOS1475/grblHAL_Backup

distebia, 2024-06-10:
grazie per la risposta..il firmware che ho generato era con LB clustering e niente wifi o sd..dopo aver generato ottengo 3 file+indirizzi nel txt, li carico con esp download tools (qualsiasi versione) e niente, creo anche il combine e lo carico con l'app ufficiale e niente

distebia, 2024-06-11:
forse sto capendo qualcosa..in sostanza mi consigli di eseguire il flash con la versione MKSLaserTool_setupV1.0.6 caricando i 3 file oppure utilizzare CombineBin? inoltre una mks dlc32 dovrebbe essere cristal 40mhz, DIO, Flash size 8mb, oppure mi sbaglio?

dJOS1475, 2024-06-11:
You need to combine the 3 files into a single binary using the ESP32 Download tool:

[image]

then follow my GitHub instructions to flash the controller via USB.

distebia, 2024-06-11:
Ok, unisco i 3 binari con versione 3.6.5 con 40 clock, dio, 64mb? Mks dlc32 ha 8mb. Ottengo un result.bin e lo carico con quale versione? Stessi parametri di flash? Indirizzo?

dJOS1475, 2024-06-11:
Sorry, I should have been more clear. You can ignore all the settings under the table, they are just is you are using the tool ESP32 tool to flash (which doesn’t work IME).

The important settings are the order of the files down in the table, and the address values to the right of them. I’ve you have this correct, click CombineBin and the file it creates is what you flash using the MKS Laser tool on my GitHub.

distebia, 2024-06-11:
Ok, capito...ma combino con 3.6.5 e flash con la versione che hai caricato nel repositor? E l'indirizzo e impostazioni di flash?

dJOS1475, 2024-06-11:
Exactly: 

Step 1: make single Bin file with ESP32 tool
Step 2: Flash bin file with MKS laser Tool

distebia, 2024-06-12:
ci rinuncio...non mi riesce a caricare il firmware. il firmware rimane nello stato "wait for download". La procedura che faccio è questa: genero il firmware per esp32+mks dlc32 con solo LB clustering attivato. Ottengo nel file zip i 3 file + indirizzi. Apro lo strumento esp 3.6.5 e imposto nel seguente ordine: Bootloader.bin 0x1000, Partition.bin 0x8000 e Firmware.bin 0x10000. Premo il tasto CombineBin e ottengo il Target.bin. Apro lo strumento mks 1.0.6 precedentemente installato. apro il file Target.bin ed eseguo il flash con cristal 80mhz, size 4mb ecc..niente da fare. rifaccio la procedura formattando la scheda e ripartendo da zero. apro strumento mks 1.0.6 e apro target.bin, cristal 40mhz e size 8mb...niente di niente. Rifaccio ancora la procedura e, dopo aver nuovamente formattato, effettuo il flash con strumento mks 3.6.5 selezionando solo target.bin e indirizzo 0x10000. faccio prima con DoNotChgBin e niente, poi con DoNotChgBin attivo. Niente da fare in nessun modo. La scheda accetta il firmware originale e carico tranquillamente FluidNc.

dJOS1475, 2024-06-12:
Do You have the LCD touch screen connected? That will prevent any 3rd party firmware from being loaded.

distebia, 2024-06-12:
Forse ho capito dove sbaglio, ma ancora devo verificarlo...in sostanza nel generare il Target.bin con strumento esp sbagliavo l'ordine dei file, mettevo bootloader (0x1000), poi partitions (0x8000) e infine firmware (0x10000). come da vostra foto dovrei fare partition.bin (0x8000) su pos.1, bootloader.bin (0x1000) su pos.2 e firmware.bin (0x10000) su pos.3
[image]
Ancora da verificare ma sicuramente l'errore sarà qua

dJOS1475, 2024-06-12:
I do think the order matters - this is the order I’ve used successfully:

**Flash adresses:**
bootloader.bin 0x1000
partitions.bin 0x8000
firmware.bin   0x10000

distebia, 2024-06-12:
allora se l'ordine è bootloader, partition e firmware non ho altro da fare, la scheda è senza niente connesso e viene eseguito il comando erase con strumento mks prima di ogni nuovo flash, spenta e riaccesa ad ogni passaggio...in rete ho trovato che bisognava mettere in corto qualcosa come gpio.4 e gnd prima del flash ma non trovo nuovamente la discussione

dJOS1475, 2024-06-12:
Hmm, I’ve never had to short any pins on my DLC32 v2.x revision board. I just power it on, select the correct com port and program it.

I assume you have the MKS USB CH340G windows drivers installed?

https://github.com/makerbase-mks/MKS-DLC32/tree/main/MKS-DLC32-main/firmware/tool

distebia, 2024-06-13:
Allora...non ho mai messo in corto i pin per eseguire il flash, l'ho solo letto...ecco la mia procedura di oggi:
Installo i driver CH340G_USB.zip sostituendo i miei del 2019 (quelli mks 2011). genero il firmware con solo LB clustering attivato 
[image]
Eseguo il comando ERASE con tools MKS 1.0.6 
[image]
e riavvio la scheda
Apro lo strumento ESP 3.6.5 e seleziono ESP32 download tools
carico Bootloader su 0x1000, partition su 0x8000 e firmware su 0x10000 ed eseguo il CombineBin
[image]
Ottengo il file Target.bin
Apro il tools MKS 1.0.6 e carico Il file Target.bin con i parametri in foto 
[image]
[image]
Riavvio la scheda e apro Lightburn
Niente da fare...Lightburn non vede la scheda su com3 e rimane in questo stato
[image]
Questa è la mia procedura...La scheda accetta senza problemi sia il firmware originale che FluidNC...

dJOS1475, 2024-06-13:
I’m at a loss to explain where things are going wrong for you. I’ve had no issues with either FluidNC or grblHAL being flashed to mine.

Are you definitely selecting the correct board when you generate the firmware via the web builder?

[image]

distebia, 2024-06-13:
> Allora...non ho mai messo in corto i pin per eseguire il flash, l'ho solo letto...ecco la mia procedura di oggi: Installo i driver CH340G_USB.zip sostituendo i miei del 2019 (quelli mks 2011). genero il firmware con solo LB clustering attivato
> 
> ![Immagine](https://private-user-images.githubusercontent.com/52679577/339345027-819c9dee-d529-4485-bfdc-7c4f4b7a0362.png?jwt=eyJhbGciOiJIUzI1NiIsInR5cCI6IkpXVCJ9.eyJpc3MiOiJnaXRodWIuY29tIiwiYXVkIjoicmF3LmdpdGh1YnVzZXJjb250ZW50LmNvbSIsImtleSI6ImtleTUiLCJleHAiOjE3MTgzMTczMTYsIm5iZiI6MTcxODMxNzAxNiwicGF0aCI6Ii81MjY3OTU3Ny8zMzkzNDUwMjctODE5YzlkZWUtZDUyOS00NDg1LWJmZGMtN2M0ZjRiN2EwMzYyLnBuZz9YLUFtei1BbGdvcml0aG09QVdTNC1ITUFDLVNIQTI1NiZYLUFtei1DcmVkZW50aWFsPUFLSUFWQ09EWUxTQTUzUFFLNFpBJTJGMjAyNDA2MTMlMkZ1cy1lYXN0LTElMkZzMyUyRmF3czRfcmVxdWVzdCZYLUFtei1EYXRlPTIwMjQwNjEzVDIyMTY1NlomWC1BbXotRXhwaXJlcz0zMDAmWC1BbXotU2lnbmF0dXJlPWFjNTcyMDUwZjQ1MzU1N2Y3N2NkZGJiYjAzOGFkMzBiZjUxMGM0MmYxMDM3MzdiZDgxNGFhYzM1ZTM3NDdiMmYmWC1BbXotU2lnbmVkSGVhZGVycz1ob3N0JmFjdG9yX2lkPTAma2V5X2lkPTAmcmVwb19pZD0wIn0.H8zD27Qczcyvd08AZCHKfH_vxoV0JyeVpKVgrv6DRvg)
> 
> Eseguo il comando ERASE con tools MKS 1.0.6
> 
> ![Immagine](https://private-user-images.githubusercontent.com/52679577/339345561-1cd9cca7-0143-4c6c-ba9a-a1776129d562.png?jwt=eyJhbGciOiJIUzI1NiIsInR5cCI6IkpXVCJ9.eyJpc3MiOiJnaXRodWIuY29tIiwiYXVkIjoicmF3LmdpdGh1YnVzZXJjb250ZW50LmNvbSIsImtleSI6ImtleTUiLCJleHAiOjE3MTgzMTczMTYsIm5iZiI6MTcxODMxNzAxNiwicGF0aCI6Ii81MjY3OTU3Ny8zMzkzNDU1NjEtMWNkOWNjYTctMDE0My00YzZjLWJhOWEtYTE3NzYxMjlkNTYyLnBuZz9YLUFtei1BbGdvcml0aG09QVdTNC1ITUFDLVNIQTI1NiZYLUFtei1DcmVkZW50aWFsPUFLSUFWQ09EWUxTQTUzUFFLNFpBJTJGMjAyNDA2MTMlMkZ1cy1lYXN0LTElMkZzMyUyRmF3czRfcmVxdWVzdCZYLUFtei1EYXRlPTIwMjQwNjEzVDIyMTY1NlomWC1BbXotRXhwaXJlcz0zMDAmWC1BbXotU2lnbmF0dXJlPTkzY2NlMWIzMjlkMWU5NDMzMjY4MTQ5N2FkMzZkM2ZkNTI5MWVlNWYzYTU4NjE5ZWNkMmIzODZmMWU4ZjM2MGUmWC1BbXotU2lnbmVkSGVhZGVycz1ob3N0JmFjdG9yX2lkPTAma2V5X2lkPTAmcmVwb19pZD0wIn0.3NIOq8gbJRbtjyiCx7AOJl2sZlV0oLfzCXP5frL7qnU) e riavviare la scheda
> 
> Apro lo strumento ESP 3.6.5 e seleziono ESP32 download tools carico Bootloader su 0x1000, partizione su 0x8000 e firmware su 0x10000 ed eseguo il CombineBin
> 
> ![Immagine](https://private-user-images.githubusercontent.com/52679577/339347535-eee79c02-e9ed-44cb-a2d1-d9b9af1a7c61.png?jwt=eyJhbGciOiJIUzI1NiIsInR5cCI6IkpXVCJ9.eyJpc3MiOiJnaXRodWIuY29tIiwiYXVkIjoicmF3LmdpdGh1YnVzZXJjb250ZW50LmNvbSIsImtleSI6ImtleTUiLCJleHAiOjE3MTgzMTczMTYsIm5iZiI6MTcxODMxNzAxNiwicGF0aCI6Ii81MjY3OTU3Ny8zMzkzNDc1MzUtZWVlNzljMDItZTllZC00NGNiLWEyZDEtZDliOWFmMWE3YzYxLnBuZz9YLUFtei1BbGdvcml0aG09QVdTNC1ITUFDLVNIQTI1NiZYLUFtei1DcmVkZW50aWFsPUFLSUFWQ09EWUxTQTUzUFFLNFpBJTJGMjAyNDA2MTMlMkZ1cy1lYXN0LTElMkZzMyUyRmF3czRfcmVxdWVzdCZYLUFtei1EYXRlPTIwMjQwNjEzVDIyMTY1NlomWC1BbXotRXhwaXJlcz0zMDAmWC1BbXotU2lnbmF0dXJlPWUzMzY0MTM4YzhkMGJiNzUzZDhiNmVlOWU4NmVkMTE1YjU4MTQyNWM4MzUxOGIzMzA0NmFmNTEyY2FmNmQxNzUmWC1BbXotU2lnbmVkSGVhZGVycz1ob3N0JmFjdG9yX2lkPTAma2V5X2lkPTAmcmVwb19pZD0wIn0.7Ss26U2bYqNJ5nAe9nLPZN7Gj6emTEZpvm4B0Kctc2Q)
> 
> Ottengo il file Target.bin
> 
> Apro il tools MKS 1.0.6 e carico Il file Target.bin con i parametri in foto
> 
> ![Immagine](https://private-user-images.githubusercontent.com/52679577/339355097-ac16b8cd-5975-4db2-81cb-a52d62afdd61.png?jwt=eyJhbGciOiJIUzI1NiIsInR5cCI6IkpXVCJ9.eyJpc3MiOiJnaXRodWIuY29tIiwiYXVkIjoicmF3LmdpdGh1YnVzZXJjb250ZW50LmNvbSIsImtleSI6ImtleTUiLCJleHAiOjE3MTgzMTczMTYsIm5iZiI6MTcxODMxNzAxNiwicGF0aCI6Ii81MjY3OTU3Ny8zMzkzNTUwOTctYWMxNmI4Y2QtNTk3NS00ZGIyLTgxY2ItYTUyZDYyYWZkZDYxLnBuZz9YLUFtei1BbGdvcml0aG09QVdTNC1ITUFDLVNIQTI1NiZYLUFtei1DcmVkZW50aWFsPUFLSUFWQ09EWUxTQTUzUFFLNFpBJTJGMjAyNDA2MTMlMkZ1cy1lYXN0LTElMkZzMyUyRmF3czRfcmVxdWVzdCZYLUFtei1EYXRlPTIwMjQwNjEzVDIyMTY1NlomWC1BbXotRXhwaXJlcz0zMDAmWC1BbXotU2lnbmF0dXJlPTNlYmFmZDY1NzE1YjM3NzlhZTE4MDIyMjg3ZWFlOTBjZjI5OGQxOWZlNDM4NjMwMmI3MGRjOThiN2MzZTA2YTEmWC1BbXotU2lnbmVkSGVhZGVycz1ob3N0JmFjdG9yX2lkPTAma2V5X2lkPTAmcmVwb19pZD0wIn0.28qURJZVrZNjNtQ_CN8MeuRrPPWfFhvFDX7KLK2SA04)
> 
> ![Immagine](https://private-user-images.githubusercontent.com/52679577/339355401-db71543a-4e2d-40b3-9654-8812783119db.png?jwt=eyJhbGciOiJIUzI1NiIsInR5cCI6IkpXVCJ9.eyJpc3MiOiJnaXRodWIuY29tIiwiYXVkIjoicmF3LmdpdGh1YnVzZXJjb250ZW50LmNvbSIsImtleSI6ImtleTUiLCJleHAiOjE3MTgzMTczMTYsIm5iZiI6MTcxODMxNzAxNiwicGF0aCI6Ii81MjY3OTU3Ny8zMzkzNTU0MDEtZGI3MTU0M2EtNGUyZC00MGIzLTk2NTQtODgxMjc4MzExOWRiLnBuZz9YLUFtei1BbGdvcml0aG09QVdTNC1ITUFDLVNIQTI1NiZYLUFtei1DcmVkZW50aWFsPUFLSUFWQ09EWUxTQTUzUFFLNFpBJTJGMjAyNDA2MTMlMkZ1cy1lYXN0LTElMkZzMyUyRmF3czRfcmVxdWVzdCZYLUFtei1EYXRlPTIwMjQwNjEzVDIyMTY1NlomWC1BbXotRXhwaXJlcz0zMDAmWC1BbXotU2lnbmF0dXJlPWM4ZTQyOGYxMTExMTg3ZGZiNDMyNGI3YzE0M2FjY2FlZjdlYjY5NjNmMWY2NjA3MzkzNzRlNGQzNTUzYWUyYzQmWC1BbXotU2lnbmVkSGVhZGVycz1ob3N0JmFjdG9yX2lkPTAma2V5X2lkPTAmcmVwb19pZD0wIn0.hKV1Gv7E9EFER594Yvjui9fVLNHFB2Hzj4ssBzcKEeY)
> 
> Riavviare la scheda e apro Lightburn
> 
> Niente da fare...Lightburn non vede la scheda su com3 e rimane in questo stato
> 
> ![Immagine](https://private-user-images.githubusercontent.com/52679577/339356411-5ddf109f-507c-4539-8990-013d22c27bf8.png?jwt=eyJhbGciOiJIUzI1NiIsInR5cCI6IkpXVCJ9.eyJpc3MiOiJnaXRodWIuY29tIiwiYXVkIjoicmF3LmdpdGh1YnVzZXJjb250ZW50LmNvbSIsImtleSI6ImtleTUiLCJleHAiOjE3MTgzMTczMTYsIm5iZiI6MTcxODMxNzAxNiwicGF0aCI6Ii81MjY3OTU3Ny8zMzkzNTY0MTEtNWRkZjEwOWYtNTA3Yy00NTM5LTg5OTAtMDEzZDIyYzI3YmY4LnBuZz9YLUFtei1BbGdvcml0aG09QVdTNC1ITUFDLVNIQTI1NiZYLUFtei1DcmVkZW50aWFsPUFLSUFWQ09EWUxTQTUzUFFLNFpBJTJGMjAyNDA2MTMlMkZ1cy1lYXN0LTElMkZzMyUyRmF3czRfcmVxdWVzdCZYLUFtei1EYXRlPTIwMjQwNjEzVDIyMTY1NlomWC1BbXotRXhwaXJlcz0zMDAmWC1BbXotU2lnbmF0dXJlPWViOGQ3N2NiZTQ0OTY4ZjkwN2M5YWI1ZDBkZTRjNzM1NmQ4ZmVjMjY4ZWVlMjkwMGZkODU5YmIyZmQ1NWQ5Y2EmWC1BbXotU2lnbmVkSGVhZGVycz1ob3N0JmFjdG9yX2lkPTAma2V5X2lkPTAmcmVwb19pZD0wIn0.ec0DSY5LGQVu7W6-WWZjN59VJA44V_kyd0xI6yMe7yI)
> 
> Questa è la mia procedura...La scheda accetta senza problemi sia il firmware originale che FluidNC...

yes, la scheda è corretta

distebia, 2024-06-15:
oggi ho voluto fare una prova...tempo fa (settembre 2023) ho ricevuto assistenza su LB cluster che generava errore e mi è stato spedito un firmware generato da web build contenente la nuova versione di LB cluster 0.6. Eseguo tutto con lo strumento esp 3.6.5, sia per erase che per flash con 40mhz, dio, 8mb e ho aggiunto DoNotChgBin. carico bootloader 0x1000, partition 0x8000 e firmware 0x10000...La scheda risponde correttamente ai comandi e possi eseguire grblhal....Non contento rifaccio tutto esattamente come sopra ma carico il firmware scaricato oggi da web build...niente non va, rimetto quello di settembre e funziona...forse qualcosa è cambiato quando viene generato il firmware?

dJOS1475, 2024-06-15:
That does sound like there’s something wrong with the current firmware being generated by the web builder. It might be worth your while to raise a bug request.

terjeio, 2024-06-17:
This issue was closed a long time ago and should not be reused. Please continue the discussion [here](https://github.com/grblHAL/ESP32/issues/111).