grblHAL/ESP32 #19: Unable to compile without USE_I2S_OUT

petervanderwalt, 2022-03-18:
I am just starting to setup a board definition to experiment with grblHal on ESP32

my pinmap:
```
#if N_ABC_MOTORS > 1
#error "Axis configuration is not supported!"
#endif

#define BOARD_NAME "Peter's little test"
#define HAS_IOPORTS
//#define HAS_BOARD_INIT

//#define USE_I2S_OUT  //  If I uncomment this and pin definitions, it compiles.  If I comment out I2S compilation fails. 
//#define I2S_OUT_BCK GPIO_NUM_16
//#define I2S_OUT_WS GPIO_NUM_17
//#define I2S_OUT_DATA GPIO_NUM_21

// timer definitions
#define STEP_TIMER_GROUP TIMER_GROUP_0
#define STEP_TIMER_INDEX TIMER_0

// Define step pulse output pins.
#define X_STEP_PIN          GPIO_NUM_12
#define Y_STEP_PIN          GPIO_NUM_27
#define Z_STEP_PIN          GPIO_NUM_15

// Define step direction output pins.
#define X_DIRECTION_PIN     GPIO_NUM_14
#define Y_DIRECTION_PIN     GPIO_NUM_26
#define Z_DIRECTION_PIN     GPIO_NUM_2

// Define homing/hard limit switch input pins and limit interrupt vectors.
#define X_LIMIT_PIN         GPIO_NUM_35
#define Y_LIMIT_PIN         GPIO_NUM_34
#define Z_LIMIT_PIN         GPIO_NUM_39

// Define ganged axis or A axis step pulse and step direction output pins.
#if N_ABC_MOTORS > 0
#define M3_AVAILABLE
#define M3_STEP_PIN         GPIO_NUM_33
#define M3_DIRECTION_PIN    GPIO_NUM_32
#define M3_LIMIT_PIN        GPIO_NUM_36
#endif

// Define spindle enable and spindle direction output pins.
#define SPINDLE_ENABLE_PIN  GPIO_NUM_4
#define SPINDLEPWMPIN       GPIO_NUM_25

// Define flood and mist coolant enable output pins.
// Only one can be enabled!

#define COOLANT_MIST_PIN    GPIO_NUM_21
//#define COOLANT_FLOOD_PIN   GPIO_NUM_21
// Define user-control CONTROLs (cycle start, reset, feed hold) input pins.
#if SAFETY_DOOR_ENABLE
#define SAFETY_DOOR_PIN     GPIO_NUM_16
#else
#define RESET_PIN           GPIO_NUM_16 //??
#endif

// Define probe switch input pin.
#if PROBE_ENABLE
#define PROBE_PIN           GPIO_NUM_22
#endif

#ifdef HAS_IOPORTS
#define AUXINPUT0_PIN       GPIO_NUM_13
#define AUXINPUT1_PIN       GPIO_NUM_0
#endif

[... 370 lines cut ...]
            ^~~~~~~~~~~~~
            i2s_out_init
At top level:
/root/ESP32/main/i2s_out.c:462:23: warning: 'i2sOutTask' defined but not used [-Wunused-function]
 static void IRAM_ATTR i2sOutTask(void* parameter)
                       ^~~~~~~~~~
/root/ESP32/main/i2s_out.c:410:23: warning: 'i2s_out_intr_handler' defined but not used [-Wunused-function]
 static void IRAM_ATTR i2s_out_intr_handler (void *arg)
                       ^~~~~~~~~~~~~~~~~~~~
/root/ESP32/main/i2s_out.c:180:23: warning: 'i2s_out_reset_fifo' defined but not used [-Wunused-function]
 static void IRAM_ATTR i2s_out_reset_fifo (void)
                       ^~~~~~~~~~~~~~~~~~
/root/ESP32/main/i2s_out.c:90:22: warning: 'i2s_out_isr_handle' defined but not used [-Wunused-variable]
 static intr_handle_t i2s_out_isr_handle;
                      ^~~~~~~~~~~~~~~~~~
cc1: some warnings being treated as errors
make[2]: *** [esp-idf/main/CMakeFiles/__idf_main.dir/build.make:169: esp-idf/main/CMakeFiles/__idf_main.dir/i2s_out.c.obj] Error 1
make[1]: *** [CMakeFiles/Makefile2:4934: esp-idf/main/CMakeFiles/__idf_main.dir/all] Error 2
make: *** [Makefile:130: all] Error 2
make failed with exit code 2
root@firmwarebuild:~/ESP32#

```

What am I doing wrong (:  or is the I2S code somewhere outside a "if" where it should have been inside?

terjeio, 2022-03-18:
> ... is the I2S code somewhere outside a "if" where it should have been inside?

Yep - a regression, uncomment these lines:

https://github.com/grblHAL/ESP32/blob/a02a4a4202fc617c944a20bdff3bc43ae632e76d/main/i2s_out.c#L47
https://github.com/grblHAL/ESP32/blob/a02a4a4202fc617c944a20bdff3bc43ae632e76d/main/i2s_out.c#L906

petervanderwalt, 2022-03-18:
Tested OK - compiled this time - onwards with testing (:   

Thank you sooooo much
[image]

petervanderwalt, 2022-04-18:
Confirming that editing the files worked a treat, can close this issue - but please do remember to fix (:

terjeio, 2022-04-19:
> but please do remember to fix

[Fix has been comitted](https://github.com/grblHAL/ESP32/commit/9436a81bfe75c569c31f1c2750fbed37677ffcf1#diff-fe0e20af8bca2a921688ff3c767b6e43e834ec624fc69d9e25fe17f1038a8899) already...

petervanderwalt, 2022-04-19:
Thank you!  Closing issue then (:

closed 2022-04-19 by petervanderwalt