grblHAL/ESP32 #20: Need your help, compilation failed, got error

hankecnc, 2022-03-21:
I downloaded esp idf 4.3 on windows system, downloaded your source code, entered cd to enter the esp directory, entered idf.py build to compile,
Got the error, depending on how some work needs to be done before this?
[image]
[image]

hankecnc, 2022-03-21:
If it can't be solved, I hope you can provide me with a bin firmware, I want to replace the Grbl_Esp32 control board sold before with grblhal_esp32, Grbl_Esp32 can be compiled with vc code Platformio and arduino, which is much simpler for me,

hankecnc, 2022-03-21:
If the compilation problem is not easy to solve, can you provide a compiled bin file? Here is his pin assignment:
#define MACHINE_NAME "ESP32_V4"

#define X_STEP_PIN GPIO_NUM_21
#define X_DIRECTION_PIN GPIO_NUM_14
#define Y_STEP_PIN GPIO_NUM_26
#define Y_DIRECTION_PIN GPIO_NUM_15
#define Z_STEP_PIN GPIO_NUM_27
#define Z_DIRECTION_PIN GPIO_NUM_33

#define X_LIMIT_PIN GPIO_NUM_17
#define Y_LIMIT_PIN GPIO_NUM_4
#define Z_LIMIT_PIN GPIO_NUM_16

#define A_STEP_PIN GPIO_NUM_25
#define A_DIRECTION_PIN GPIO_NUM_13
//Adir GPIO_NUM_13
//Apul GPIO_NUM_25
#define SPINDLE_TYPE SpindleType::PWM
#define SPINDLE_OUTPUT_PIN GPIO_NUM_2 // PWM pin
#define SPINDLE_ENABLE_PIN GPIO_NUM_22 // Spindle start

#define COOLANT_FLOOD_PIN GPIO_NUM_12 // Coolant
#define PROBE_PIN GPIO_NUM_32 //Tool detection

hankecnc, 2022-03-21:
Judging from the more than 150 esp32 control boards that have been sold, for amateur CNC, people seem to like the grbl of the esp32 port. It has many functions. It can connect to the control computer without plugging in usb, and can use wifi, bluetooth, etc. control,

hankecnc, 2022-03-21:
In addition, I have established a Chinese grbl website to provide users with Chinese grbl tutorials, http://www.grbl.cc/

terjeio, 2022-03-21:
[image]

Looks like you have not downloaded the submodules with the [--recurse-submodules](https://github.com/grblHAL/core/wiki/Compiling-GrblHAL) option.

The simplest way to add your pin mapping is to create _my_machine_map.h_ in the main directory, add definitions, and enable that in [CMakelists.txt](https://github.com/grblHAL/ESP32/blob/a02a4a4202fc617c944a20bdff3bc43ae632e76d/main/CMakeLists.txt#L16). BTW the mappings shown above is not compatible with grblHAL.

>  Grbl_Esp32 can be compiled with vc code Platformio and arduino

It should be possible to build grblHAL with VSCode - I did that earlier, but I have gave up on it after an automatic upgrade broke something. It should be possible to use platformio as well if you are able to create an ini-file. Compiling with Arduino is not possible without more fundamental code changes?

FYI there is a [regression in the source](https://github.com/grblHAL/ESP32/issues/19#issuecomment-1072578529) that I have not yet comitted a fix for.

hankecnc, 2022-03-21:
It's frustrating, or it doesn't work, I don't know what I'm doing wrong? English is not my native language and it's a headach
[image]
e

hankecnc, 2022-03-21:
I used the --recurse-submodules download, also enabled a pinout file in my_machine.h, also in CMakeLists.txt
[image]
[image]
[image]
,

hankecnc, 2022-03-21:
[image]

terjeio, 2022-03-21:
You have to run the ESP-IDF startup script to setup the environment for compiling. On my machine there are two versions found on the Start menu in the ESP-IDF "folder" - this is what I get when starting _ESP-IDF 4.3 CMD_ shortcut:

[image]

Then I have to change directory to the project folder before building/flashing.

When building with idf.py _my_machine.h_ is [not used at all](https://github.com/grblHAL/ESP32/blob/a02a4a4202fc617c944a20bdff3bc43ae632e76d/main/my_machine.h#L4), configuration has to be done in [CMakeLists.txt](https://github.com/grblHAL/ESP32/blob/master/main/CMakeLists.txt), if you added _my_machine_map.h_ for your board you have to disable (set to OFF) this line:
https://github.com/grblHAL/ESP32/blob/a02a4a4202fc617c944a20bdff3bc43ae632e76d/main/CMakeLists.txt#L7
and enable (set to ON) this line:
https://github.com/grblHAL/ESP32/blob/a02a4a4202fc617c944a20bdff3bc43ae632e76d/main/CMakeLists.txt#L16

I recommend using _my_machine_map.h_ for custom mappings as this will not be overwritten on updates.

closed 2022-03-22 by hankecnc