grblHAL/ESP32 #21: Gang Axis with I2S does not work

fireup924, 2022-03-26:
I'm trying to Gang the Y axis with the 4th axis and it does not seem to be working for me. I'm using I2S mode, the M3_DIRECTION_PIN works but the M3_STEP_PIN and M3_ENABLE_PIN are not working.

fireup924, 2022-03-26:
I think I've fixed the problem in driver.c

after line 509 I added the following:

```
#ifdef GANGING_ENABLED
  #ifdef X2_ENABLE_PIN
    DIGITAL_OUT(X2_ENABLE_PIN, enable.x);
  #endif
  #ifdef Y2_ENABLE_PIN
    DIGITAL_OUT(Y2_ENABLE_PIN, enable.y);
  #endif
  #ifdef Z2_ENABLE_PIN
    DIGITAL_OUT(Z2_ENABLE_PIN, enable.z);
  #endif
#endif

```

and also on lines 616,  619 and 622

```
#ifdef X2_STEP_PIN
    DIGITAL_OUT(X_STEP_PIN, step_outbits_2.x);
#endif
#ifdef Y2_STEP_PIN
    DIGITAL_OUT(X_STEP_PIN, step_outbits_2.y);
#endif
#ifdef Z2_STEP_PIN
    DIGITAL_OUT(X_STEP_PIN, step_outbits_2.z);
#endif 
```

I changed them to :

```
#ifdef X2_STEP_PIN
    DIGITAL_OUT(X2_STEP_PIN, step_outbits_2.x);
#endif
#ifdef Y2_STEP_PIN
    DIGITAL_OUT(Y2_STEP_PIN, step_outbits_2.y);
#endif
#ifdef Z2_STEP_PIN
    DIGITAL_OUT(Z2_STEP_PIN, step_outbits_2.z);
#endif 
```

Those changes made it work for me, I haven't tested Auto Square feature yet...

terjeio, 2022-03-27:
Thanks for reporting, fix comitted.

closed 2023-04-16 by terjeio