grblHAL/ESP32 #22: Problem with ESP-WROOOM-32

vvv1954, 2022-04-09:
Hello,

I compiled codes with basic functions (no WiFi, no SD, etc) and uploaded the firmware to ESP-WROOOM-32.  But when connected it goes to the loop sending the welcome message. 
[image]
Also I tried one of yours build (Fix for issue #21) - the same problem.

terjeio, 2022-04-10:
Is this when connecting with a sender and/or a terminal?

vvv1954, 2022-04-10:
With Candle and LaserGRBL.

vvv1954, 2022-04-10:
With terminal (Espressif IDE) as follows
[image]

terjeio, 2022-04-10:
Try with [compatibility level](https://github.com/grblHAL/core/wiki/Compatibility-level) > 0, or [ioSender](https://github.com/terjeio/ioSender) if on Windows.

vvv1954, 2022-04-10:
I uploaded your build with "ninja flash" and everything is working (it was uploaded before with Espressif Download Tool). However when I am doing the same with my build it goes to the loop above. 

It seems there is something wrong with my Espressif IDE installation: mainly it reports no errors, but somtimes the errors like "bool not defined" of the warnings like "Member 'flash_size' was not initialized in this constructor" appear.

terjeio, 2023-01-29:
If you are not modifying code you may try the [Web Builder](http://svn.io-engineering.com:8080/) to generate firmware.
I am closing this issue as I have no idea about what is causing this issue and I have never seen it during development.

closed 2023-01-29 by terjeio