grblHAL/ESP32 #23: Recent changes to Serial queue?

petervanderwalt, 2022-04-26:
I last did a pull on March 18th, added a board definition and all was working great. 
Did a clean clone down today, put back my board definition (just a pinmap) and cmakelist entries (for the board include and name), built, and flashed - all good there just as always 

But something seems broken? 

[image]

Same if I try to connect from OpenBuildsCONTROL

terjeio, 2022-04-28:
Odd, I just flashed my test board with the latest version - no issues.
Add your CMakeLists.txt and pin map and I'll check.

petervanderwalt, 2022-04-28:
Thanks for the confirmation.  It always helps as a sanity check

So one other thing I just realized - along with the clean clone, I did install IDF again too:
The readme here says "First you have to prepare esp-idf v4.3:"
I did actually somehow install 4.4 (esp-idf-v4.4.1)

Could be related? I'll get going on reinstalling as 4.3 though

petervanderwalt, 2022-04-29:
Confirmed, removed IDF4.4, Installed IDF4.3 - nothing else changed and all working again.

closed 2022-04-29 by petervanderwalt