grblHAL/ESP32 #28: Help setting up as AP

Henrikastro, 2022-06-20:
So, I'm trying to set up my ESP32 as an AP and I've already setted the CMakesList.txt settings of Networking as SoftAP as on and I still have no AP created. Any setting above $70 gives me an "error:53 - Setting is not available, possibly due to limited driver support.". It cannot be a limited driver issue since I've just setted up my ESP as an AP using FluidNC and it seems like I'm missing something on the setup of GrblHAL itself. Any ideas?

terjeio, 2022-06-20:
Have you enabled [Wifi+protocols](https://github.com/grblHAL/ESP32/blob/265ae0639eee46bd36e163163bc3f0088ebf4e40/main/CMakeLists.txt#L28)?

Henrikastro, 2022-06-20:
yes I have

terjeio, 2022-06-20:
Your `$$=73` output? I get
```
$73: WiFi Mode:
    0 - Off
    1 - Station
    2 - Access Point
    3 - Access Point/Station
ok

```

Henrikastro, 2022-06-20:
Soo, I just tried it after having used fluidNC and it worked. Maybe the version I was flashing was not enabling it or something. Thank you for the attention!

closed 2022-06-20 by Henrikastro