grblHAL/ESP32 #30: Direct connect using Ethernet with a W5500

Henrikastro, 2022-06-23:
So, I'm trying to direct connect GrblHAL with an Ethernet port to my Computer. I've seen the telnet and websocket plugins and it seems that if I wanna make this happen I would have to create something from them. I've also seen that there is already an SPI support because of the trinamic drivers and I was wondering if maybe that could somehow make my life easier setting this up?

I've already figured out how to set an static IP to my ESP32 and connect to it using other examples from the ESP-IDF framework, but I'm having trouble on how to integrate this with HAL. I was wondering if you could give me a guiding light on how should I be tackling this to make it work. 

I've tried using the websocket and telnet that exists on HAL but since the ESP only supports wi-fi I can't even change the settings related to Ethernet connection and also the HAL doesnt know how my W5500 is connected to the board and how to acess it. I have a working example with the SPI and maybe all I need is a way to integrate that to the HAL?

terjeio, 2022-06-23:
The websocket, telnet, http and ftp protocol implementations are on top of the lwIP tcp/ip stack (raw mode), if you are to use that code you have to get/implement a W5500 lwIP driver. I found [this thread](https://esp32.com/viewtopic.php?t=19018) when doing a quick search.

> I have a working example with the SPI and maybe all I need is a way to integrate that to the HAL?

If this provides the lwIP driver then it is a matter of initializing it in grblHAL?

> Or maybe at this point I should change microcontrollers, I don't know, what's your opnion on this?

It may well be easier to get a controller that has an ethernet phy built-in and is supported by the existing code such as the iMXRT1062 (Teensy 4.1).

Henrikastro, 2022-06-23:
Thank you for the reply, I'll definetly check the IwIP and see if I can get anything out of it, it seems fairly easy after setting the W5500 with it. I had checked the Teensy out but it's kinda expensive to have it in Brasil and the ESP with the W5500 is a much more inexpensive solution. Once again, thank you for the attention Terje!

closed 2022-06-23 by Henrikastro