grblHAL/ESP32 #31: Problem when using grblHAL with Fysetc E4 v1.0

7cnc, 2022-06-25:
I'm in need of some help!
When I compile GRBLHAL for the "Fysetc E4 v1.0" board and use IOSender to send a movement instruction to the X or Y axes, the two axes move simultaneously, I already checked the IO map but the error persists.
I'm making the necessary changes to CMakeList.txt. I even thought it was a defect on the board, but I compiled the GRBL_ESP32 version and the axes work normally with it.
I'm using GRBLHAL ESP32 clone from githabe compiling with Visual Studio Code.

terjeio, 2022-06-25:
Seems that I have messed up when testing by [leaving COREXY enabled](https://github.com/grblHAL/core/blob/f446881da4a85602928120308cf77985024d5bf4/config.h#L82). Sorry about that.

7cnc, 2022-06-25:
I changed #define COREXY as instructed and it worked.

After the change I noticed that the Y axis was not changing direction, after changing the option
OPTION(NOPROBE "Compile without probe support" ON) in CMakeLists.txt worked correctly. I don't know if this was the best way to solve the problem of not changing the direction of the Y axis but it worked. If you have another solution, please let me know. But if it's correct, you can close the topic as a resolved issue. Thanks for your help and I'm at your disposal.

closed 2022-07-14 by 7cnc