grblHAL/ESP32 #32: Invert Y Axis and A axis

7cnc, 2022-07-14:
Hello, I need a suggestion on how to create a function in settings to flip pins from Y axis to A axis.

Example: $499=0 (Normal mode)
#define Y_ENABLE_PIN           I2SO(4)
#define Y_STEP_PIN                I2SO(5)
#define Y_DIRECTION_PIN      I2SO(6)
#define Y_LIMIT_PIN GPIO_NUM_35
/ /Define ganged axis or A axis step pulse and step direction output pins.
#if N_ABC_MOTORS > 0
#define M3_AVAILABLE
#define M3_ENABLE_PIN             I2SO(7)
#define M3_STEP_PIN                  I2SO(8)
#define M3_DIRECTION_PIN        I2SO(9)
#endif

When changing the setting invert the pins
Example: $499=1 (Invert Y and A axis)
#define Y_ENABLE_PIN           I2SO(7)
#define Y_STEP_PIN                I2SO(8)
#define Y_DIRECTION_PIN      I2SO(9)
#define Y_LIMIT_PIN GPIO_NUM_35
/ /Define ganged axis or A axis step pulse and step direction output pins.
#if N_ABC_MOTORS > 0
#define M3_AVAILABLE
#define M3_ENABLE_PIN             I2SO(4)
#define M3_STEP_PIN                  I2SO(5)
#define M3_DIRECTION_PIN        I2SO(6)
#endif

In this way, using ioSender, I can use a probe on the rotating axis on cylindrical people, being able to map the probe.

terjeio, 2022-07-14:
An example for how to add settings can be found [here](https://github.com/grblHAL/Templates/blob/master/my_plugin/settings/my_plugin.c). However, would it be better to add a [M-code](https://github.com/grblHAL/Templates/blob/master/mcodes.c) for swapping or do you need a setting so that the current configuration can survive a cold start?

You have to modify existing functions in driver.c to swap pins.
E.g the step pins are set [here](https://github.com/grblHAL/ESP32/blob/66db555c6779d3e0a3d1d74809b83b4b6d1872d6/main/driver.c#L734-L759).

7cnc, 2022-07-14:
Another way that would solve my case would be to enable the surface probing of the A axis, where the X axis represents the width of an alliance or cylindrical piece and the A axis the circumference. But I don't know if I would need to change it in GrblHAL or in ioSender.

7cnc, 2022-07-14:
I made the map for the Makerbase MKS TinyBee board, and it is working perfectly, I will make it available so that other users can use the updated map.

terjeio, 2022-07-15:
> But I don't know if I would need to change it in GrblHAL or in ioSender.

The change has to be made in ioSender - not something I want to attempt without a machine with a rotary axis.

> ... and it is working perfectly

Great, you implemented it via a M-code?

7cnc, 2022-07-16:
It still didn't work!

I'll try to simplify what I need, because the way I tried to do it wasn't good.

I need to create parameter $499 (Invert Axis).

$499=0 (normal mode)
#define Y_ENABLE_PIN I2SO(4)
#define Y_STEP_PIN I2SO(5)
#define Y_DIRECTION_PIN I2SO(6)
#define Y_LIMIT_PIN GPIO_NUM_35

$499=1 (inverted mode)
#define Y_ENABLE_PIN I2SO(7)
#define Y_STEP_PIN I2SO(8)
#define Y_DIRECTION_PIN I2SO(9)
#define Y_LIMIT_PIN GPIO_NUM_35

I need that every time I change the $499 setting the grblHAL changes the pins without needing to restart the firmware.

7cnc, 2022-07-16:
> I made the map for the Makerbase MKS TinyBee board, and it is working perfectly, I will make it available so that other users can use the updated map.

mks_TinyBee_1_0_map.h

/*
  mks_TinyBee_1_0_map.h - An embedded CNC Controller with rs274/ngc (g-code) support

  Driver code for ESP32

  Part of grblHAL

  Copyright (c) 2022 Evan F. Gos

  Grbl is free software: you can redistribute it and/or modify
  it under the terms of the GNU General Public License as published by
  the Free Software Foundation, either version 3 of the License, or
  (at your option) any later version.

  Grbl is distributed in the hope that it will be useful,
  but WITHOUT ANY WARRANTY; without even the implied warranty of
  MERCHANTABILITY or FITNESS FOR A PARTICULAR PURPOSE.  See the
  GNU General Public License for more details.

  You should have received a copy of the GNU General Public License
  along with Grbl.  If not, see <http://www.gnu.org/licenses/>.
*/
#if N_ABC_MOTORS > 1
#error "Axis configuration is not supported!"
#endif

#if VFD_SPINDLE
#error "Board BOARD_MKS_TinyBee_V1.0 does not have support for VFD spindle."
#endif

#define BOARD_NAME " 7CNC / MKS-TinyBee V1.0 "

#define USE_I2S_OUT
#define USE_I2S_IN
#define I2S_OUT_PIN_BASE 64

// timer definitions
#define STEP_TIMER_GROUP TIMER_GROUP_0
#define STEP_TIMER_INDEX TIMER_0

#if SDCARD_ENABLE
// Pin mapping when using SPI mode.
// With this mapping, SD card can be used both in SPI and 1-line SD mode.
// Note that a pull-up on CS line is required in SD mode.
#define PIN_NUM_MISO GPIO_NUM_12
#define PIN_NUM_MOSI GPIO_NUM_13
#define PIN_NUM_CLK GPIO_NUM_14
#define PIN_NUM_CS GPIO_NUM_15
#endif // SDCARD_ENABLE

#define I2S_OUT_BCK GPIO_NUM_25
#define I2S_OUT_WS GPIO_NUM_26
#define I2S_OUT_DATA GPIO_NUM_27

#define X_ENABLE_PIN               I2SO(0)
#define X_STEP_PIN                 I2SO(1)
#define X_DIRECTION_PIN            I2SO(2)
#define X_LIMIT_PIN GPIO_NUM_33

#define Y_ENABLE_PIN               I2SO(3)
#define Y_STEP_PIN                 I2SO(4)
#define Y_DIRECTION_PIN            I2SO(5)
#define Y_LIMIT_PIN GPIO_NUM_32

#define Z_ENABLE_PIN               I2SO(6)
#define Z_STEP_PIN                 I2SO(7)
#define Z_DIRECTION_PIN            I2SO(8)
#define Z_LIMIT_PIN GPIO_NUM_22

// Define ganged axis or A axis step pulse and step direction output pins.
#if N_ABC_MOTORS > 0
#define M3_AVAILABLE
#define M3_ENABLE_PIN   	        I2SO(9)
#define M3_STEP_PIN               I2SO(10)
#define M3_DIRECTION_PIN          I2SO(11)
#endif

// Define ganged axis or B axis step pulse and step direction output pins.
#if N_ABC_MOTORS > 0
#define M4_AVAILABLE
#define M4_ENABLE_PIN   	        I2SO(12)
#define M4_STEP_PIN               I2SO(13)
#define M4_DIRECTION_PIN          I2SO(14)
#endif

// Define spindle enable and spindle direction output pins.

#define SPINDLE_ENABLE_PIN        I2SO(19)
#define SPINDLEPWMPIN             I2SO(20)

// Define flood and mist coolant enable output pins.

#define COOLANT_MIST_PIN         I2SO(7)
#define COOLANT_FLOOD_PIN GPIO_NUM_5

// Define user-control CONTROLs (cycle start, reset, feed hold) input pins.

// N/A

// Define probe switch input pin.
#if PROBE_ENABLE
#define PROBE_PIN GPIO_NUM_39
#endif

#if I2C_ENABLE
// Define I2C port/pins
#define I2C_SDA GPIO_NUM_0
#define I2C_SCL GPIO_NUM_4
#define I2C_CLOCK 100000
#endif

terjeio, 2022-07-18:
> It still didn't work!

You have to modify _driver.c_ to get it to work. Add a bool variable to control the pin swapping and a function to set it. Call the function either when the setting is changed or when a M-code is issued. IMO it is best to use a M-code as this does not write anything to flash...

closed 2023-01-08 by 7cnc