grblHAL/ESP32 #36: ESP32 Wroom + SDcard plugin error

SuksesAmin, 2022-07-25:
You have developed grblhal for people and you help them voluntarily, you are Hero!!.

I have a problem when using the sdcrad plugin. The error shown in the console is as below
`[MSG:Warning: Failed to initialize SD card]
Guru Meditation Error: Core  1 panic'ed (LoadProhibited). Exception was unhandled.
Core  1 register dump:
PC      : 0x400e7af0  PS      : 0x00060530  A0      : 0x800e4c64  A1      : 0x3ffb64b0  
A2      : 0x00000000  A3      : 0x3f404c8a  A4      : 0x00000000  A5      : 0x0000001f  
A6      : 0x00001004  A7      : 0x3ffaed3c  A8      : 0x00000000  A9      : 0x00000000  
A10     : 0x00000000  A11     : 0x3ffb1ca0  A12     : 0x3ffb1c54  A13     : 0x00000000  
A14     : 0x3ffb1ca0  A15     : 0x3ffb1c54  SAR     : 0x00000020  EXCCAUSE: 0x0000001c  
EXCVADDR: 0x00000000  LBEG    : 0x4000c46c  LEND    : 0x4000c477  LCOUNT  : 0xffffffff  
Backtrace:0x400e7aed:0x3ffb64b0 0x400e4c61:0x3ffb64d0 0x400e605c:0x3ffb6510 0x400e327c:0x3ffb6550 0x400e4293:0x3ffb65d0 0x400e42dc:0x3ffb65f0 0x400e1d71:0x3ffb6610 0x400db6f1:0x3ffb6660 0x400d5b7c:0x3ffb66a0 0x400d5883:0x3ffb66c0
ELF file SHA256: a20ced2169f81b78
CPU halted.
`

terjeio, 2022-07-25:
Change this line
https://github.com/grblHAL/ESP32/blob/6d08a106ff2ad1ab7911b6fec7f91c2c1b20a550/main/driver.c#L1816
to
`    if(card && fs) {`

and the guru should keep meditating.

If you get the _Failed to initialize SD card_ message when a formatted card is inserted then check your wiring as well.

SuksesAmin, 2022-07-25:
yes. now
[MSG:Warning: Failed to initialize SD card]
error:60 - SD Card mount failed.

I have tested, the wiring work well using this sketch

#include "FS.h"
#include "SD.h"
#define SD_CS 5
void listDir(fs::FS &fs, const char * dirname, uint8_t levels){
    Serial.printf("Listing directory: %s\n", dirname);
    File root = fs.open(dirname);
    if(!root){
        Serial.println("Failed to open directory");
        return;
    }
    if(!root.isDirectory()){
        Serial.println("Not a directory");
        return;
    }
    File file = root.openNextFile();
    while(file){
        if(file.isDirectory()){
            Serial.print("  DIR : ");
            Serial.println(file.name());
            if(levels){
                listDir(fs, file.name(), levels -1);
            }
        } else {
            Serial.print("  FILE: ");
            Serial.print(file.name());
            Serial.print("  SIZE: ");
            Serial.println(file.size());
        }
        file = root.openNextFile();
    }
}
void setup(){
  Serial.begin(115200);
  SD.begin(SD_CS);
  if (!SD.begin(SD_CS)) {
    Serial.println("failed");
    return;
  }
  uint8_t cardType = SD.cardType();
  if (cardType == CARD_NONE) {
    Serial.println("No sdcard");
    return;
  }
  Serial.println("Reading SD...");
  listDir(SD, "/", 0);
}
void loop(){
}

terjeio, 2022-07-26:
Your test program is using the same pins for communicating with the card as grblHAL does?
Which board map are you using - there could be pin conflicts?

SuksesAmin, 2022-07-26:
Yes, it is using the same pins. 8GB microsd and bdring v4

SuksesAmin, 2022-07-27:
I left all settings as default except CMakeLists.txt. I changed it on line 33 from OFF to ON

terjeio, 2022-07-27:
Which version of the SDK are you using? You can use `$I` to check:
```
...
[DRIVER OPTIONS:v4.3.1-dirty]
...
```

SuksesAmin, 2022-07-27:
[DRIVER OPTIONS:v4.3.2-dirty]

terjeio, 2022-07-27:
I have now tested with the same board map as you and it works for me. A bug/regression/breaking change in the IDF framework?
Anyway here is the binary that works for me:
[app-template.zip](https://github.com/grblHAL/ESP32/files/9200814/app-template.zip)

SuksesAmin, 2022-07-28:
I have tried compiling the sd_card example program from esp-idf which also doesn't work. I'm confused, it's working on arduino ide with a libraries that also made by Espressif.

SuksesAmin, 2022-07-28:
Now it works with grblhal after I removed the LVC125A IC and connected it directly to the eps32 pin. Still working with the Arduino IDE sketch above.

Hope this helps all of you who are having the same problem. Don't use the sdcard module that uses LVC125A IC for grblhal.

terjeio, 2022-07-28:
> after I removed the LVC125A IC

On which hardware? The schematic I have for for the V4.1 board do not show any pin buffering.

SuksesAmin, 2022-07-28:
[image]
this sdcard module that I use.

terjeio, 2022-07-28:
> this sdcard module that I use.

ok, so the buffer is on module - I have not seen that before. It needs 5V VCC? This since there is a regulator on the module, did you supply that?

> where can i get the schematic for v4.1?

I believe I found it [here](https://github.com/bdring/4_Axis_External_Driver).

terjeio, 2022-07-28:
Hmm - maybe this module is not suited for the ESP32 as, IIRC, it is not 5V tolerant. If supplied with 5V it will fry the ESP32 inputs?

SuksesAmin, 2022-07-28:
Yes, I'm supplying this with 5v connected to v-in pin of esp32. The AMS117 is 3.3V. so i think it's safe.

I also use this module for stm32f103. the miso, mosi, clk and cs pins are connected to the controller via the LVC12SA ic.

terjeio, 2022-07-28:
Most STM32F103 pins are 5v tolerant, not sure about the ESP32 - the MISO pin is an input to the MCU.

terjeio, 2022-09-17:
Closing as I've had hours upon hours of successful SD card streaming with the ESP32 recently.

closed 2022-09-17 by terjeio