grblHAL/ESP32 #40: [COMPILATION ISSUE]cannot compile latest git https://github.com/grblHAL/ESP32/commit/f91fe73a730cd7a15f173915c422f654a17bd5f5

luc-github, 2022-08-01:
I think some sanity check are missing 
if I compile with WiFi enabled but `OPTION(SDcard "SD Card Streaming" OFF)` (I forget to enable it)
```
../main/webui/commands.c: In function 'handle_job_status':
../main/webui/commands.c:1001:5: error: unknown type name 'sdcard_job_t'
     sdcard_job_t *job = sdcard_get_job_info();
     ^~~~~~~~~~~~
../main/webui/commands.c:1001:25: error: implicit declaration of function 'sdcard_get_job_info' [-Werror=implicit-function-declaration]
     sdcard_job_t *job = sdcard_get_job_info();
                         ^~~~~~~~~~~~~~~~~~~
../main/webui/commands.c:1001:25: warning: initialization of 'int *' from 'int' makes pointer from integer without a cast [-Wint-conversion]
../main/webui/commands.c:1044:77: error: request for member 'size' in something not a structure or union
                     ok &= !!cJSON_AddStringToObject(data, "total", uitoa(job->size));
                                                                             ^~
../main/webui/commands.c:1045:81: error: request for member 'pos' in something not a structure or union
                     ok &= !!cJSON_AddStringToObject(data, "processed", uitoa(job->pos));
                                                                                 ^~
../main/webui/commands.c:1047:70: error: request for member 'name' in something not a structure or union
                     ok &= !!cJSON_AddStringToObject(data, "name", job->name);
                                                                      ^~
../main/webui/commands.c: In function 'get_sd_status':
../main/webui/commands.c:1200:14: warning: unused variable 'refresh' [-Wunused-variable]
         bool refresh = !!get_arg(argc, argv, "REFRESH"), release = !!get_arg(argc, argv, "RELEASE");
              ^~~~~~~
../main/webui/commands.c:1208:64: error: implicit declaration of function 'sdcard_getfs'; did you mean 'state_get'? [-Werror=implicit-function-declaration]
         msg = hal.stream.type == StreamType_SDCard ? "Busy" : (sdcard_getfs() ? "SD card detected" : "Not available");
                                                                ^~~~~~~~~~~~
                                                                state_get
At top level:
../main/webui/commands.c:1229:22: warning: 'sd_print' defined but not used [-Wunused-function]
 static status_code_t sd_print (const struct webui_cmd_binding *command, uint_fast16_t argc, char **argv, bool json, bool isv3)
                      ^~~~~~~~
../main/webui/commands.c:1221:22: warning: 'get_sd_content' defined but not used [-Wunused-function]
 static status_code_t get_sd_content (const struct webui_cmd_binding *command, uint_fast16_t argc, char **argv, bool json, bool isv3)
                      ^~~~~~~~~~~~~~
../main/webui/commands.c:1194:22: warning: 'get_sd_status' defined but not used [-Wunused-function]
 static status_code_t get_sd_status (const struct webui_cmd_binding *command, uint_fast16_t argc, char **argv, bool json, bool isv3)
                      ^~~~~~~~~~~~~
cc1.exe: some warnings being treated as errors
ninja: build stopped: subcommand failed.
ninja failed with exit code 1
```

compilation still failed if `OPTION(SDcard "SD Card Streaming" ON)` due to authentication
if `OPTION(WebAuth "WebUI authentication" OFF)`

```
../main/web/backend.c:812:19: error: 'login_handler_get' undeclared here (not in a function); did you mean 'sighandler_t'?
       .handler  = login_handler_get,
                   ^~~~~~~~~~~~~~~~~
                   sighandler_t
../main/web/backend.c:817:19: error: 'login_handler_post' undeclared here (not in a function); did you mean 'basic_handlers'?
       .handler  = login_handler_post,
                   ^~~~~~~~~~~~~~~~~~
                   basic_handlers
[60/65] Building C object esp-idf/main/CMakeFiles/__idf_main.dir/webui/commands.c.obj
../main/webui/commands.c: In function 'get_sd_status':
../main/webui/commands.c:1200:14: warning: unused variable 'refresh' [-Wunused-variable]
         bool refresh = !!get_arg(argc, argv, "REFRESH"), release = !!get_arg(argc, argv, "RELEASE");
              ^~~~~~~
ninja: build stopped: subcommand failed.
ninja failed with exit code 1
```

which should be solved in backend.c by adding compilation directives :
```
#if WEBUI_AUTH_ENABLE
    { .uri      = "/login",
      .method   = HTTP_GET,
      .handler  = login_handler_get,
[... 50 lines cut ...]
cmd.exe /C "cd . && C:\Users\luc\.espressif\tools\xtensa-esp32-elf\esp-2021r2-patch3-8.4.0\xtensa-esp32-elf\bin\xtensa-esp32-elf-g++.exe  -mlongcalls -Wno-frame-address   @CMakeFiles\grbl.elf.rsp  -o grbl.elf  && cd ."
c:/users/luc/.espressif/tools/xtensa-esp32-elf/esp-2021r2-patch3-8.4.0/xtensa-esp32-elf/bin/../lib/gcc/xtensa-esp32-elf/8.4.0/../../../../xtensa-esp32-elf/bin/ld.exe: esp-idf/main/libmain.a(backend.c.obj):(.rodata.basic_handlers+0x28): undefined reference to `sdcard_handler'
c:/users/luc/.espressif/tools/xtensa-esp32-elf/esp-2021r2-patch3-8.4.0/xtensa-esp32-elf/bin/../lib/gcc/xtensa-esp32-elf/8.4.0/../../../../xtensa-esp32-elf/bin/ld.exe: esp-idf/main/libmain.a(backend.c.obj):(.rodata.basic_handlers+0x38): undefined reference to `sdcard_upload_handler'
c:/users/luc/.espressif/tools/xtensa-esp32-elf/esp-2021r2-patch3-8.4.0/xtensa-esp32-elf/bin/../lib/gcc/xtensa-esp32-elf/8.4.0/../../../../xtensa-esp32-elf/bin/ld.exe: esp-idf/main/libmain.a(backend.c.obj):(.rodata.basic_handlers+0x88): undefined reference to `sdcard_handler'
c:/users/luc/.espressif/tools/xtensa-esp32-elf/esp-2021r2-patch3-8.4.0/xtensa-esp32-elf/bin/../lib/gcc/xtensa-esp32-elf/8.4.0/../../../../xtensa-esp32-elf/bin/ld.exe: esp-idf/main/libmain.a(backend.c.obj):(.rodata.basic_handlers+0x98): undefined reference to `sdcard_upload_handler'
c:/users/luc/.espressif/tools/xtensa-esp32-elf/esp-2021r2-patch3-8.4.0/xtensa-esp32-elf/bin/../lib/gcc/xtensa-esp32-elf/8.4.0/../../../../xtensa-esp32-elf/bin/ld.exe: esp-idf/main/libmain.a(backend.c.obj):(.rodata.basic_handlers+0xf8): undefined reference to `login_handler_get'
c:/users/luc/.espressif/tools/xtensa-esp32-elf/esp-2021r2-patch3-8.4.0/xtensa-esp32-elf/bin/../lib/gcc/xtensa-esp32-elf/8.4.0/../../../../xtensa-esp32-elf/bin/ld.exe: esp-idf/main/libmain.a(backend.c.obj):(.rodata.basic_handlers+0x108): undefined reference to `login_handler_post'
c:/users/luc/.espressif/tools/xtensa-esp32-elf/esp-2021r2-patch3-8.4.0/xtensa-esp32-elf/bin/../lib/gcc/xtensa-esp32-elf/8.4.0/../../../../xtensa-esp32-elf/bin/ld.exe: esp-idf/main/libmain.a(server.c.obj):(.literal.is_authorized+0x14): undefined reference to `get_auth_level'
c:/users/luc/.espressif/tools/xtensa-esp32-elf/esp-2021r2-patch3-8.4.0/xtensa-esp32-elf/bin/../lib/gcc/xtensa-esp32-elf/8.4.0/../../../../xtensa-esp32-elf/bin/ld.exe: esp-idf/main/libmain.a(server.c.obj):(.literal.webui_init+0x14): undefined reference to `login_init'
c:/users/luc/.espressif/tools/xtensa-esp32-elf/esp-2021r2-patch3-8.4.0/xtensa-esp32-elf/bin/../lib/gcc/xtensa-esp32-elf/8.4.0/../../../../xtensa-esp32-elf/bin/ld.exe: esp-idf/main/libmain.a(server.c.obj): in function `is_authorized':
c:\users\luc\documents\esp\esp-idf\esp32\build/../main/esp_webui/server.c:156: undefined reference to `get_auth_level'
c:/users/luc/.espressif/tools/xtensa-esp32-elf/esp-2021r2-patch3-8.4.0/xtensa-esp32-elf/bin/../lib/gcc/xtensa-esp32-elf/8.4.0/../../../../xtensa-esp32-elf/bin/ld.exe: esp-idf/main/libmain.a(server.c.obj): in function `webui_http_command_handler':
c:\users\luc\documents\esp\esp-idf\esp32\build/../main/esp_webui/server.c:255: undefined reference to `get_auth_level'
c:/users/luc/.espressif/tools/xtensa-esp32-elf/esp-2021r2-patch3-8.4.0/xtensa-esp32-elf/bin/../lib/gcc/xtensa-esp32-elf/8.4.0/../../../../xtensa-esp32-elf/bin/ld.exe: c:\users\luc\documents\esp\esp-idf\esp32\build/../main/esp_webui/server.c:299: undefined reference to `get_auth_level'
c:/users/luc/.espressif/tools/xtensa-esp32-elf/esp-2021r2-patch3-8.4.0/xtensa-esp32-elf/bin/../lib/gcc/xtensa-esp32-elf/8.4.0/../../../../xtensa-esp32-elf/bin/ld.exe: esp-idf/main/libmain.a(server.c.obj): in function `webui_init':
c:\users\luc\documents\esp\esp-idf\esp32\build/../main/esp_webui/server.c:677: undefined reference to `login_init'
collect2.exe: error: ld returned 1 exit status
ninja: build stopped: subcommand failed.
ninja failed with exit code 1
```

sorry I did not digged more - let me know if you want me to - but I get you will be faster than me as I am not yet familliar with your code, may be a missing commit ?

I enclose my config File for reference
[CMakeLists.txt](https://github.com/grblHAL/ESP32/files/9230313/CMakeLists.txt)

terjeio, 2022-08-01:
I have just commited a quick update that should fix these issues. It takes a long time to recompile with different combinations of settings so I tend not to do that - errors may sneak in. One reason is that changes to the CMakeLists.txt casues the framework (1200 or so files) to be recompiled as well and this is slow...

> if I compile with WiFi enabled but OPTION(SDcard "SD Card Streaming" OFF) (I forget to enable it)

I now enable the SD card option if WebUI is enabled, I guess I should enable WiFi too.

> if OPTION(WebAuth "WebUI authentication" ON) there are another issues

This is due to the default compiler settings beeing very strict. I added a compiler flag to [CMakeLists.txt](https://github.com/grblHAL/ESP32/blob/9b774a3ae8c0e294970627986caac3a3c9aa4468/main/CMakeLists.txt#L391) to not treat the warning as an error. If you copied the CMakeLists.txt from the previous version this might be the reason.

luc-github, 2022-08-01:
Ok I will give a try, I always use CMakeLists.txt but use the provided one and do change according needs

luc-github, 2022-08-01:
Fixed - Thank you ^_^

closed 2022-08-01 by luc-github