grblHAL/ESP32 #42: Effect of speed and acceleration settings

solawc, 2022-08-15:
Hello, I am also a fan of grbl. After I saw the project of grblhal, I was very interested because I have done similar things in private. I want to ask, the speed and acceleration of the motor can be set. If I set the maximum speed to 20000, the acceleration to 6000, and the speed during slicing to 20000, can grblhal work normally?

terjeio, 2022-08-15:
> can grblhal work normally?

That depends on the steps/mm settings (not to exceed max step rate - which I have not profiled for this MCU) and your hardware beeing capable of handling the acceleration without stalling/missing steps.

The [iMXRT1062 driver](https://github.com/grblHAL/iMXRT1062) has the fastest MCU at 600MHz - I have managed to push that to 300KHz+ step rates and speeds > 50000 mm/min even when laser engraving short line segments. This with significantly increased planner buffer size.

terjeio, 2022-08-15:
What are the steps/mm settings (`$101-103`)?
Step pulse time (`$0`)?
Step pulse delay (`$29`)?

These are the basic parameters limiting max step rate. E.g. with $0 set to 10 (the default value) you will be limited to approx. 80KHz depending on the stepper driver specs (min. acceptable pulse off time).
Next is the MCU interrupt response time++. Then, for lots of short segments, the planner buffer size.

solawc, 2022-08-15:
> What are the steps/mm settings (`$101-103`)? Step pulse time (`$0`)? Step pulse delay (`$29`)?
>
> These are the basic parameters limiting max step rate. E.g. with $0 set to 10 (the default value) you [...]

the $101-$103 set 20000 all, and Step pulse time is 10 ($0)

solawc, 2022-08-15:
> What are the steps/mm settings (`$101-103`)? Step pulse time (`$0`)? Step pulse delay (`$29`)?
>
> These are the basic parameters limiting max step rate. E.g. with $0 set to 10 (the default value) you [...]

Oh, maybe I know the problem, let me check the code tomorrow, thanks for your help.

solawc, 2022-08-16:
In the code, I see that todo is written to handle S0. Is there a single line of S0 that can not be engraved normally?

terjeio, 2022-08-16:
> In the code, I see that todo is written to handle S0. Is there a single line of S0 that can not be engraved normally?

Nothing will be engraved when S0  is active (spindle speed = zero = laser off). BTW the TODO comment is related to CSS (constant surface speed mode) which is typically used for a lathe, not when engraving.

closed 2022-08-17 by solawc