grblHAL/ESP32 #46: Crash using modbus

JoachimF, 2022-11-24:
Hello Terjeio

I bought a  TinyBee v1.0 and I test it before mounting in my CNC, Wifi, Webui, SDcard, 5 axis, 3 limits + probes are OK.
Firmware downloaded yersterday, last IoSender

I enabled VFD, HUANYANG , the status message is ok on the TX2, I sniffed it, but when I send a command, M3S0 for example, the TX2 send the right message command to the VFD including the CRC, then the ESP32 crash "Core 0 panic'ed (InstrFetchProhibited)"

I haven't the logs files, because I'm not at home, i'll send you later.

have you a clue?

Also I need on this board the dual spindle, do you plan to add it?

Thanks for your work and your help

Joachim

JoachimF, 2022-11-25:
Here is the console :

**m3s0**
ALARM:14
Guru Meditation Error: Core  1 panic'ed (InstrFetchProhibited). Exception was unhandled.
Core  1 register dump:
PC      : 0x00000000  PS      : 0x00060e30  A0      : 0x800dd73b  A1      : 0x3ffb6560  
A2      : 0x3ffc2e30  A3      : 0x00000000  A4      : 0x3ffbf228  A5      : 0x00000011  
A6      : 0x00000030  A7      : 0x3ffc41a6  A8      : 0x800d860c  A9      : 0x00000000  
A10     : 0x3ffc2a3c  A11     : 0x00000000  A12     : 0x0001947d  A13     : 0x3ffc79c0  
A14     : 0x3ffc79cd  A15     : 0x3ffc7b14  SAR     : 0x00000004  EXCCAUSE: 0x00000014  
EXCVADDR: 0x00000000  LBEG    : 0x4000c2e0  LEND    : 0x4000c2f6  LCOUNT  : 0xffffffff  
Backtrace:0xfffffffd:0x3ffb6560 0x400dd738:0x3ffb6580 0x400dd9f2:0x3ffb65b0 0x400ddcd7:0x3ffb65d0 0x400d7d68:0x3ffb6610 0x400d7a67:0x3ffb6640
ELF file SHA256: 6921ac44f4be452f
CPU halted.

Here is my map : 
`#define BOARD_NAME "MKS Tinybee V1.0"

#define USE_I2S_OUT
#define I2S_OUT_PIN_BASE 64
#define I2S_SPINDLE
#define I2S_COOLANT

// timer definitions
#define STEP_TIMER_GROUP TIMER_GROUP_0
#define STEP_TIMER_INDEX TIMER_0

#define I2S_OUT_BCK     GPIO_NUM_25
#define I2S_OUT_WS      GPIO_NUM_26
#define I2S_OUT_DATA    GPIO_NUM_27

#define X_STEP_PIN      	I2SO(1) 
#define X_DIRECTION_PIN 	I2SO(2) 
#define X_ENABLE_PIN   		I2SO(0) 
#define X_LIMIT_PIN     GPIO_NUM_33

#define Y_STEP_PIN      	I2SO(4)
#define Y_DIRECTION_PIN 	I2SO(5)
#define Y_ENABLE_PIN   		I2SO(3)
#define Y_LIMIT_PIN     GPIO_NUM_32

#define Z_STEP_PIN      	I2SO(7)
#define Z_DIRECTION_PIN 	I2SO(8)
#define Z_ENABLE_PIN   		I2SO(6)
#define Z_LIMIT_PIN     GPIO_NUM_22

#if N_ABC_MOTORS >= 1
#define M3_AVAILABLE
#define M3_STEP_PIN      	I2SO(10)
#define M3_DIRECTION_PIN 	I2SO(11)
#define M3_ENABLE_PIN   	I2SO(9)
//#define M3_LIMIT_PIN     	GPIO_NUM_19
#endif

#if N_ABC_MOTORS >= 2
#define M4_AVAILABLE
#define M4_STEP_PIN      	I2SO(13)
#define M4_DIRECTION_PIN	I2SO(14)
#define M4_ENABLE_PIN   	I2SO(12)
//#define M4_LIMIT_PIN     	GPIO_NUM_18
#endif

// Define spindle enable and spindle direction output pins.
#if !VFD_SPINDLE
#define SPINDLEPWMPIN           GPIO_NUM_2
#define SPINDLE_ENABLE_PIN      I2SO(16)
#define SPINDLE_DIRECTION_PIN   I2SO(19)
#endif

[... 82 lines cut ...]
 Motor plugin options
 NOTE: SPI mode Trinamic drivers cannot currently be used with SD Card enabled!
OPTION(TMC2130 "Trinamic TMC2130 driver support" OFF)
OPTION(TMC2209 "Trinamic TMC2209 driver support" OFF)
OPTION(TMC2209_HWADDR "Trinamic TMC2209 drivers uses hardware addressing" OFF)
OPTION(TMC5160 "Trinamic TMC5160 driver support" OFF)
OPTION(Trinamic "Trinamic driver support via I2C" OFF)

 Spindle plugin options
OPTION(HUANYANG "Compile with Huanyang RS485 Spindle support" ON)
OPTION(RS485_DIR_OUT "Compile with RS485 direction pin enabled" OFF)
OPTION(DUAL_SPINDLE "Compile with dual spindle enabled" ON)

 EEPROM/FRAM plugin options
OPTION(EEPROM "Compile with I2C EEPROM support" OFF)
OPTION(FRAM "Compile with I2C FRAM support" OFF)

 Laser plugin options
OPTION(LBCLUSTER "Compile with LaserBurn cluster support" OFF)
OPTION(LASERCOOLANT "Compile with Laser coolant plugin support" OFF)

 Keypad plugin options
OPTION(Keypad "I2C Keypad" OFF)`

I hope this help

JoachimF, 2022-11-27:
Hello Terjeio,

I have a little bit of progress, if I plug the VFD, it works nice, I can drive the VFD. It's strange that the mcu crash when the VFD is not connected (same on a STM32F7Xx target).

I have succeded to use dual spindle, on PWM for hotwire and the VFD, but if I swap to VFD, the mcu crash. I setup using VFD on startup, but now it crash when a want to go back on PWM, also when I do $395=0

I am waiting for a JTAG debugger, so I can't do more.

If you have and idea?

Regards,

Joachim

JoachimF, 2022-12-04:
Hello Terjeio,

I have received the JTAG debugger and I think that I found why it crashed, here is the code correction :

in huanyang.c

```
#ifdef GRBL_ESP32
static void v1_spindleOff(void)
{
    v1_spindleUpdateRPM(0.0f);
}
#endif

#ifdef GRBL_ESP32
static void v2_spindleOff(void)
{
    v2_spindleUpdateRPM(0.0f);
}
#endif
```

```
#if VFD_ENABLE == SPINDLE_HUANYANG1 || VFD_ENABLE == SPINDLE_ALL
    static const vfd_spindle_ptrs_t v1_spindle = {
        .spindle.type = SpindleType_VFD,
        .spindle.cap.variable = On,
        .spindle.cap.at_speed = On,
        .spindle.cap.direction = On,
        .spindle.config = v1_spindle_config,
        .spindle.set_state = v1_spindleSetState,
        .spindle.get_state = v1_spindleGetState,
        .spindle.update_rpm = v1_spindleUpdateRPM,
#ifdef GRBL_ESP32
        .spindle.esp32_off = v1_spindleOff,            //!< Workaround handler for snowflake ESP32 Guru awaken by floating point data in ISR context.
#endif        
        .vfd.get_load = spindleGetLoad
    };
```
```
#if VFD_ENABLE == SPINDLE_HUANYANG2 || VFD_ENABLE == SPINDLE_ALL
    static const vfd_spindle_ptrs_t v2_spindle = {
        .spindle.type = SpindleType_VFD,
        .spindle.cap.variable = On,
        .spindle.cap.at_speed = On,
        .spindle.cap.direction = On,
        .spindle.config = v2_spindle_config,
        .spindle.set_state = v2_spindleSetState,
        .spindle.get_state = v2_spindleGetState,
#ifdef GRBL_ESP32
        .spindle.esp32_off = v2_spindleOff,            //!< Workaround handler for snowflake ESP32 Guru awaken by floating point data in ISR context.
#endif           
        .spindle.update_rpm = v2_spindleUpdateRPM
    };
```
Or you can adapt this code in gcode.c because esp32_off() doesn't exist in VFD spindle : 
```
void gc_spindle_off (void)
{
    gc_state.spindle.rpm = sys.spindle_rpm = 0.0f;
    gc_state.modal.spindle.value = 0;
#ifndef GRBL_ESP32
    hal.spindle.set_state(gc_state.modal.spindle, 0.0f);
#else
    hal.spindle.esp32_off();
#endif
    sys.report.spindle = On;
}
```

Regards,

Joachim

terjeio, 2022-12-05:
Try replacing [vfd_spindle_select()](https://github.com/grblHAL/Plugins_spindle/blob/6b70b5cfe2f3573b0cb33bd178e08c4e29b76353/vfd/spindle.c#L188-L205) with this code (should work with all VDF spindles):

```
#ifdef GRBL_ESP32
static void esp32_spindle_off (void)
{
    hal.spindle.set_state((spindle_state_t){0}, 0.0f);
}
#endif

static bool vfd_spindle_select (spindle_id_t spindle_id)
{
    uint_fast8_t idx = n_spindle;

    spindle_changed = true;
    memset(&vfd_spindle, 0, sizeof(vfd_ptrs_t));
    if(n_spindle) do {
        if(vfd_spindles[--idx].id == spindle_id) {
            memcpy(&vfd_spindle, &vfd_spindles[idx].vfd->vfd, sizeof(vfd_ptrs_t));
#ifdef GRBL_ESP32
            hal.spindle.esp32_off = esp32_spindle_off;
#endif
            break;
        }
    } while(idx);

    if(on_spindle_select)
        on_spindle_select(spindle_id);

    return true;
}
```

I am currently not able to test this code so I leave it to you.'

> Also I need on this board the dual spindle, do you plan to add it?

I do want to add multiple spindle support, LinuxCNC style. I have done some preliminary work, however a final version may take some time.

terjeio, 2023-01-28:
Fix above [committed](https://github.com/grblHAL/Plugins_spindle/commit/9868a3b70186b7dfb5ab3cb0e4b225afcd9956ff).

closed 2023-01-28 by terjeio