grblHAL/ESP32 #48: Multiple Spindle with VFD code correction

JoachimF, 2022-12-04:
In driver.c line 1899

PWM is not initialized when VFD is enable
```
#if (!VFD_SPINDLE || N_SPINDLE > 1) && defined(SPINDLE_ENABLE_PIN)
//#if VFD_SPINDLE != 1 && defined(SPINDLEPWMPIN)

```
If VFD is set by default, PWM is set to 100%, I don't know where to change that.
Is there a way to have PMW on I2S Pin?

Regards,

Joachim

terjeio, 2022-12-05:
> PWM is not initialized when VFD is enable

Good catch, will fix in next commit.

> If VFD is set by default, PWM is set to 100%, I don't know where to change that.

Set PWM as default and add [M104](https://github.com/grblHAL/Plugins_spindle/tree/6b70b5cfe2f3573b0cb33bd178e08c4e29b76353#readme) as a startup command ($Nx) to switch to the VFD?

> Is there a way to have PMW on I2S Pin?

It may be possible with code changes - but I will not be able to look into that for a while. You can give it a try?

terjeio, 2023-02-02:
> PWM is not initialized when VFD is enable

Set [N_SPINDLE](https://github.com/grblHAL/core/blob/e2ba6db69cda9d7ab664c3d4bbce0c1737b81768/config.h#L58) to 2 to have the PWM spindle enabled together with the VFD. In the[ Web Builder](http://svn.io-engineering.com:8080/) this can be done by checking _Add PWM spindle_ in the _General_ tab.

In an upcoming build I have added code to ensure the PWM spindle pins will be initialized to the off state on startup (if enabled).

terjeio, 2023-02-14:
The [latest commit](https://github.com/grblHAL/core/commit/04ce2c5c8c1129ed98e79b91e4041cd74cd6d08a) has code added for initializing the PWM output even if the PWM spindle is not the default. Handling of multiple spindles has been changed somewhat - see the [readme](https://github.com/grblHAL/Plugins_spindle#readme).

closed 2023-02-14 by terjeio