grblHAL/ESP32 #50: BlackBox X32 defaults / pin inversions

petervanderwalt, 2022-12-12:
Hi!  

So by default grblHAL has a couple pins inverted from our old standard (inhereted from Grbl days, existing machine profiles, users own grbl backups, etc), what is the best way to implement predefined settings defaults - that is compatible with Web Builder and rest of your vision? 
There are also a few parameters we change from default because the now have extended functionality - and it would be beneficial to use if they are preset in a new compile (Web Builder > Select BlackBox X32 > settings are already there when you start building your profile) 

We currently a function in CONTROL to set these for profile loads, backup restores, etc:

https://github.com/OpenBuilds/OpenBuilds-CONTROL/blob/5e3c66ac7275399580109d3e31c35c17420d38aa/app/js/grbl-settings-defaults.js#L4-L24

Basically just:
```
$10=511 ; we had $10=1 for Grbl, but we need 511 for the Troubleshooting tab in CONTROL to show the Pin States as people test endstops
$5=7 ; we have filtered switches and controller inputs, use NO switches
$6=7 ; our probe is Normally Open
#4=0 ; We had $4=1 in Grbl (the stepper drivers in BlackBox needed an inverted input). With grblHAL that "inversion" is already active by default, but default setting re-inverts it ($4=7 is the default I think) 
```

Is this something we can do from our pinmap? (if we have to do a PR) or if you don't mind, you can help us implement the options - we just don't know where to start to keep it compatible with the mainline and build tools etc without affecting other boards (:

terjeio, 2022-12-12:
> Is this something we can do from our pinmap? 

No. However it can be done from [CMakeLists.txt](https://github.com/grblHAL/ESP32/blob/ae61ef4e0280ff6f80d05cb76cabc6d2ba3eb259/main/CMakeLists.txt#L272-L273) or in the WebBuilder by adding preprocessor symbols that "override" the defaults. See [grbl/config.h](https://github.com/grblHAL/core/blob/d10804433c2b1060bb8ad6c5fa04218c2cf435a8/defaults.h) and [grbl/config.h](https://github.com/grblHAL/core/blob/d10804433c2b1060bb8ad6c5fa04218c2cf435a8/config.h).

On restoring a Grbl backup to grblHAL new $10 flags to be kept could be added and other extended settings modified to match the extended versions?

terjeio, 2023-01-26:
Override for some default settings values added to CMakeLists.txt:

https://github.com/grblHAL/ESP32/blob/f3cd99058897f7f32e90659065762b000befeb05/main/CMakeLists.txt#L274-L277

closed 2023-01-26 by terjeio

petervanderwalt, 2023-01-27:
Thank you! Any chance to have that work with Web Builder too?

terjeio, 2023-01-28:
It works for me, it does not for you?

petervanderwalt, 2023-01-30:
> However it can be done from [CMakeLists.txt](https://github.com/grblHAL/ESP32/blob/ae61ef4e0280ff6f80d05cb76cabc6d2ba3eb259/main/CMakeLists.txt#L272-L273) or in the WebBuilder

I interpreted the "OR" in between CMakeLists and WebBuilder as meaning seperate options.  I realize now that the options in CMakeLists.txt would work with WebBuilder already (: