grblHAL/ESP32 #51: probing not supported by driver or unsupported G38.x command

MickyPf, 2022-12-21:
Hi,

I own the OPENBUILDS Blackbax X32. Since I want to activate safety door, I need a new GRBL. The compilation worked great. Unfortunately I get the following error message "probing not supported by driver or unsupported G38.x command" when I start a probe. I ask for help.

terjeio, 2022-12-22:
Did you compile with an auto-squared axis? That assigns the probe input to the [second limit switch](https://github.com/grblHAL/ESP32/blob/ae61ef4e0280ff6f80d05cb76cabc6d2ba3eb259/main/BlackBoxX32_map.h#L56-L63).

MickyPf, 2022-12-22:
Hi,
thanks for the quick reply. I set Compile with ganged Y axis to on because I have two motors on the Y axis. Compile with auto-squared Y axis is off.

terjeio, 2022-12-22:
What is your `$pins` output?

MickyPf, 2022-12-22:
As intended by Openbuild's Blackbox X 32. The two Y endstops are connected together (soldered // two endstop cables become one) and this is plugged into the Y endstop socket.

terjeio, 2022-12-22:
> As intended by Openbuild's Blackbox X 32. 

Ok, so the probe pin is listed in the `$pins` output then. I'll try to figure out why it fails, but it may take some time as I am on holiday and do not have access to relevant hardware.

MickyPf, 2022-12-22:
OK, thanks for your help.

Can I help myself if I just comment out the area you showed me from here " #if N_AUTO_SQUARED".

 I wish you a Merry Christmas and a Happy New Year.

terjeio, 2022-12-22:
Since the probe pin is listed in the `$pins` output I assume the source code (driver.c) has somehow been corrupted. Changing the map file will not help as this correctly enables the pin, if it did not it would not show up in the `$pins` output.

MickyPf, 2022-12-22:
Thanks for the answer