grblHAL/ESP32 #52: Installation Instuctions

nsiatras, 2022-12-25:
I am tryingto upload the firmware to an ESP32 dev module.

Instructions says "Go into the **ESP32 directory** and modify settings in grbl/config.h and CMakeLists.txt as needed. Pin assignments and board specific config is in *_map.h files for each individual board."

**Where is the ESP32 directory located ?**

Thank you

terjeio, 2022-12-26:
> Where is the ESP32 directory located ?

In the directory where you downloaded/unpacked the source.

nsiatras, 2022-12-26:
I see you changed the instructions to "Go into the **ESP32/main** directory and modify settings in grbl/config.h" 

Makes sense now
Thank you

nsiatras, 2022-12-27:
I am using windows OS and I am trying to install the ESP32 grblHAL driver.
Has anyone ever installed it using Windows OS ?

drom89, 2022-12-27:
Im uploading the firmware to MCU by Visual Studio Code.

terjeio, 2022-12-28:
Development is done on a Win10 machine, usually I build and flash using Eclipse but occasinally flash with the idf.py command line utility. I have also flashed [Web Builder](http://svn.io-engineering.com:8080/) builds with Win10.

nsiatras, 2022-12-28:
@terjeio That Web Builder looks really nice !

Bravo!!!

nsiatras, 2022-12-29:
I am using the Flash Download Tool to upload the firmware to my ESP32-WROOM-32D chip.
The settings are as the following
[image]

Unfortunately this doesn't work for me.
Any ideas for different settings ?

nsiatras, 2022-12-29:
I managed to upload the firmware.
I use the SourceRabbit 4Axis CNC motherboard

If ound out that if if use 4 motors the X axis doesn't work
The problem is at the config file
https://github.com/grblHAL/ESP32/blob/master/main/sourcerabbit_4axis.h

The  
```
#if N_ABC_MOTORS > 0
#define M3_AVAILABLE
#define M3_STEP_PIN         GPIO_NUM_33
#define M3_DIRECTION_PIN    GPIO_NUM_12
#endif
```

Should be changed to

```
#if N_ABC_MOTORS > 0
#define M3_AVAILABLE
#define M3_STEP_PIN         GPIO_NUM_12
#define M3_DIRECTION_PIN    GPIO_NUM_13
#endif
```

terjeio, 2023-01-26:
Fixed in the latest commit.

closed 2023-01-26 by terjeio