grblHAL/ESP32 #53: Spindle Stops after using Feed Ovveride command

nsiatras, 2022-12-30:
I just found out that the Spindle Stops in the middle of the gcode cycle after I use feed override commands.

Tested it with the following control software:

- Focus (https://www.sourcerabbit.com/Shop/pr-i-91-t-focus-cnc-control-software.htm)
- Universal GCode Sender (https://winder.github.io/ugs_website/)

terjeio, 2022-12-31:
Works ok for me with ioSender and a STM32F756 board. Which board map are you using and is laser mode enabled?

nsiatras, 2022-12-31:
I am using an  ESP32-DevKitC-32D with the SourceRabbit 4-Axis CNC motherboard. 

SourceRabbit 4-Axis CNC motherboard works totally fine with the Grbl_ESP32. I changed to GrblHAL in order to check the backlash compensation.

terjeio, 2022-12-31:
It seems the pin assignments for the spindle is wrong, [this is the official mapping](https://github.com/SourceRabbit/GRBL_ESP32/blob/239a53cd1d2043a17684fb8b99822bea035ee8df/Grbl_Esp32/src/Machines/sourcerabbit_4axis_cnc_motherboard.h#L69-L70) (?) vs the [grblHAL mapping](https://github.com/grblHAL/ESP32/blob/575f315457e0775aa7a30cc56445d46fb10eb6d0/main/sourcerabbit_4axis.h#L74-L75). Can you try with fixing the grblHAL map?
However, I have no good explanation for why feedrate overrides should affect the spindle output - and I am not able to do any ESP32 debugging before I am back home at the end of January.

nsiatras, 2023-01-01:
Hello, happy new year !

Indeed, the pins are wrong. The correct pins can be found here
https://github.com/SourceRabbit/GRBL_ESP32/blob/main/Grbl_Esp32/src/Machines/sourcerabbit_Quantum.h
I used the correct pins for my tests.

Correction: **The speed (RPM ovveride) stops the spindle** and NOT the  feedrate overrides 

I earlier managed to implement the backlash compensation to our Grbl _ESP32 fork (https://github.com/SourceRabbit/GRBL_ESP32) so I really do not need to use GrblHAL at the moment.

In case you can take a look at the bug I will be happy to test it.

evangois, 2023-01-08:
> Parece que as atribuições de pinos para o fuso estão erradas, [este é o mapeamento oficial](https://github.com/SourceRabbit/GRBL_ESP32/blob/239a53cd1d2043a17684fb8b99822bea035ee8df/Grbl_Esp32/src/Machines/sourcerabbit_4axis_cnc_motherboard.h#L69-L70) (?) versus o [mapeamento grblHAL](https://github.com/grblHAL/ESP32/blob/575f315457e0775aa7a30cc56445d46fb10eb6d0/main/sourcerabbit_4axis.h#L74-L75) . Você pode tentar consertar o mapa grblHAL? No entanto, não tenho uma boa explicação de por que as substituições da taxa de avanço devem afetar a saída do fuso - e não posso fazer nenhuma depuração do ESP32 antes de voltar para casa no final de janeiro.

Hello, I'm using GRBLHAL with ioSender and it's been great, I recommend it to anyone who needs great performance.

Case I need help with the map of pins I'm available to adjust you.

terjeio, 2023-01-25:
I am not able to replicate this with a upcoming release, tested with ioSender, UGS and the SourceRabbit map on a bare dev board.
Do you remember if it was the spindle enable or PWM signal (or both) that went away?
Was _Laser mode_ enabled?

nsiatras, 2023-01-26:
Spindle is PWM, no laser enabled.