grblHAL/ESP32 #54: Wishlist for Web Builder:  Door Enable/Disable

petervanderwalt, 2023-01-02:
Hey!  On the BlackBox X32 we have a door input. 
Would love if it if Web Builder could allow users to build with Door enabled or Disabled (: 

We leave it OFF in our default builds, as it expects an NC switch - most of our users doesn't have a switch hooked up, leading to door alarm on power up

terjeio, 2023-01-02:
> Would love if it if Web Builder could allow users to build with Door enabled or Disabled (:

It is already available in the _General_ tab, _Safety door input_ check-box:

[image]

FYI I'll add URI parameters for linking to the web-builder that allows preselection of driver and board, and later on I might also add a tab/tabs for predefined machine profiles (overriding default values in grbl/defaults.h).

petervanderwalt, 2023-01-02:
> It is already available in the _General_ tab, _Safety door input_ check-box:

Oh my word, I am embarrassed I don't know how I missed it, Brain still on holiday I guess (: LOL! 

> FYI I'll add URI parameters for linking to the web-builder that allows preselection of driver and board, and later on I might also add a tab/tabs for predefined machine profiles (overriding default values in grbl/defaults.h).

Looking forward to that! URI parameters will make it nice for use to preselect items for users and just give them the link!

closed 2023-01-02 by petervanderwalt

terjeio, 2023-01-26:
URI parameters for driver and board selection has now been added. Try [this link](http://svn.io-engineering.com:8080/?driver=ESP32&board=BlackBox%20X32).

petervanderwalt, 2023-01-27:
Fantastic!  Well done Will give it some more extensive testing next week and get back to you!

terjeio, 2023-01-30:
> Now just wishing I could preselect other options for users (:

I could add a parameter to load a saved board from a server somewhere (if it allows cross site requests) as adding URI parameters for the whole UI will, IMO, make the code messy and hard to maintain.

And later I want to add machine profiles...

petervanderwalt, 2023-01-30:
> And later I want to add machine profiles...

Yeah in OpenBuildsCONTROL we have Profiles that sets the $$-settings for each of our machines. Apart from 4th axes vs 3-axes-dual-y we don't have much compile-time wise to set.  

It's more for the "I didn't get a kit" guys - they often want a little bit of firmware compiled customization like dual-x instead of dual-y; or enabling the not-enabled-by-default Door, or things like that, that we can't just do with $$ yet. 

Web Builder is a game changer in that finally I can direct users to say "go build your own firmware" and it really is easy! But if I could avoid having to tell them to also enable ganged-Y, and also set the Networking/WebUI, Enable SD on Plugins, etc - instead of just "enable safety door" they wanted - its a lot less to talk them through, and less they have to remember to do. 

> I could add a parameter to load a saved board from a server somewhere (if it allows cross site requests) as adding URI parameters for the whole UI will, IMO, make the code messy and hard to maintain.

Alternatively, would you be open to us not speficially having to use the grblHAL UI - but just using it as a backend, sending our payload with the options we need, grabbing the firmware, which then goes into our flashing tool for example. 

If so, we can do our GUI in CONTROL to let people configure custom build options, but in a way that we still set the defaults to avoid support on that. 

Or we could self-host if it helps with compute costs? Take the load of your server. Not sure if WebBuilder is open source? 

Sorry rambling here in between the Monday morning daily tasks but just putting some thoughts down.

terjeio, 2023-02-02:
> Alternatively, would you be open to us not speficially having to use the grblHAL UI - but just using it as a backend, sending our payload with the options we need, grabbing the firmware, which then goes into our flashing tool for example.

Possibly - I cannot stop you doing that as it is easy to figure out the format of the message sent to the backend. 

> Or we could self-host if it helps with compute costs? Take the load of your server.

Currently the server is a dedicated old desktop machine upstairs so yes that could help.

>  Not sure if WebBuilder is open source?

The WebBuilder is not open source - making it generally available may only add to my workload... However I can give access the code if you agree to not publish or share it.

petervanderwalt, 2023-02-02:
> The WebBuilder is not open source - making it generally available may only add to my workload... However I can give access the code if you agree to not publish or share it.
On the workload increase - I can imagine yes! 
Thanks for the offer, good to know it is a possibility - we'll strategise a little internally and get back to you privately if we need to go this route (: 

> Currently the server is a dedicated old desktop machine upstairs so yes that could help.
Noted! Will avoid abusing it! 

> Possibly - I cannot stop you doing that as it is easy to figure out the format of the message sent to the backend.
True, but we'd never do that without permission - that would just be bad on someone doing that!