grblHAL/ESP32 #56: ESP32 Crash

errolt, 2023-01-05:
Board: MKS-DLC32

I set something in the ESP3D UI(To test saving as it was giving errors), now it crashes at startup?
```
--- idf_monitor on /dev/ttyUSB0 115200 ---
--- Quit: Ctrl+] | Menu: Ctrl+T | Help: Ctrl+T followed by Ctrl+H ---
ets Jun  8 2016 00:22:57

rst:0x1 (POWERON_RESET),boot:0x1b (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:2
load:0x3fff0030,len:1504
load:0x40078000,len:12960
load:0x40080400,len:3020
0x40080400: _init at ??:?

entry 0x400805e4

GrblHAL 1.1f ['$' or '$HELP' for help]
ALARM:12
[MSG:Check Limits]

assert failed: xTimerGenericCommand timers.c:390 (xTimer)

Backtrace:0x400818b2:0x3ffb3d60 0x4008be95:0x3ffb3d80 0x40092a65:0x3ffb3da0 0x4008eb0d:0x3ffb3ec0 0x400e5749:0x3ffb3ef0 0x4008ed10:0x3ffb3f10
0x400818b2: panic_abort at /home/errol/esp/esp-idf-4.3/components/esp_system/panic.c:404

0x4008be95: esp_system_abort at /home/errol/esp/esp-idf-4.3/components/esp_system/system_api.c:112

0x40092a65: __assert_func at /home/errol/esp/esp-idf-4.3/components/newlib/assert.c:85

0x4008eb0d: xTimerGenericCommand at /home/errol/esp/esp-idf-4.3/components/freertos/timers.c:390
 (inlined by) xTimerGenericCommand at /home/errol/esp/esp-idf-4.3/components/freertos/timers.c:385

0x400e5749: vTimerCallback at /home/errol/git/ESP32/build/../main/driver.c:468

0x4008ed10: prvProcessReceivedCommands at /home/errol/esp/esp-idf-4.3/components/freertos/timers.c:781
 (inlined by) prvTimerTask at /home/errol/esp/esp-idf-4.3/components/freertos/timers.c:558

ELF file SHA256: 7dcd6de6653da1bc

CPU halted.

```

terjeio, 2023-01-05:
Do you remember what you changed that triggered this?

Looking at the code I believe critical section(s) should be added to `vTimerCallback()` and prehaps also `driver_delay_ms()` in _driver.c_. 

A quick fix is to erase the flash and reinstall?

errolt, 2023-01-05:
Was able to reproduce:
I enabled Hard Limit and Strict. After reset it keeps crashing.
[image]
Maybe related to https://github.com/luc-github/ESP3D-WEBUI/issues/290#issuecomment-1372163249?

terjeio, 2023-01-05:
Thanks, I will look into this when I am back home and able to debug. 

> Maybe related to https://github.com/luc-github/ESP3D-WEBUI/issues/290#issuecomment-1372163249?

I do not think so, but I cannot be sure since the crash seems to be timing related.

terjeio, 2023-01-26:
Can you retest this with the latest build? I have added critical sections around some code segments and I am not able to replicate the issue.

errolt, 2023-01-26:
Seems fixed.

closed 2023-01-26 by errolt