grblHAL/ESP32 #57: Slow file upload

mtk11, 2023-01-19:
I am experiencing slow file upload via the web interface – around 64 kB/s. Any way to increase the speed ?

terjeio, 2023-01-19:
IIRC I achieve around 300 Kb/s with my test setup. I'll retest when I am back home next week.

Which versions (grblHAL and WebUI) are you running and with wich board?

mtk11, 2023-01-19:
Board: modded version of [ESP32Controller R2](https://github.com/simon-jouet/ESP32Controller)

WebUI version: 3.0.0-a19.GH2
FW version: 1.1f-20221115
Browser version: Chrome 98
Free memory: 123.12 KB
CPU Frequency: 240 MHz
Filesystem type: SD
Filesystem usage: 1.86 GB/22.34 MB

terjeio, 2023-02-10:
I have retested a bit and found this:

2GB (Team brand) and 4GB (SanDisk HC) cards: upload ~140 Kb/s
16GB (Verbatim Premium HC 10): upload ~325 Kb/s

Could it be that your SD card is the limiting factor?

luc-github, 2023-02-10:
Signal strength is also a possible limitation

mtk11, 2023-02-11:
With 16 GB Transcend Premium, UHS Speed Class - Class 1 (U1), I had file transfer rates between 137-170.10 kB/s. I have tried 47K and 10K pullups. Had issues with large file upload. When tried to upload a 14.9 MB file, the upload failed few time, showing the file size - 0 B. I will try to test higher speed class SD cards later.

mtk11, 2023-03-17:
Tested few more microSD cards: 

- MediaRange, Class 10, 16 GB, recognized as 2.55 GB – average transfer speed 111 kB/s
- AData,  UHS U1, 64 GB – not recognized
- Kingstone, UHS U1, 64 GB – not recognized

terjeio, 2023-03-18:
> MediaRange, Class 10, 16 GB, recognized as 2.55 GB – average transfer speed 111 kB/s

The size info is wrong due to a bug (the `size_t` type was 32 bit, not 64 as required) will fix in the next build.

Here is the upload speed to my Verbatim 16GB card:

[image]

This is with WebDAV (using WinSCP) - can you try that too?

> AData, UHS U1, 64 GB – not recognized

FatFs do not supports cards > 32 GB? Due to different file system?

FYI for some reason FTP transfers makes the SD card go belly up on the ESP32 - I do not know why... Flaky ESP32 code somewhere? No such issue with the iMXRT1062 driver which uses the same grblHAL code for the protocol layers.