grblHAL/ESP32 #58: Probe pin on DLC32

errolt, 2023-01-24:
I'm trying to get Probing enabled on the DLC32. After compiling and uploading this is what $pins gives me:
```
[PIN:36,X limit min]
[PIN:35,Y limit min]
[PIN:34,Z limit min]
[PIN:27,Spindle on]
[PIN:5,Flood]
[PIN:71,Mist]
[PIN:66,X dir]
[PIN:70,Y dir]
[PIN:68,Z dir]
[PIN:15,SD card CS]
[PIN:34,RX,Primary UART]
[PIN:35,TX,Primary UART]
[PIN:32,Spindle PWM]
[PIN:14,SCK]
[PIN:13,MOSI]
[PIN:12,MISO]
```

Looking at $I I get this, looks like NOPROBE is set:
```
[VER:1.1f.20221115:]
[OPT:VNMSL,35,1024,3,0]
[AXS:3:XYZ]
[NEWOPT:ENUMS,RT+,NOPROBE,REBOOT,TC,SED,RTC,WIFI,FTP,SD]
[FIRMWARE:grblHAL]
[NVS STORAGE:*FLASH]
[DRIVER:ESP32]
[DRIVER VERSION:221205]
[DRIVER OPTIONS:v4.3.4-332-g56ee43ec5e]
[BOARD:MKS DLC32 2.0]
[WIFI MAC:C8:F0:9E:A1:C8:74]
[IP:0.0.0.0]
[PLUGIN:WebUI v0.13]
[PLUGIN:SDCARD v1.07]
ok
```

Digging through the code probing seems to be disabled here:
https://github.com/grblHAL/ESP32/blob/4ec36b75264ea119bb4c0fe19cca3c7f60043901/main/driver.h#L128

Am I missing a PROBE_ENABLE somewhere? I don't see a PROBE_ENABLE for any other board?

[CMakeLists.txt](https://github.com/grblHAL/ESP32/files/10486976/CMakeLists.txt)

errolt, 2023-01-25:
Must be something on my setup, as I'm the only one with this issue.

closed 2023-01-25 by errolt

terjeio, 2023-01-25:
> Must be something on my setup, as I'm the only one with this issue.

Perhaps not, I am just back home and tired from jet lag so give me some time to fix this...
It could well be due to a breaking change. BTW were you building with the Web Builder?

errolt, 2023-01-25:
No, building locally. 
If I build with the web builder then the probe pin works, but the web builder has an option to **Enable** the probe pin, while the CMakeList.txt has an option to **Disable** the probe pin(NOPROBE). 

Looking at the code there seems to be two #defines that "affect" probe availability.

Defining PROBE_ENABLE will enable the probe, and this is not defined by default.
Defining NOPROBE will "disable" the probe, but this does not seem to be referenced anywhere in the code.

Changing CMakeList.txt to this:
```
if(NOPROBE)
target_compile_definitions("${COMPONENT_LIB}" PUBLIC NOPROBE)
else()
target_compile_definitions("${COMPONENT_LIB}" PUBLIC PROBE_ENABLE)
endif()
```

does fix the issue for me, but I suspect that NOPROBE can be removed completely, and just PROBE_ENABLE used.

terjeio, 2023-01-26:
The latest commit fixes this issue.

errolt, 2023-01-26:
Works. Thanks.