grblHAL/ESP32 #59: Tool change = "Manual Touch off" causes Error 46

errolt, 2023-01-24:
My machine doesn't have end stops and firmware is compiled with homing disabled.
I need to manually zero the tool after tool change, hence "Manual Touch Off".

But when I try to run an NC file I get error 46, "Home machine to continue.". If I set Tool Change back to Normal, then I don't get Error 46.

Anyway around this? Or must I add limit switches to be able to manually zero a tool after tool change?

errolt, 2023-01-25:
Found a sender that solves this

closed 2023-01-25 by errolt

terjeio, 2023-01-26:
> Found a sender that solves this

Which one?

FYI _Manual Touch Off_ requires at least the Z-axis to be homed, grblHAL allows manual homing by moving the spindle to the home position and issuing the `$H` command there if configured so:

[image]

errolt, 2023-01-26:
> Which one?

gSender. I have not looked at the data stream, but it allows me to jog and manually zero after an M6 command, but before resuming the program. I suspect it might intercept the M6, or handle it somehow. Anyway, with gSender I can leave Tool Change = Normal, and gSender handles the zeroing.

I have not found a sender that handles the TOOL state and allows entering jog mode during TOOL state as described here:
https://github.com/grblHAL/core/wiki/Manual-tool-change-protocol