grblHAL/ESP32 #62: Spindle sync

MsnSazabi, 2023-03-18:
Is it possible to read spindle rpm on esp32 based controller? Last info available dated 2020, maybe any progress since than. I have both: stock hall and quadrature encoder installed on my lathe. Better to read encoder of course.

terjeio, 2023-03-19:
No, somebody has to write and debug supporting code for that.

I have no plans to implement spindle sync for the ESP32 myself since the resident meditating guru shuts down the processor on the first sniff of a floating point instruction in an interrupt context (fixed point calculations could be used instead for the PID controller). Limited number of pins and no real debug capability also puts me off implementing spindle sync for this processor. 
iMXRT1062, STM32F4xx and MSP432P401R are the drivers that currently has experimental spindle sync support.

MsnSazabi, 2023-03-19:
Thank. Stm32 works with quadrature encoder or hall?

terjeio, 2023-03-19:
The F4xx driver should work with both - but not yet with full support for quadrature encoders. Note for spindle sync an index pulse (one per revolution) is required as well.

MsnSazabi, 2023-03-19:
So encoder + hall or hall+hall? Any detailed info on this topic to read?

terjeio, 2023-03-19:
You can use the quadrature encoder for the main pulse input, but only one phase. The hall sensor can be used for the index pulse if it delivers one per rev.

> Any detailed info on this topic to read?

I have not made any - you may find some if you search for _linuxcnc spindle sync_. grblHAL uses a PID control loop for spindle sync that has to be tuned via settings. FYI [ioSender](https://github.com/terjeio/ioSender) has a tool for plotting the response (actual position vs programmed), I use this in combination with G33 commands to tune the PID parameters. The tool is enabled when [PID logging is enabled in the controller](https://github.com/grblHAL/core/blob/cad1dc8476b9bdc8e751d2535dae7019f701a835/config.h#L499-L500).

closed 2023-08-26 by terjeio