grblHAL/ESP32 #65: Compile with vscode win 10.

RDS356, 2023-03-30:
Hi. Is there a tutorial to compile using vscode on windows ? I didn't find anything that can help me.

terjeio, 2023-03-30:
[A search returns plenty of hits](https://duckduckgo.com/?t=ffab&q=esp32+vscode&ia=web).

However, if you do not need to change code and/or a map file then using the [Web Builder](http://svn.io-engineering.com:8080/) is the easiest way to create a binary.

RDS356, 2023-03-30:
I have my own board, actually I'm using fluidnc, but I could see that grblhal is much more robust. I'm not advanced user regarding low level programing. Unfortunately I will leave grblhall by  this. Sucees for all.
Below a picture of the my board.

[image]

closed 2023-04-16 by terjeio