grblHAL/ESP32 #68: Reduced Stepper Rate

amctgt, 2023-04-03:
I need to use grblHAL on a BlackBox X32 to control the motion of a workpiece. My process requires very slow movement rates on the order of 1mm/hour. I attempted to edit the config file to reduce the MINIMUM_FEED_RATE from 1.0f to 0.001f. I then changed the cMakeLists.txt file so that "OPTION(BOARD_BLACKBOX_X32 "Compile for BlackBox X32" ON)" was "ON", rebuilt the project and flashed it to my BlackBox X32. When I reset it, I get a repeated error
[image]
It eventually says "ok" then I can send a move command, it will execute but then return "ok" several times followed by the "GrblHAL 1.1f ['$' or '$HELP' for help]" error code several times. My two main questions are:
1. How can I rebuild the code without causing all of these errors?
2. Is this the best way to go about reducing the minimum speed?

I am running Windows 11 and using Eclipse IDE for Embedded C/C++ Developers (includes Incubating components) Version: 2021-09 (4.21.0) Build id: 20210910-1417 and built and flashed using ESP-IDF 4.4 command prompt.

terjeio, 2023-04-04:
1. By using [ESP-IDF v4.3](https://github.com/grblHAL/ESP32#how-to-build-using-esp-idf-v43)? I have no idea what might cause the repeated messages except perhaps some breaking changes in the IDF framework. Try this binary:

[app-template.zip](https://github.com/grblHAL/ESP32/files/11144488/app-template.zip)

2. Minimum feedrate is also [capped by code in driver.c](https://github.com/grblHAL/ESP32/blob/8f4f19795e9e86df75bb53326de6747e774b21e9/main/driver.c#L569-L571). This has to be changed as well - taking into account your step/mm setting as the capping was added to get rid of some, seemingly random, very long step intervals.
Note that the attached binary has the cap removed.

terjeio, 2023-04-05:
>  what exactly did you change to remove the cap?

`    TIMERG0.hw_timer[STEP_TIMER_INDEX].alarm_low = cycles_per_tick;`

closed 2023-04-16 by terjeio