grblHAL/ESP32 #69: Cannot download any file from local FS but preferences.json

luc-github, 2023-04-09:
It seems on ESP32 files can be uploaded on local FS, but no files but preferences.json can be downloaded
Like the webui theme files.
  
[image]

I know on other platform the FS is limited and that is why we introduced the `HostPath:"/"` parameter in [ESP800], should we add a new parameter `ReadOnlyFS` to let know webui to disable the theme features  and any FS upload

In same idea, it seems webui (index.html.gz) can be updated but not downloaded so it is weird - I did not digged grblHAL code to be honest 

Original issue was raised here : https://github.com/luc-github/ESP3D-WEBUI/issues/350

terjeio, 2023-04-09:
There are several issues at play here.
First, I do not set `HostPath` as I should, [code is commented out](https://github.com/grblHAL/Plugin_WebUI/blob/d18a9cd7fbea02b22655e37f47e1a5c2905b1a57/commands_v3.c#L1499-L1502) for that - cannot remember why now.

When removing the commenting and fixing some related issues things improves.

Saving files from _Flash_ still does not work, due to `HostPath` not beeing prepended to the filename:

[image]

However, saving from the maintenance page works:

[image]

Also the dark theme does nothing even if it is loaded, due to it beeing gzipped? The purple theme works as does the dark theme if I download it (from the controller to be sure it has not been corrupted), unzip and upload it again.

[image]

[image]

luc-github, 2023-04-09:
if gz theme does not work it is may be due to fact it is gzziped yes, 
the webui only request the file name  `theme-dark` and ESP3D web server check if the file exists, or if  `theme-dark.gz` exists, in this case ( `theme-dark.gz` exists) it add `Content-Encoding:gzip` header before streaming it, it is same trick like for index.html.gz to save space
 Sorry I did not thinked that may not be implemented on your webserver
 
 The dark theme should give this ui as I tested for the issue on test server and fixed the grblHAL logo not visible on it
[image]

>Saving files from Flash still does not work, due to HostPath not beeing prepended to the filename:

That is weird I can upload files on FS on current FW
[image]

Do you need me to test something ?