grblHAL/ESP32 #70: All io stop work when I2S is actived.

RDS356, 2023-04-16:
I'm trying to use my self made board with GRBLhall. This board works perfectly in grbl_esp32, Fluidnc, etc.
My problem is, when I active the I2S(with shift register) all IO not work more. Look my config, i commented the lines(*) and all ports like M8, M7, M3, stepper enable works , but if I uncommented  no one works.

/*
  generic_map.h - driver code for ESP32

  Part of grblHAL

  Copyright (c) 2020-2023 Terje Io

  Grbl is free software: you can redistribute it and/or modify
  it under the terms of the GNU General Public License as published by
  the Free Software Foundation, either version 3 of the License, or
  (at your option) any later version.

  Grbl is distributed in the hope that it will be useful,
  but WITHOUT ANY WARRANTY; without even the implied warranty of
  MERCHANTABILITY or FITNESS FOR A PARTICULAR PURPOSE.  See the
  GNU General Public License for more details.

  You should have received a copy of the GNU General Public License
  along with Grbl.  If not, see <http://www.gnu.org/licenses/>.
*/
#define BOARD_NAME "RDS_ESP32"
#define BOARD_URL "https://rdscnc.com.br"

#if VFD_SPINDLE
#error "Generic map does not have support for VFD spindle."
#endif

//#define USE_I2S_OUT *
//#define I2S_OUT_PIN_BASE 64 *
// timer definitions
#define STEP_TIMER_GROUP TIMER_GROUP_0
#define STEP_TIMER_INDEX TIMER_0

// Define step pulse output pin27
#define X_STEP_PIN              GPIO_NUM_13//I2SO(1)
#define Y_STEP_PIN              GPIO_NUM_12//I2SO(3)
#define Z_STEP_PIN              GPIO_NUM_14//I2SO(5)

// Define step direction output pins. NOTE: All direction pins must be on the same port.
#define X_DIRECTION_PIN         GPIO_NUM_13//I2SO(2)
#define Y_DIRECTION_PIN         GPIO_NUM_12//I2SO(4)
#define Z_DIRECTION_PIN         GPIO_NUM_14//I2SO(6)

// Define stepper driver enable/disable output pin(s).
#define STEPPERS_ENABLE_PIN     GPIO_NUM_27

// Define homing/hard limit switch input pins and limit interrupt vectors.
#define X_LIMIT_PIN             GPIO_NUM_33
#define Y_LIMIT_PIN             GPIO_NUM_32
#define Z_LIMIT_PIN             GPIO_NUM_35

#if N_ABC_MOTORS >= 1
#define M3_AVAILABLE
#define M3_STEP_PIN       GPIO_NUM_12//I2SO(7)
#define M3_DIRECTION_PIN  GPIO_NUM_13//I2SO(0)
#define M3_LIMIT_PIN      GPIO_NUM_34
#endif

//#define I2S_OUT_BCK         GPIO_NUM_13 *
//#define I2S_OUT_WS          GPIO_NUM_12 *
//#define I2S_OUT_DATA        GPIO_NUM_14 *

#if DRIVER_SPINDLE_ENABLE
// Define spindle enable and spindle direction output pins.
#define SPINDLE_ENABLE_PIN      GPIO_NUM_4
#define SPINDLEPWMPIN           GPIO_NUM_2
#endif

// Define flood and mist coolant enable output pins.
#define COOLANT_FLOOD_PIN       GPIO_NUM_15
#define COOLANT_MIST_PIN        GPIO_NUM_21

// Define user-control CONTROLs (cycle start, reset, feed hold) input pins.
//#define RESET_PIN               GPIO_NUM_34
//#define FEED_HOLD_PIN           GPIO_NUM_36
//#define CYCLE_START_PIN         GPIO_NUM_39
#if SAFETY_DOOR_ENABLE
 #define SAFETY_DOOR_PIN         GPIO_NUM_25
#endif

// Define probe switch input pin.
#if PROBE_ENABLE
#define PROBE_PIN               GPIO_NUM_26
#endif

#if KEYPAD_ENABLE
#error No free pins for keypad!
#endif

#if SDCARD_ENABLE
// Pin mapping when using SPI mode.
// With this mapping, SD card can be used both in SPI and 1-line SD mode.
// Note that a pull-up on CS line is required in SD mode.
#define PIN_NUM_MISO            GPIO_NUM_19
#define PIN_NUM_MOSI            GPIO_NUM_23
#define PIN_NUM_CLK             GPIO_NUM_18
#define PIN_NUM_CS              GPIO_NUM_5
#endif

terjeio, 2023-04-16:
Default is to use I2S for all output signals. Spindle and coolant signals can be switched to normal GPIO, see [here](https://github.com/grblHAL/ESP32/blob/f6f089a125958069514a12d07147c60aa511b96b/main/mks_tinybee_1_0_map.h#L30-L31).

I have not made it possible to use normal GPIO for motor enable signals, this could be added to driver.c in the same way as for spindle and [coolant](https://github.com/grblHAL/ESP32/blob/f6f089a125958069514a12d07147c60aa511b96b/main/driver.c#L1327-L1381).

RDS356, 2023-04-16:
Unfortunately doesn't work. when I compile with I2S the GPIO.15 stay ever high, regardless if this GPIO.15 is configured or not.

terjeio, 2023-04-16:
Try with the latest version and these #defines in your map file:

```
#define I2S_SPINDLE 0
#define I2S_COOLANT 0
#define I2S_STEPPER_ENA 0
```

RDS356, 2023-04-17:
I tried like you said, but still the same problem. When I associet these I2S ports and compile the GPIO.15 go to high and nothing more work. Replacing the I2S ports by GPIO_NUM_ all ports GPIO work.(M3, M7, M8,INPUTS).

RDS356, 2023-04-17:
/*
    Driver code for ESP32

  Part of grblHAL

  Copyright (c) 2020-2023 Terje Io

  Grbl is free software: you can redistribute it and/or modify
  it under the terms of the GNU General Public License as published by
  the Free Software Foundation, either version 3 of the License, or
  (at your option) any later version.

  Grbl is distributed in the hope that it will be useful,
  but WITHOUT ANY WARRANTY; without even the implied warranty of
  MERCHANTABILITY or FITNESS FOR A PARTICULAR PURPOSE.  See the
  GNU General Public License for more details.

  You should have received a copy of the GNU General Public License
  along with Grbl.  If not, see <http://www.gnu.org/licenses/>.
*/

#if VFD_SPINDLE
#error "Board RDS does not have support for VFD spindle."
#endif

#define BOARD_NAME "RDS"
#if VFD_SPINDLE
 #error "Generic map does not have support for VFD spindle."
#endif

#define USE_I2S_OUT
#define I2S_OUT_PIN_BASE 64
#define I2S_SPINDLE 0
#define I2S_COOLANT 0
#define I2S_STEPPERS_ENA 0

#define I2S_OUT_BCK GPIO_NUM_13
#define I2S_OUT_WS GPIO_NUM_12
#define I2S_OUT_DATA GPIO_NUM_14

// timer definitions
#define STEP_TIMER_GROUP TIMER_GROUP_0
#define STEP_TIMER_INDEX TIMER_0

// Define step pulse output pin27
#define X_STEP_PIN I2SO(1)
#define Y_STEP_PIN I2SO(3)
#define Z_STEP_PIN I2SO(5)

// Define step direction output pins. NOTE: All direction pins must be on the same port.
#define X_DIRECTION_PIN I2SO(2)
#define Y_DIRECTION_PIN I2SO(4)
#define Z_DIRECTION_PIN I2SO(6)

// Define stepper driver enable/disable output pin(s).
#define STEPPERS_ENABLE_PIN GPIO_NUM_27

// Define homing/hard limit switch input pins and limit interrupt vectors.
#define X_LIMIT_PIN GPIO_NUM_33
#define Y_LIMIT_PIN GPIO_NUM_32
#define Z_LIMIT_PIN GPIO_NUM_35

#if N_ABC_MOTORS >= 1
 #define M3_AVAILABLE
 #define M3_STEP_PIN I2SO(7)
 #define M3_DIRECTION_PIN I2SO(0)
 #define M3_LIMIT_PIN GPIO_NUM_34
#endif

#if DRIVER_SPINDLE_ENABLE
// Define spindle enable and spindle direction output pins.
#define SPINDLE_ENABLE_PIN GPIO_NUM_4
#define SPINDLEPWMPIN GPIO_NUM_2
#endif

// Define flood and mist coolant enable output pins.
#define COOLANT_FLOOD_PIN GPIO_NUM_15
#define COOLANT_MIST_PIN GPIO_NUM_21

// Define user-control CONTROLs (cycle start, reset, feed hold) input pins.
//#define RESET_PIN GPIO_NUM_34
//#define FEED_HOLD_PIN GPIO_NUM_36
//#define CYCLE_START_PIN GPIO_NUM_39
#if SAFETY_DOOR_ENABLE
#define SAFETY_DOOR_PIN GPIO_NUM_25
#endif

// Define probe switch input pin.
#if PROBE_ENABLE
#define PROBE_PIN GPIO_NUM_26
#endif

#if KEYPAD_ENABLE
#error No free pins for keypad!
#endif

#if SDCARD_ENABLE
// Pin mapping when using SPI mode.
// With this mapping, SD card can be used both in SPI and 1-line SD mode.
// Note that a pull-up on CS line is required in SD mode.
#define PIN_NUM_MISO GPIO_NUM_19
#define PIN_NUM_MOSI GPIO_NUM_23
#define PIN_NUM_CLK GPIO_NUM_18
#define PIN_NUM_CS GPIO_NUM_5
#endif

RDS356, 2023-04-17:
My board;

[image]

terjeio, 2023-04-18:
Did you update to the latest version? `$I` should report driver version 230416.

RDS356, 2023-04-18:
Yes, I did.

$i
[VER:1.1f.20230411:]
[OPT:VNMSL,35,1024,4,0]
[AXS:4:XYZA]
[NEWOPT:ENUMS,RT+,REBOOT,TC,SED,RTC]
[FIRMWARE:grblHAL]
[NVS STORAGE:*FLASH]
[DRIVER:ESP32]
[DRIVER VERSION:230416]
[DRIVER OPTIONS:v4.3.5]
[BOARD:RDS_ESP32]

RDS356, 2023-04-18:
I updated  merging these files manually;
        main/CMakeLists.txt
	main/driver.c
	main/driver.h
	main/embedded/index.html.gz
	main/mks_tinybee_1_0_map.h

On time, what this line means ?
**#define I2S_OUT_PIN_BASE 64**

terjeio, 2023-04-18:
Try this driver:

[driver.zip](https://github.com/grblHAL/ESP32/files/11264079/driver.zip)

> On time, what this line means ?
#define I2S_OUT_PIN_BASE 64

It is a pin number offset to distinguish I2S pins from GPIO pins.

RDS356, 2023-04-20:
> Try this driver:
> 
> [driver.zip](https://github.com/grblHAL/ESP32/files/11264079/driver.zip)
> 
> > On time, what this line means ?
> > #define I2S_OUT_PIN_BASE 64
> 
> It is a pin number offset to distinguish I2S pins from GPIO pins.

Hi..
Work fine.
Thank you very much.
Best Regards,
Reinaldo de Souza

closed 2023-05-24 by terjeio