grblHAL/ESP32 #76: 4th Axis (Rotary)

ronenoosthuizen, 2023-06-20:
Hello,

We are currently using the repository with an ESP32. We are using it with 3 stepper motor to specify movement of an object in the X,Y,Z directions. We are looking to add a 4th motor that will enable the object rotate about the Z-axis. How can I set this up, if possible?

terjeio, 2023-06-20:
You will need to select or create a map file that supports at least four axes. In the [Web Builder](http://svn.io-engineering.com:8080) it is easy to create a binary by selecting four axis build, if compiling locally [grbl/config.h](https://github.com/grblHAL/core/blob/8c7f84184028fc8ad74baef5a5fd43bfbcbd12ad/config.h#L37-L43) has to be changed.

The fourth axis (A)  can be set up as a rotary (this disables the inches to mm transform) by setting $376=1.
Note that there is an option to "fix" a [feed rate issue](https://github.com/grblHAL/core/discussions/241) that may cause slow movements when both linear and rotary motion is commanded at the same time. The need to apply this fix depends on the output from the CAM program. It can be activated by a checking a box in the Web Builder or by changing [this symbol](https://github.com/grblHAL/core/blob/8c7f84184028fc8ad74baef5a5fd43bfbcbd12ad/planner.c#L34-L36) to 1.

ronenoosthuizen, 2023-06-21:
> You will need to select or create a map file that supports at least four axes. In the [Web Builder](http://svn.io-engineering.com:8080) it is easy to create a binary by selecting four axis build, if [...]

[image]
[image]
[image]

So I believe I created the Map File according to what is seen in "binary" above.
That left me with the files seen in "firmware" above.
If I want the axis to be rotary do I have to change the 0 to 1 in "$376" above?

What would the next steps look like?

ronenoosthuizen, 2023-06-22:
https://github.com/grblHAL/ESP32/issues/68#issue-1652910567

We are attempting to add this rotary axis to the same system referenced in the issue above. This brings up 2 questions:

1. Are there any other changes to the code required to run the system?
2. In the referenced issue above, we were never able to successfully compile the code you sent locally - any advice on avoiding these rebuild errors?

petervanderwalt, 2023-06-22:
As you have selected the BlackBox, does this pertain the OpenBuilds BlackBox? If so, 4th axes firmware is made available to you in OpenBuildsCONTROL (https://software.openbuilds.com) > Wizards and tools > Firmware flashing tool already

terjeio, 2023-06-22:
> If I want the axis to be rotary do I have to change the 0 to 1 in "$376" above?

No, that is just a default value. $376 is a run time setting that overrides the default, send `$376=1` from a terminal (e.g. puTTY) or a sender to the controller to set it.

> What would the next steps look like?

Have you sucessfully [programmed the processor](https://github.com/grblHAL/core/wiki/Compiling-GrblHAL) with the files provided by the Web Builder? 

> Are there any other changes to the code required to run the system?

No.

> In the referenced issue above, we were never able to successfully compile the code you sent locally - any advice on avoiding these rebuild errors?

Which errors?

> ...since we have issues compiling it locally, would you be willing to compile it for us?

I see no need for that if you can use the binary provided by the Web Builder. If I compile for you I will just send you the same files.

You can also use OpenBuilds Control sender to program the MCU even if it is not a BlackBox X32 but a "bare" board. You will then have to wire up the board to match the [BlackBox X32 map](https://github.com/grblHAL/ESP32/blob/master/main/BlackBoxX32_map.h).