grblHAL/ESP32 #78: QUESTION? what sender is needed for embroidery mode?

Ams-Laser, 2023-09-06:
what UI or sender shall be used for embroidery mode?
Thanks

terjeio, 2023-09-07:
Any sender that can list files and start streaming a selected one from the SD card -> most?
If it does not have a SD card UI then if $F lists the files in the console you can copy/paste the desired path to a $F=<path> command in the MDI to start streaming?

gwcSoftware, 2024-11-09:
I have tried using iosender with grblHAL ESP32 to access the sd card on an ESP32-Wrover but the sd card initialization always fails. I can access the card using SDMMC mode tutorial provided by Freenove the mfr of the Wrover controller that I have. 

I want to test out the embroidery plugin but I can not get the sd card to mount. 

Any info on how to get access to the sd card?

terjeio, 2024-11-10:
I just downloaded a Web Builder build to my test board and the SD card works with that. 
Which board are you using?
Which version of grlHAL ($I output)?

gwcSoftware, 2024-11-11:
> I just downloaded a Web Builder build to my test board and the SD card works with that. Which board are you using? Which version of grlHAL ($I output)?

What was the board you used?

```
[18:33:02] [ $I ] [VER:1.1f.20241016:]

[18:33:02] [ $I ] [OPT:VNSL,100,1024,3,0]

[18:33:02] [ $I ] [AXS:3:XYZ]

[18:33:02] [ $I ] [NEWOPT:ENUMS,RT+,REBOOT,SED,RTC,WIFI,SD]

[18:33:02] [ $I ] [FIRMWARE:grblHAL]

[18:33:02] [ $I ] [SIGNALS:RHSP]

[18:33:02] [ $I ] [FREE MEMORY:190K]

[18:33:02] [ $I ] [DRIVER:ESP32]

[18:33:02] [ $I ] [DRIVER VERSION:241003]

[18:33:02] [ $I ] [DRIVER OPTIONS:v4.4.3-dirty]

[18:33:02] [ $I ] [BOARD:BlackBox X32]

[18:33:02] [ $I ] [AUX IO:1,0,0,0]

[18:33:02] [ $I ] [WIFI MAC:cc:7b:5c:98:8d:59]

[18:33:02] [ $I ] [IP:192.168.5.1]

[18:33:02] [ $I ] [ESP-NOW Initialized]

[18:33:02] [ $I ] [PLUGIN:SDCARD v1.13]

[18:33:02] [ $I ] [PLUGIN:FS macro plugin v0.09]

[18:33:02] [ $I ] [PLUGIN:EMBROIDERY v0.06]

[18:33:02] [ $I ] ok
```

terjeio, 2024-11-11:
Have you tried cards of different brands?
FYI both my test board and my Blackbox X32 works so I assume the firmware is ok. I am using a 32GB Sandisk card:

[image]

gwcSoftware, 2024-11-11:
Ok, thanks for your response. I'll keep trying on my end. It's good to know you are working on the X32.

terjeio, 2024-11-14:
> Does the WebBuilder use the same source as is available for download from github for a local build?

Yes, but it sometimes lags a little behind when changes are mostly inconsequential for web builds.

> I see the embroidery plugin listed when the controller starts but how do I submit a job for the embroidery plugin to process?

By running it from the SD card with the $F command, e.g. $F=\myfile.pes. More info [here](https://github.com/grblHAL/Plugin_embroidery/blob/main/README.md). You can _not_ send .pes or .dst files via the serial port!

> Does the $f= command only work with the embroidery plugin? 

No, it works with regular gcode as well. The embroidery plugin kicks in when a .pes or .dst file is opened on the SD card and translates the commands directly to motion, bypassing the gcode parser.

> Is there any other configuration needed to enable the plugin?

Yes, see the [readme](https://github.com/grblHAL/Plugin_embroidery/blob/main/README.md).

FYI [ioSender](https://github.com/terjeio/ioSender) can be used to start the files, and has a download and run option that gets a gcode version of the translated file that it uses for displaying before running it from the SD card. See [this discussion](https://github.com/grblHAL/Plugin_embroidery/discussions/1) for some background.

gwcSoftware, 2024-11-14:
Thanks for your response.

I have been able to use iosender to submit a .dst and it runs my x/y steppers. So I am confident that is working now. I just need to get the machine connected to test out the next steps and actually get it stitching.

terjeio, 2024-11-16:
Please add your DST file to a comment so I can have a look.

gwcSoftware, 2024-11-17:
I found what was causing the alarm after moving a short distance on the y-axis. The grbl settings for the embroidery plugin had the sync option enabled. After setting this to 0, when submitting the dst file from the sdcard, the job continues to run. However, the rate that each stitch is being submitted is very slow. Only about one stitch/second! I only have the x/y steppers connected to my controller at this point. What controls the rate of the stitches being read from the sdcard. Even if a gcode file is submitted from the sdcard the rate is the same. If you submit a gcode file that does not reside on the sdcard, it runs at proper speed.

petervanderwalt, 2024-11-18:
> Edit 2: Tried it again after turning x32 off and it is back to running the job slow again as stated above! Settings are the same but only processes a stitch about every second!

Sorry to jump in, but just gotta say, I love hearing that is an X32 being used for this machine conversion, and we would love it if you to post a Build on https://builds.openbuilds.com/ when you have it going - would be a great advantage if others if could learn from your work! If you need any additional help from our side, don't hesitate to tag me in! This is going to be an awesome project

terjeio, 2024-11-18:
@gwcSoftware If your "spindle" is PWM controlled [try this](https://github.com/grblHAL/Plugin_embroidery/discussions/1#discussioncomment-7735831).