grblHAL/ESP32 #81: coordinate not update in candle

PloyPanmaneeCNC, 2023-09-12:
Run grblHAL ESP32 control via candle 1.1 sometime coordinate freezing after run file or freezing after inactive for 5 min. Sometime only number not update as I jog and sometime after finish file, program end didn't change program state I can't abort or reset.
I'll try with candle 1.2 tomorrow.

It's there's any additional config I should do?

terjeio, 2023-09-15:
You have EMI issues causing the controller to crash or the USB connection to fail? Bad USB cable?
E.g. if I connect my ESP32 to my USB hub I sometimes get random failures and I have to reconnect the USB cable to make it work again. When I connect directly to a PC port then it never fails.

PloyPanmaneeCNC, 2023-09-15:
But I connect USB directly.  

Sorry I'm new to this but what is EMI? when I search google it's give me company name.

terjeio, 2023-09-15:
EMI = [ElectroMagnetic Interference](https://en.wikipedia.org/wiki/Electromagnetic_interference).

First thing to try is a differnt (and high quality) USB cable. If possible with a [ferrite bead](https://en.wikipedia.org/wiki/Ferrite_bead). 
I this does not help then do some "dry runs", first without the spindle connected and if this does not help with the spindle and stepper motors disconnected. Be sure to power the system off when disconnecting/connecting. If this resolves the issue then moving and/or shielding cables may help. [More info here](https://openbuilds.com/search/199181855/?q=EMI&o=relevance).

PloyPanmaneeCNC, 2023-09-16:
I see. I'll try and look more about this, Thank You