grblHAL/ESP32 #83: Spindle encoder

arlimbad222, 2023-10-01:
I have spare input in esp32 board. can I use lathe mode with esp32 if yes then kindly inform me how to set gpio pin for spindle encoder.

I am using 1.5 kw servo as a spindle which can do both operation speed and position ,0 to 10v analog for speed& pulse direction for position.

Currently I am planning to make desktop cnc turn mill machine with grbl hal.

terjeio, 2023-10-01:
Spindle sync is not implemented for the ESP32 since the FPU cannot be used from an interrupt context. Someone has to write and test a fixed point PID implementation for it if it is ever to get spindle sync support.

> I am using 1.5 kw servo as a spindle which can do both operation speed and position ,0 to 10v analog for speed& pulse direction for position.

FYI some grblHAL supported processors might get step/dir spindle support as a spin-off from the plasma THC code I am working on. iMXRT1062, STM32F4xx and STM32F7xx is first in line for that.

arlimbad222, 2023-10-01:
Okay can you suggest me best micro controller for my application I want to make cnc mill turn with threading capabilities I am not good at coding so i will use different micro controller.

terjeio, 2023-10-02:
Some iMXRT1062 or STM32F4xx boards/board maps support spindle encoder input/spindle sync as does the MSP432 driver.
The iMXRT1062 T41U5XBB board _can_ be modified for spindle sync but I have not added it as an option to the [Web Builder](http://svn.io-engineering.com:8080).
Note that pin selection for the encoder inputs are limited for the time beeing, only a few are available.