grblHAL/ESP32 #84: Maximum Angular Position

amctgt, 2023-10-06:
I am using grblHAL/ESP32 running on an OpenBuilds BlackBox X32 to control a rotary actuator. I need to spin the part continuously while moving it in the Z direction. To do so, I enter the final Z position and an A position that will get me the desired rotation e.g. if I want to move 1 mm up and rotate 4 times, G0 Z1 A1440. However, for my actual use I need to rotate for several hours. When I enter G0 Z34000000, the BlackBox X32 says it is in "run" mode but it doesn't actually move. Soft limits are not enabled.

Is this due to a limit on the maximum position? Where would I go in the settings or firmware to change this?

terjeio, 2023-10-06:
> Is this due to a limit on the maximum position?

There are two limits, first is the step counter which is 32 bit signed integer - second is the float resolution used for storing the commanded positition which is also 32 bit (single precision).
What is your step/deg setting for the A axis? And what is the number of degrees you commanded in the example (the A word is missing): `G0 Z34000000` - tell me and I will check if limits are somehow hit.

A few days a started playing with repurposing an axis for a stepper driven spindle, this is only controlled by RPM, not distance, so can be run indefinitely. Can this be used in your application or do you need coordinated motion between Z and A?

Longer term I want to create a 64 bit version of grblHAL for processors that has a double precision FPU - when and if that will be done I do not know. Such a version will dramatically increase the range of reachable positions...

amctgt, 2023-10-09:
Thank you for your reply. Please see my responses below:

 

What is your step/deg setting for the A axis? 4.44444 step/deg. And what is the number of degrees you commanded in the example (the A word is missing): G0 Z34000000 - tell me and I will check if limits are somehow hit. Sorry for the typo the command I want to send is G0 Z15 A31104000. Goal is to rotate the part at 4 Hz while moving it up 15mm along the Z axis. It seems like this would be 138239861 steps, which would be lower than the max value of 32 bit signed integer or 32 bit single precision. I can confirm that the G-code command works fine with lower values, I checked and it works up to G0 A3500000 but not at G0 A4000000.

A few days a started playing with repurposing an axis for a stepper driven spindle, this is only controlled by RPM, not distance, so can be run indefinitely. Can this be used in your application or do you need coordinated motion between Z and A? This sounds like it could work for us. We have already modified the firmware to remove the lower limit on movement speed. Would the change to create a stepper driven spindle be something we could easily drop in to our version of the firmware, or could you create a version with the lower speed limit removed?

Longer term I want to create a 64 bit version of grblHAL for processors that has a double precision FPU - when and if that will be done I do not know. Such a version will dramatically increase the range of reachable positions... That would be great, I’m interested in hearing how this turns out!

*CONFIDENTIALITY WARNING*: This message may contain confidential, 
proprietary, or business sensitive information of Top Grain Technologies, 
Inc. If the reader of this message is not the intended recipient, you are 
hereby notified that any review, use, dissemination, distribution, or 
copying of this transmission or its contents is strictly prohibited. If you 
are not the intended recipient(s) please: (1) do not disclose, copy, 
distribute, or use this message or its contents; (2) immediately advise the 
sender by return e-mail; and (3) delete the original and all copies 
(including all attachments) from your computer.

amctgt, 2023-10-16:
Hi Terjeio,

Have you had any luck re-creating my issue? Is there any other information I could provide to help with this?

terjeio, 2023-10-17:
I am away from home for a while and I forgot to bring a mini USB cable for my dev board - and these are hard to come by nowadays. Luckily I managed to get hold of one yesterday and have now verified that the single precision float is the reason - as I suspected. [This calculation](https://github.com/grblHAL/core/blob/04a5edcf8353f7294167d12d32e4b073d9be9ce4/stepper.c#L923) does not change the target value due to limited precision, causing a divide by zero later on.
This means that no grbl port can be used for your application since none I know of uses double precision floats to store floating point variables.

The stepper driven spindle sound like it could be a solution for you, however I am not yet ready to commit the needed changes to github. Anyway, I can provide precompiled firmware for ESP32 if you are willing to test it for me - but I need a bit more time to prepare that. FYI I am going to spend some time on this going forward, using my STM32F446 based dev board as a test platform.
BTW which ESP32 board are you using? Currently there are two step generators available, I am pretty sure the RMT version should work ok but I am not so sure about the one that uses shift registers.

amctgt, 2023-10-17:
Thank you for your reply and for digging up a mini USB! I took a look at the calculation you referenced, and am thinking maybe the simplest solution is to increase the acceleration or go in steps less than 3000000 degrees. I'm testing this now and is seems to be working. Testing by sending "G0 Z0 A3000000", "G0 Z0 A6000000", "G0 Z0 A9000000","G0 Z5 A12000000" and so on, with $123=100000. Do you see any potential issues with running like this?

terjeio, 2023-10-18:
> Do you see any potential issues with running like this?

If it works for you then no. But you are likely still outside the range where single precision floating point math used by grbl/grblHAL will always give the expected results.

amctgt, 2023-10-18:
OK, we are testing the system with a full length run today. I will keep you posted of the results.

I would still be interested in testing the stepper driven spindle code you mention. For our application we had removed the minimum speed limit so we can move in the Z axis at speed around 0.04 mm/min. Could you provide a pre-compiled version of the firmware for the stepper driven spindle with this limit removed so we can test with our application?

terjeio, 2023-10-19:
> Could you provide a pre-compiled version of the firmware for the stepper driven spindle with this limit removed so we can test with our application?

Which board map? Or do you use a custom version?

> For our application we had removed the minimum speed limit...

By this you mean the [MINIMUM_FEED_RATE](https://github.com/grblHAL/core/blob/04a5edcf8353f7294167d12d32e4b073d9be9ce4/config.h#L279) defined in config.h?

terjeio, 2023-10-19:
Here is the firmware, with some luck it should be working.

[firmware.zip](https://github.com/grblHAL/ESP32/files/13044674/firmware.zip)

The stepper spindle "claims" the A-axis motor. The A-axis is still available (for now?) and should not be moved with gcode commands. A-axis settings is in steps/revolution etc. Max travel is ignored.
Since steps are output by the foreground process microstepping should not be set to a too high value, no microstepping is best - also for maximum torque?

amctgt, 2023-10-20:
Great, thank you! We have some projects running on the machine at the moment but I will try to test in the next week or so and will update this thread when I have some results to share.