grblHAL/ESP32 #86: Webbuilder - VFD + Diode laser

boboxx, 2023-11-20:
In the rebuilder how do I configure the web build to have the option to build a FW that support a HY01D523B and a + Diode laser.

From the Web Builder I have selected the MKS Tinybee board, under Spindle Huanyang v1 but the "add PWM Spindle" is not enable. I have also tried adding the "LightBurn cluster" but that did not defined a laser either in ioSender

In FluidNC I was using:
Huanyang:
  uart:
    txd_pin: gpio.17 ### J3 header
    rxd_pin: gpio.16 ### J3 header
    rts_pin: NO_PIN
    its_pin: NO_PIN
    baud: 9600
    mode: 8N1
  modbus_id: 1
  tool_num: 0
  # Typ. Air cooled 120, Water cooled 100 Spindle
  #120hz * 60sec/min = 7200 min speed
  speed_map: 0=0% 0=25% 7200=25% 24000=100%

Laser:
  pwm_hz: 5000
  # on 3D Touch connector (has pdwn + PWM)
  output_pin: gpio.15:high ### EXP2
  enable_pin: NO_PIN
  disable_with_s0: false
  s0_with_disable: true
  tool_num: 1
  speed_map: 0=0.000% 255=100.000%
  #   speed_map: 0=0.000% 1000=100.000%
  off_on_alarm: true

terjeio, 2023-11-21:
> From the Web Builder I have selected the MKS Tinybee board, under Spindle Huanyang v1 but the "add PWM Spindle" is not enable

It is enabled for me, your web browser has cached some incorrect data?

[image]

I have also tried adding the "LightBurn cluster" but that did not defined a laser either in ioSender

"LightBurn cluster" only adds decoding of the non-standard gcode, nothing else.

boboxx, 2023-11-21:
I was able to generate a FW with only Huanyang and a PWM. 

Where is the 19200 baud rate settingcoming from? In FluidNC (and VFD) I was using 9600 baud. If I select 19200 in the web builder and change it later via $374 will it take affect after a reboot or is it hard coded in the firmware?

terjeio, 2023-11-21:
Since the baud rate is a setting it is not hard coded and it will take effect immediately on a change. However, a reboot is still needed after a change to correctly initialize the spindle driver.

boboxx, 2023-11-21:
Perfect thsank you [terjeio]!

closed 2023-11-21 by boboxx