grblHAL/ESP32 #87: Erratic step output lathe mode

JoachimF, 2023-12-17:
Hello,

I have flashed a brand new tinybee with the firmware build on the webbuilder, config in lathe mode, no option, no wifi, nothing.
Just two motors, Z and X, no limits.
Reset params before tests

I have already the step by mm, because the lathe run already on grbl uno.
When I move, from 0.01 to 1mm, the number of step send to tje driver is more than ten times that should be!! And more, it's not regular, sometimes its 10, sometimes 5, and sometimes no step send!

I think a problem from the i2S,  but what..... ?

Spindle sync will be really a must on this board!

Thanks terj for your work

terjeio, 2023-12-17:
> I think a problem from the i2S, but what..... ?

The main issue was that the I2S task had too low priority causing the signals to go crazy. Hopefully fixed now.

> Spindle sync will be really a must on this board!

Since accessing the FPU in an interrupt context crashes the MCU and the current implementation uses it like that I am not to keen to add it myself. Fixed point math is a possible solution - perhaps someone will add it?

JoachimF, 2023-12-17:
I'll test i2s correction tonight or tomorrow, i'll tell you if it's ok

Thanks

JoachimF, 2023-12-18:
Everything is ok, tested on the lathe with backlash compensation.

Thanks

JoachimF, 2023-12-18:
Here is two methods to use float in interrupt, use double then fpu is not used, or start the fpu manually.

https://esp32.com/viewtopic.php?t=1292

I hope this help