grblHAL/ESP32 #90: How to use UART1?

KohajdaG, 2024-01-03:
Hi!

The ESP32 have 3 UART.
UART0 is connected to the USB on PIN 34, 35.
The UART2 is configurable in the map.h.

I made some modification on the MKS DLC32 board, and now the MPG working on the UART2.
But I plan to use TMC2209 motor driver with UART, so I need the UART1 too.

How can I configure it, and use with the trinamic plugin?

terjeio, 2024-01-03:
Code has to be added for it in [uart_serial.c](https://github.com/grblHAL/ESP32/blob/master/main/uart_serial.c) - copy-paste code inside the #if SERIAL2_ENABLE - #endif code blocks to new #if SERIAL3_ENABLE - #endif blocks and rename functions and data from \*2\* to \*3\*, then [trinamic_if.c ](https://github.com/grblHAL/ESP32/blob/21111b1f27aece110a6a2ff3761b46f0293d47ba/main/trinamic_if.c#L264) has to be modified to use a confiigured stream like has been done for [other drivers](https://github.com/grblHAL/STM32F4xx/blob/082445d9f526d89ee0fc8ef714d14de315165f65/Src/tmc_uart.c#L86-L88).

KohajdaG, 2024-01-03:
Thank you!
I made the changes, and the compiler throw an error:

../main/trinamic_if.c:163:10: fatal error: esp32-hal-uart.h: No such file or directory
 #include "esp32-hal-uart.h"
               ^~~~~~~~~~~~~~~~~~

There is no such file in the esp-idf too.

What could be the problem?

terjeio, 2024-01-03:
Oops, my bad. I have renamed the file to _uart-serial.h_ in the latest version. Please change it in your copy for now.

KohajdaG, 2024-01-04:
I swap the Serail2 and Serial3 with MPG_STREAM 2 and TRINAMIC_STREAM 1, and I get the status reports, so the Serial3 working fine. Same time there is no activity on Serial2 with M122 I.

KohajdaG, 2024-01-13:
The drivers arrived, but the EPS32 can't communicate whit it. 
`[MSG:Warning: Could not communicate with stepper driver!]`

The Serail3 configured to GPIO25 & GPIO26.
`#define TRINAMIC_ENABLE 2209
#define TRINAMIC_UART_ENABLE 1`
and set the TMC stream to 2.

Trying [this](https://github.com/grblHAL/ESP32/issues/33#issuecomment-1200725742) getting "Fail!"
I think I have a misstake in the configuration but can't find where.

Can You explain what settings needed to work it correctly?

KohajdaG, 2024-01-13:
Ok, it works, there is some wiring problem.

closed 2024-01-13 by KohajdaG