grblHAL/ESP32 #91: Difficult to compile ESP32 Project!

szelote, 2024-01-14:
download the ESP32-master.ZIP and unzip it,  there are no any files in many sub-directories.  
I spend a lot of time to compile it and always check the failure.

later , I find I can clone by:
git clone --recursive https://github.com/grblHAL/ESP32.git

the files are much more than ZIP downloaded. why different?

but still fail to compile it. because in driver.c
#include "esp32s3/clk.h"        //exit because of error " Not found the file"

comment this #include "esp32s3/clk.h".   the last result is like that . Is it successful?
Used static DRAM:   20388 bytes ( 104192 remain, 16.4% used)
      .data size:    9868 bytes
      .bss  size:   10520 bytes
Used static IRAM:   57110 bytes (  73962 remain, 43.6% used)
      .text size:   56083 bytes
   .vectors size:    1027 bytes
Used Flash size :  263887 bytes
      .text     :  193415 bytes
      .rodata   :   70216 bytes
Total image size:  330865 bytes (.bin may be padded larger)

Spend 2 days to download ,install and compile. I almost lose interest of it .

terjeio, 2024-01-14:
> the files are much more than ZIP downloaded. why different?

Because [github works like that](https://github.com/grblHAL/core/wiki/Compiling-grblHAL#downloading-zip) for some reason.

> #include "esp32s3/clk.h" //exit because of error " Not found the file"

Oops - I have made a mistake, will fix soon.

> Spend 2 days to download ,install and compile. I almost lose interest of it .

The easiest way to build the firmware is to use the [Web Builder](http://svn.io-engineering.com:8080/) - unless you have reasons  for modifying the code.

RealTadango, 2024-03-11:
Using VSCode and the ESP-IDF plugin it compiles perfectly. I don't get the Web builder because it is very limited for DIY machines. It is not hard to create a custom mapping once you understand the little quirks.

closed 2024-06-25 by terjeio