grblHAL/ESP32 #92: "Task watchdog got triggered." when WEBSOCKET_ENABLE or TELNET_ENABLE

grantroux, 2024-01-19:
Hi @terjeio,

Firstly, this is an awesome project, thanks!

I connect to my network successfully when `WIFI_ENABLE` is `1`. However, as soon as I set `WEBSOCKET_ENABLE` or `TELNET_ENABLE` then I get the following error:

```
␛[0;31mE (6674) task_wdt: Task watchdog got triggered. The following tasks did not reset the watchdog in time:␛[0m
␛[0;31mE (6674) task_wdt:  - IDLE (CPU 1)␛[0m
␛[0;31mE (6674) task_wdt: Tasks currently running:␛[0m
␛[0;31mE (6674) task_wdt: CPU 0: IDLE␛[0m
␛[0;31mE (6674) task_wdt: CPU 1: tiT␛[0m
␛[0;31mE (6674) task_wdt: Print CPU 0 (current core) backtrace␛[0m

Backtrace:0x401608DF:0x3FFB0E30 0x4008570D:0x3FFB0E50 0x401637C7:0x3FFBE400 0x4016041F:0x3FFBE420 0x4008D571:0x3FFBE440 0x4008F3CD:0x3FFBE460

␛[0;31mE (6674) task_wdt: Print CPU 1 backtrace␛[0m

Backtrace:0x40089F65:0x3FFB1430 0x4008570D:0x3FFB1450 0x4000BFED:0x3FFC6820 0x4008F63A:0x3FFC6830 0x4008CD9A:0x3FFC6850 0x40107853:0x3FFC6890 0x400F6A44:0x3FFC68C0 0x400F6AFB:0x3FFC68E0 0x4008F3CD:0x3FFC6910
```

My board was purchased from AliExpress and manufactured by Macrobase-tech. It has a ESP32-WROOM-32 and came pre-installed with Grbl_ESP32 / FluidNC. Here is the link: https://www.aliexpress.com/item/1005004644140352.html . I'm compiling in Visual Studio Code using PlatformIO on a Windows 11 machine.

[image]

I have not made any changes to `CMakeLists.txt` and none were required in `grbl/config.h`. Here is my `my_machine.h`:

```
#define BOARD_MY_MACHINE  // Add my_machine_map.h in the boards directory before enabling this!
...
#define WIFI_ENABLE           1 //
...
#if WIFI_ENABLE || ETHERNET_ENABLE || WEBUI_ENABLE
#define TELNET_ENABLE         1 // Telnet daemon - requires WiFi streaming enabled.
#define WEBSOCKET_ENABLE      1 // Websocket daemon - requires WiFi streaming enabled.
...
#define NETWORK_TELNET_PORT  23
//#define NETWORK_HTTP_PORT    80
#if HTTP_ENABLE
//#define NETWORK_WEBSOCKET_PORT  81
#else
#define NETWORK_WEBSOCKET_PORT  80
#endif
#endif // WIFI_ENABLE
```

Here is my boot log and everything immediately before the error log (above):

```
rst:0x1 (POWERON_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:2
load:0x3fff0030,len:7580
ho 0 tail 12 room 4
load:0x40078000,len:14912
load:0x40080400,len:3688
entry 0x4008067c
␛[0;32mI (29) boot: ESP-IDF 4.3.2 2nd stage bootloader␛[0m
␛[0;32mI (29) boot: compile time 22:31:17␛[0m
␛[0;32mI (29) boot: chip revision: 1␛[0m
␛[0;32mI (31) boot_comm: chip revision: 1, min. bootloader chip revision: 0␛[0m
␛[0;32mI (38) boot.esp32: SPI Speed      : 40MHz␛[0m
␛[0;32mI (43) boot.esp32: SPI Mode       : DIO␛[0m
␛[0;32mI (48) boot.esp32: SPI Flash Size : 4MB␛[0m
␛[0;32mI (52) boot: Enabling RNG early entropy source...␛[0m
␛[0;32mI (58) boot: Partition Table:␛[0m
␛[0;32mI (61) boot: ## Label            Usage          Type ST Offset   Length␛[0m
␛[0;32mI (68) boot:  0 nvs              WiFi data        01 02 00009000 00006000␛[0m
␛[0;32mI (76) boot:  1 phy_init         RF data          01 01 0000f000 00001000␛[0m
␛[0;32mI (83) boot:  2 factory          factory app      00 00 00010000 00100000␛[0m
␛[0;32mI (91) boot:  3 grbl             Unknown data     01 99 00110000 00001000␛[0m
␛[0;32mI (98) boot:  4 storage          Unknown data     01 82 00111000 000f0000␛[0m
␛[0;32mI (106) boot: End of partition table␛[0m
␛[0;32mI (110) boot_comm: chip revision: 1, min. application chip revision: 0␛[0m
␛[0;32mI (117) esp_image: segment 0: paddr=00010020 vaddr=3f400020 size=20bcch (134092) map␛[0m
␛[0;32mI (174) esp_image: segment 1: paddr=00030bf4 vaddr=3ffb0000 size=0451ch ( 17692) load␛[0m
␛[0;32mI (181) esp_image: segment 2: paddr=00035118 vaddr=40080000 size=0af00h ( 44800) load␛[0m
␛[0;32mI (200) esp_image: segment 3: paddr=00040020 vaddr=400d0020 size=96528h (615720) map␛[0m
␛[0;32mI (423) esp_image: segment 4: paddr=000d6550 vaddr=4008af00 size=0d614h ( 54804) load␛[0m
␛[0;32mI (446) esp_image: segment 5: paddr=000e3b6c vaddr=50000000 size=00010h (    16) load␛[0m
␛[0;32mI (458) boot: Loaded app from partition at offset 0x10000␛[0m
␛[0;32mI (458) boot: Disabling RNG early entropy source...␛[0m
␛[0;32mI (470) cpu_start: Pro cpu up.␛[0m
␛[0;32mI (470) cpu_start: Starting app cpu, entry point is 0x40084a60␛[0m
␛[0;32mI (0) cpu_start: App cpu up.␛[0m
␛[0;32mI (486) cpu_start: Pro cpu start user code␛[0m
␛[0;32mI (486) cpu_start: cpu freq: 160000000␛[0m
␛[0;32mI (486) cpu_start: Application information:␛[0m
␛[0;32mI (491) cpu_start: Project name:     grbl␛[0m
␛[0;32mI (496) cpu_start: App version:      298bd27-dirty␛[0m
␛[0;32mI (501) cpu_start: Compile time:     Jan 18 2024 22:28:58␛[0m
␛[0;32mI (507) cpu_start: ELF file SHA256:  796505eac07560bc...␛[0m
␛[0;32mI (513) cpu_start: ESP-IDF:          4.3.2␛[0m
␛[0;32mI (518) heap_init: Initializing. RAM available for dynamic allocation:␛[0m
␛[0;32mI (525) heap_init: At 3FFAE6E0 len 00001920 (6 KiB): DRAM␛[0m
␛[0;32mI (531) heap_init: At 3FFBAD10 len 000252F0 (148 KiB): DRAM␛[0m
␛[0;32mI (537) heap_init: At 3FFE0440 len 00003AE0 (14 KiB): D/IRAM␛[0m
␛[0;32mI (544) heap_init: At 3FFE4350 len 0001BCB0 (111 KiB): D/IRAM␛[0m
␛[0;32mI (550) heap_init: At 40098514 len 00007AEC (30 KiB): IRAM␛[0m
␛[0;32mI (558) spi_flash: detected chip: generic␛[0m
␛[0;32mI (561) spi_flash: flash io: dio␛[0m
␛[0;32mI (566) cpu_start: Starting scheduler on PRO CPU.␛[0m
␛[0;32mI (0) cpu_start: Starting scheduler on APP CPU.␛[0m
␛[0;32mI (30) gpio: GPIO[4]| InputEn: 1| OutputEn: 1| OpenDrain: 0| Pullup: 0| Pulldown: 0| Intr:0 ␛[0m
␛[0;32mI (40) gpio: GPIO[12]| InputEn: 1| OutputEn: 1| OpenDrain: 0| Pullup: 0| Pulldown: 0| Intr:0 ␛[0m
␛[0;32mI (40) gpio: GPIO[16]| InputEn: 1| OutputEn: 1| OpenDrain: 0| Pullup: 0| Pulldown: 0| Intr:0 ␛[0m
I (60) wifi:wifi driver task: 3ffc9e18, prio:23, stack:6656, core=0
␛[0;32mI (694) system_api: Base MAC address is not set␛[0m
␛[0;32mI (694) system_api: read default base MAC address from EFUSE␛[0m
I (704) wifi:wifi firmware version: eb52264
I (704) wifi:wifi certification version: v7.0
I (714) wifi:config NVS flash: enabled
I (714) wifi:config nano formating: disabled
I (714) wifi:Init data frame dynamic rx buffer num: 32
I (724) wifi:Init management frame dynamic rx buffer num: 32
I (724) wifi:Init management short buffer num: 32
I (734) wifi:Init dynamic tx buffer num: 32
I (734) wifi:Init static rx buffer size: 1600
I (744) wifi:Init static rx buffer num: 10
I (744) wifi:Init dynamic rx buffer num: 32
␛[0;32mI (754) wifi_init: rx ba win: 6␛[0m
␛[0;32mI (754) wifi_init: tcpip mbox: 32␛[0m
␛[0;32mI (754) wifi_init: udp mbox: 6␛[0m
␛[0;32mI (764) wifi_init: tcp mbox: 6␛[0m
␛[0;32mI (764) wifi_init: tcp tx win: 5744␛[0m
␛[0;32mI (764) wifi_init: tcp rx win: 5744␛[0m
␛[0;32mI (774) wifi_init: tcp mss: 1440␛[0m
␛[0;32mI (774) wifi_init: WiFi IRAM OP enabled␛[0m
␛[0;32mI (784) wifi_init: WiFi RX IRAM OP enabled␛[0m
I (784) wifi:Set ps type: 0

␛[0;32mI (794) phy_init: phy_version 4670,719f9f6,Feb 18 2021,17:07:07␛[0m
I (894) wifi:mode : sta (34:98:7a:65:dd:64)
I (894) wifi:enable tsf
␛[0;32mI (894) gpio: GPIO[35]| InputEn: 1| OutputEn: 0| OpenDrain: 0| Pullup: 0| Pulldown: 0| Intr:0 ␛[0m
␛[0;32mI (904) gpio: GPIO[32]| InputEn: 1| OutputEn: 0| OpenDrain: 0| Pullup: 1| Pulldown: 0| Intr:0 ␛[0m
I (904) wifi:␛[0;32mI (914) gpio: GPIO[33]| InputEn: 1| OutputEn: 0| OpenDrain: 0| Pullup: 1| Pulldown: 0| Intr:0 ␛[0m
new:<1,0>, old:<1,0>, ap:<255,255>, sta:<1,0>, prof:1
␛[0;32mI (924) gpio: GPIO[34]| InputEn: 1| OutputEn: 0| OpenDrain: 0| Pullup: 0| Pulldown: 0| Intr:0 ␛[0m
GrblHAL 1.1f ['$' or '$HELP' for help]
I (1664) wifi:state: init -> auth (b0)
I (1674) wifi:state: auth -> assoc (0)
I (1684) wifi:state: assoc -> run (10)
W (1694) wifi:<ba-add>idx:0 (ifx:0, d4:35:1d:0b:84:9b), tid:0, ssn:0, winSize:64
W (1694) wifi:<ba-add>idx:1 (ifx:0, d4:35:1d:0b:84:9b), tid:5, ssn:0, winSize:64
I (1714) wifi:connected with TelstraABC, aid = 3, channel 1, BW20, bssid = d4:35:1d:0b:84:9b
I (1714) wifi:security: WPA2-PSK, phy: bgn, rssi: -60
I (1714) wifi:pm start, type: 0

I (1794) wifi:AP's beacon interval = 102400 us, DTIM period = 1
[MSG:WIFI STA ACTIVE, IP=192.168.0.113]
␛[0;32mI (2684) esp_netif_handlers: sta ip: 192.168.0.113, mask: 255.255.255.0, gw: 192.168.0.1␛[0m
```

I appreciate any advice / guidance that you may have.

Please let me know if I can provide any more information.

grantroux, 2024-01-20:
I have tried the following ESP-IDF's with no success:

```
ESP-IDF 4.2.0 (compile error)
ESP-IDF 4.3.2
ESP-IDF 4.4.3
ESP-IDF 5.1.2 (compile error)
```

Here is my output from `$I` (after disabling telnet and websockets):

```
[VER:1.1f.20240115:]
[OPT:VNSL,100,1024,3,0]
[AXS:3:XYZ]
[NEWOPT:ENUMS,RT+,REBOOT,SED,RTC,WIFI,SD]
[FIRMWARE:grblHAL]
[SIGNALS:RHS]
[NVS STORAGE:*FLASH]
[FREE MEMORY:180K]
[DRIVER:ESP32]
[DRIVER VERSION:240114]
[DRIVER OPTIONS:4.3.2]
[BOARD:Macrobase 6-axis]
[WIFI MAC:34:98:7a:65:dd:64]
[IP:192.168.0.113]
[PLUGIN:SDCARD v1.12]
[PLUGIN:FS macro plugin v0.06]
ok
```

I have also tried various platforms in my `platformio.ini`:

```
platform =
  espressif32 @ ~4.0.0 # (original)
; espressif32 @ ~5.3.0
; espressif32 @ ~6.5.0 # (breaking changes) 
platform_packages =
; framework-espidf@~3.40200.0 # ESP-IDF 4.2.0
  framework-espidf@~3.40302.0 # ESP-IDF 4.3.2
; framework-espidf@~3.40403.0 # ESP-IDF 4.4.3
framework = espidf
board = esp32dev
```

terjeio, 2024-01-20:
Post the board map for me to try with. If it works for me I'll upload the binaries for you to test with.

grantroux, 2024-01-20:
Thanks! Please find them attached.

[my_machine.txt](https://github.com/grblHAL/ESP32/files/13997487/my_machine.txt)
[my_machine_map.txt](https://github.com/grblHAL/ESP32/files/13997509/my_machine_map.txt)

terjeio, 2024-01-20:
Here you go:

[my_machine.zip](https://github.com/grblHAL/ESP32/files/13997602/my_machine.zip)

```
GrblHAL 1.1f ['$' or '$HELP' for help]
[MSG:WIFI STA ACTIVE, IP=10.0.0.82]
[VER:1.1f.20240118:]
[OPT:VNSL,100,1024,3,0]
[AXS:3:XYZ]
[NEWOPT:ENUMS,RT+,REBOOT,SED,RTC,WIFI,SD]
[FIRMWARE:grblHAL]
[SIGNALS:RHS]
[NVS STORAGE:*FLASH]
[FREE MEMORY:176K]
[DRIVER:ESP32]
[DRIVER VERSION:240119]
[DRIVER OPTIONS:v4.4.6-dirty]
[BOARD:Macrobase 6-axis]
[WIFI MAC:e4:65:b8:19:b5:b0]
[IP:10.0.0.82]
[PLUGIN:SDCARD v1.12]
[PLUGIN:FS macro plugin v0.06]
ok
[MSG:WIFI AP SCAN COMPLETED]
```

Running on a "naked" dev board.

grantroux, 2024-01-20:
That works (i.e. no more errors) on my board too!? Thank you @terjeio.

I will try to build without PlatformIO or investigate other possible causes. You have pointed me in the right direction.

closed 2024-01-22 by grantroux