grblHAL/ESP32 #95: Changing pin mapping - MKS Tinybee

terjeio, 2024-02-12:
i got a question, not related  to the ticket. how can i single control these pins for another project...  https://github.com/makerbase-mks/MKS-TinyBee/issues/113

_Originally posted by @SphaeroX in https://github.com/grblHAL/ESP32/issues/27#issuecomment-1939419368_

terjeio, 2024-02-12:
You can create a new pin map based on the original, easiest is just copying it to _my_machine_map.h_ and enable it in [my_machine.h](https://github.com/grblHAL/ESP32/blob/3935fbd33d2e5391c429f588a77a74bb29a6bb74/main/my_machine.h#L40). FYI there is support for an [I2C I/O expander](https://github.com/grblHAL/ESP32/blob/master/main/ioexpand.c) in the codebase as well, it will be a bit more work to integrate that in a derived board map.

SphaeroX, 2024-02-13:
As I already said, it has nothing to do with grblHAL, I just hope that someone here can kindly help me so that I can switch the outputs. Due to the complexity of the code, I unfortunately can't see through it. I tried to create it in a simple way, unfortunately, the outputs do not switch.

```c
// IOExpander.h
#ifndef IO_EXPANDER_H
#define IO_EXPANDER_H

#define DATA_PIN 27
#define LATCH_PIN 26
#define CLOCK_PIN 25

void setupIOExpander()
{
    pinMode(DATA_PIN, OUTPUT);
    pinMode(LATCH_PIN, OUTPUT);
    pinMode(CLOCK_PIN, OUTPUT);
}

void test()
{
    byte patternHigh = 0b11111111;
    byte patternLow = 0b00000000;

    Serial.println("HIGH");
    digitalWrite(LATCH_PIN, LOW);
    for (int i = 0; i < 3; i++)
    {
        shiftOut(DATA_PIN, CLOCK_PIN, LSBFIRST, patternHigh);
    }
    digitalWrite(LATCH_PIN, HIGH);

    delay(1000);

    Serial.println("LOW");
    digitalWrite(LATCH_PIN, LOW);
    for (int i = 0; i < 3; i++)
    {
        shiftOut(DATA_PIN, CLOCK_PIN, LSBFIRST, patternLow);
    }
    digitalWrite(LATCH_PIN, HIGH);
}

#endif
```

terjeio, 2024-02-13:
Here is the code for [shifting out the data](https://github.com/grblHAL/ESP32/blob/3935fbd33d2e5391c429f588a77a74bb29a6bb74/main/i2s_out.c#L233-L242). You have the latch signal the wrong way? It should be high shifting in, low when latching. See the datasheet for the shift register.

SphaeroX, 2024-02-13:
no idea how I did it but it works now. I had completely disconnected the power and tried to switch and measure only the data pin on the esp332, which worked. After that, the code worked.

heres my code now (working):
```c
// IOExpander.h
#ifndef IO_EXPANDER_H
#define IO_EXPANDER_H

#define NUM_REGISTERS 3   
#define PINS_PER_REGISTER 8 

byte registerStates[NUM_REGISTERS] = {0, 0, 0};

void setupIOExpander()
{
    pinMode(DATA_PIN, OUTPUT);
    pinMode(LATCH_PIN, OUTPUT);
    pinMode(CLOCK_PIN, OUTPUT);
}

void updateShiftRegisters()
{
    digitalWrite(LATCH_PIN, LOW);
    for (int i = NUM_REGISTERS - 1; i >= 0; i--)
    {
        shiftOut(DATA_PIN, CLOCK_PIN, MSBFIRST, registerStates[i]);
    }
    digitalWrite(LATCH_PIN, HIGH);
}

void setPin(int pin, bool state)
{
    int registerIndex = (pin - 1) / PINS_PER_REGISTER;
    int pinIndex = (pin - 1) % PINS_PER_REGISTER;

    if (state)
        registerStates[registerIndex] |= (1 << pinIndex);
    else
        registerStates[registerIndex] &= ~(1 << pinIndex);

    updateShiftRegisters();
}
#endif
```

closed 2024-02-23 by terjeio