grblHAL/ESP32 #99: X limit no pullup?

RealTadango, 2024-03-05:
Using the generic mapping (pins 2,4,15) pin 2(X) has no pullup active? I only measure 600mv. Tested on 2 ESP32 boards.  Can this be a configuration issue or should this pullup always be active?

terjeio, 2024-03-05:
$18 (pullup disable) is not 0?
My dev board works: ~2.5V with pullup enabled, ~0V when disabled.

RealTadango, 2024-03-06:
$18 = 0. If i set it to 1 the X pullup goes to 0... so the pullup is working but very weak. The other 2 pins are at 3.2V and working normally..... strange. I will swap some pins i guess and see if it works correctly as an output

RealTadango, 2024-03-06:
Wait... GPIO02 has a LED? That might interfere.....

But when i use GPIO23 for Probe it works correct with the pullup. I can enable and disable it using $19 and it works correct. If i use GPIO23 for E-Stop (Reset) it does not? I cannot control the pullup using $17 (motor fault pullup works). It always is at 0V...  The pin works for Probe with pullup so it is working correctly. Can you explain this?

terjeio, 2024-03-06:
Missing pullup for estop is a bug, add this code [here](https://github.com/grblHAL/ESP32/blob/fa7b9196ec64ad6bec013a3ff3d5d29f24ba94e4/main/driver.c#L2124) for a quick fix:

```
                case Input_EStop:
                    signal->mode.pull_mode = settings->control_disable_pullup.e_stop ? PullMode_Down : PullMode_Up;
                    signal->mode.inverted = control_fei.e_stop;
                    break;
```

And P2 as a limit input is a bad choice for the generic map, my dev boards has a pulldown resistor connected, yours has a LED.

RealTadango, 2024-03-06:
Thanks for the bugfix, it works. And i now see the remark for my specific board and indeed 2 is pulled low.... All working now as expected :)

closed 2024-03-06 by RealTadango