grblHAL/ESP32 #106: Compilation error

auxp, 2024-04-12:
The ESP32 program has been modified to be different from before.Compilation error of reloaded library.build using ESP-IDF v4.3

CMake Error at ../tools/cmake/component.cmake:475 (add_library):
  Cannot find source file:

    D:/espidf/ESP32/main/networking/strutils.c

  Tried extensions .c .C .c++ .cc .cpp .cxx .cu .m .M .mm .h .hh .h++ .hm
  .hpp .hxx .in .txx
Call Stack (most recent call first):
  main/CMakeLists.txt:239 (idf_component_register)

Can help solve it? Thank you

CMake Error at ../tools/cmake/component.cmake:475 (add_library):
  No SOURCES given to target: __idf_main
Call Stack (most recent call first):
  main/CMakeLists.txt:239 (idf_component_register)

CMake Generate step failed.  Build files cannot be regenerated correctly.
cmake failed with exit code 1

terjeio, 2024-04-12:
If the file is missing then you forgot to [download the submodules](https://github.com/grblHAL/core/wiki/Compiling-grblHAL#using-git-command-line)?

auxp, 2024-04-12:
> 如果文件丢失，那么您忘记[下载子模块](https://github.com/grblHAL/core/wiki/Compiling-grblHAL#using-git-command-line)了吗？

The subroutines have already been downloaded

auxp, 2024-04-12:
> > 如果文件丢失，那么您忘记[下载子模块](https://github.com/grblHAL/core/wiki/Compiling-grblHAL#using-git-command-line)了吗？
> 
> The subroutines have already been downloaded

Download all the latest

terjeio, 2024-04-12:
So ﻿_D:/espidf/ESP32/main/networking/strutils.c_ exists?

auxp, 2024-04-12:
> 那么 D：/espidf/ESP32/main/networking/strutils.c 存在吗？

Yes, there is
[image]

terjeio, 2024-04-12:
Does it help to "clean" the project? Or just delete the content of the build folder?

auxp, 2024-04-12:
> 它有助于“清理”项目吗？或者只是删除构建文件夹的内容？

My mistake, I forgot a subroutine that can now be burned. However, if there is a new issue, the key pin cannot be reversed,
Guru Meditation Error: Core  0 panic'ed (LoadProhibited). Exception was unhandled.
Core  0 register dump:
PC      : 0x400e8b8f  PS      : 0x00060630  A0      : 0x800e2d50  A1      : 0x3ffcbe20  
A2      : 0x00000000  A3      : 0x3ffc84c4  A4      : 0x00000001  A5      : 0x3ffc3bc8  
A6      : 0x3ffcbf33  A7      : 0x00000000  A8      : 0x8008e87e  A9      : 0x3ffbe960  
A10     : 0x00000000  A11     : 0x00000000  A12     : 0x00000000  A13     : 0x3ffc2320  
A14     : 0x00000001  A15     : 0x00000001  SAR     : 0x00000019  EXCCAUSE: 0x0000001c  
EXCVADDR: 0x00000000  LBEG    : 0x400014fd  LEND    : 0x4000150d  LCOUNT  : 0xffffffff  
Backtrace:0x400e8b8c:0x3ffcbe20 0x400e2d4d:0x3ffcbe50 0x400e4114:0x3ffcbe70 0x400f736e:0x3ffcbea0 0x400f7ab1:0x3ffcbec0 0x400f8e07:0x3ffcbef0 0x400efd91:0x3ffcbf20 0x400ece62:0x3ffcbfc0 0x40112f12:0x3ffcc010 0x4011780a:0x3ffcc080 0x4011b61e:0x3ffcc0b0 0x4010d4d9:0x3ffcc0d0
ELF file SHA256: 9415e131e8efd75b
CPU halted.

terjeio, 2024-04-12:
> However, if there is a new issue, the key pin cannot be reversed,

I do not know what the key pin is...

The backtrace makes no sense to me, you have to [decode it](https://esp32.com/viewtopic.php?t=22162) against the .elf file to find out where it goes wrong.

auxp, 2024-04-13:
> > 但是，如果出现新问题，则无法反转键引脚，
> 
> 我不知道钥匙针是什么......
> 
> 回溯对我来说毫无意义，您必须根据 .elf 文件[对其进行解码](https://esp32.com/viewtopic.php?t=22162)才能找出它出错的地方。

YES

closed 2024-04-13 by auxp